slimbootloader/slimbootloader #1252: [TGL] SBL fails to initialize the HD audio codec

PennZheng, 2021-08-06:
Hi,

I encounter an issue on an EC-less Tiger Lake UP3 board that the SBL fails to initialize the high definition audio codec.

The FSP UPD setting is as follows.

1. The PchHdaEnable is set to 1.
2. The PchHdaDspUaaCompliance is set to 1.
3. The PchHdaDspEnable is set to 0.
4. The PchHdaVerbTablePtr is set to the address of the codec verb table.
5. The PchHdaVerbTableEntryNum is set to 1.

Does the SBL support the HD audio codec initialization now?

The following is the Slim Bootloader and UEFI Payload source code version :

Slim Bootloader Project

> Git Repository: https://github.com/slimbootloader/slimbootloader.git
> Download Date: 2021/07/16
> Git Commit ID: ec20bf3

EDK II Project (UEFI Payload)

> Git Repository: https://github.com/tianocore/edk2.git
> Download Date: 2021/07/09
> Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks

gdong1, 2021-08-09:
Thanks PennZheng to report this issue. SBL supports HDA codec, we will look at this issue and back to this thread.

randylintw, 2021-08-11:
Hi Reporter, 
If you use 3rd party BIOS, what kind of driver loaded in the device manage? Is SST type?

randylintw, 2021-08-12:
Can you test it on ubuntu? To see if it related to ACPI.

randylintw, 2021-08-16:
Can you try with this ?
https://github.com/slimbootloader/slimbootloader/pull/1264

PennZheng, 2021-08-20:
I have verified the solution and it works.
Thank you for your support.

closed 2021-08-20 by PennZheng