slimbootloader/slimbootloader #1265: [TGL] S3 information in TSEG area still exists after doing a hard reset

PennZheng, 2021-08-16:
Hi,

In the follow code statement, Slim Bootloader appends some information in the TSEG area to share with UEFI Payload for S3 save/restore purpose.
https://github.com/slimbootloader/slimbootloader/blob/9081525430e236ace87aa2c880cb9264b829259c/Platform/TigerlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c#L973

**I have done the following verification:**
1. The S3 information appended by SBL during last boot still exists after the TGL board is reset by writing 0x06 to reset control register 0xCF9 (**doing a hard reset**). (In the latest SBL version, rebooting the system by executing "reset -w" or "reset -c" command in the EDK2 Shell and clicking restart in the Windows 10 is implemented by writing 0x06 to reset control register 0xCF9.)
2. The S3 information appended by SBL during last boot does not exist anymore after the TGL board is reset by writing 0x0E to reset control register 0xCF9 (**doing a full reset**).

My question is, why does the S3 information in the TSEG area still exist after doing a hard reset? Is that normal?

The following is the Slim Bootloader and UEFI Payload source code version I used:

Slim Bootloader Project

> Git Repository: https://github.com/slimbootloader/slimbootloader.git
> Download Date: 2021/07/16
> Git Commit ID: ec20bf3

EDK II Project (UEFI Payload)

> Git Repository: https://github.com/tianocore/edk2.git
> Download Date: 2021/07/09
> Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks

gdong1, 2021-08-16:
Writing 0x6 ( means soft reset or system reset) to 0xCF9 Often the memory content will not lost.
Writing 0xE ( means hard reset or full reset) to 0xCF9 will cause a full power cycle, so the memory content will lost.
Refer SBL writing warm reset and code reset https://github.com/slimbootloader/slimbootloader/blob/master/BootloaderCommonPkg/Library/ResetSystemLib/ResetSystemLib.c
BTW, Saving info to SMM region for S3 is only used when SMM is supported in UEFI payload. If you are using open sourced UEFI payload (currently SMM is not supported), it might be not necessary to save info to SMM.

PennZheng, 2021-08-16:
The Intel chipset spec says there are two ways to initiate a hard reset:
1.Write 06h to Reset Control register at CF9h (as known as platform Warm Reset) (SLP_S3#, SLP_S4#, SLP_S5# are not asserted)
2.Write 0Eh to Reset Control register at CF9h (as known as platform Cold Reset) (SLP_S3#, SLP_S4#, SLP_S5# are asserted)

So I said writing 0x06 to reset control register 0xCF9 would initiate a hard reset. To aovid ambiguity, I will use the "warm reset" instead of "hard reset".

In the current SBL open source code, **the reserved S3 info after warm reset will cause SBL to hang** since the ​S3 info saved in SMM region is not completely reserved after warm reset.
Specifically, sometimes the CommHdr->TotalSize in the following code statement is 0 after warm reset and it results in infinite loop.
https://github.com/slimbootloader/slimbootloader/blob/9081525430e236ace87aa2c880cb9264b829259c/BootloaderCorePkg/Library/S3SaveRestoreLib/S3SaveRestore.c#L180

I think SBL should clear the S3 info saved in the last boot before the first time append S3 info even if there is no hang issue. Otherwise SBL will run out of the SMM region after many times of warm reset.

Thanks

stalamudupula, 2021-08-16:
Hi PennZheng,

I dont understand how the 'TotalSize' can be 0. Today, that size is calculated in Stage code for each platform from statically defined structs, and can not be 0.

I agree with you, we can clear S3 Save/Restore area on each Non-S3 resume boot, otherwise after few warm resets, TSEG region might be overflowing, and we cant append anymore required data.

PennZheng, 2021-08-17:
Hi,

I encounter an issue on our TGL board that the SBL hangs during the repetitive reboot test. After debug, I find the while loop of AppendS3Info function is dead since the the CommHdr->TotalSize is 0.
It seems that the TotalSize fields of the BL_PLD_COMM_HDR are changed from **0x0028** to **0x0080/0x00FF/0x0000** during warm reset.

The debug log of the next-to-last boot:

> CommHdr = 0x7B000000
> CommHdr->Signature = 0x4F435042
> CommHdr->Id        = 0x00000001
> CommHdr->Count     = 0x00000004
> CommHdr->TotalSize = 0x00000028
> 
> CommHdr = 0x7B000028
> CommHdr->Signature = 0x4F435042
> CommHdr->Id        = 0x00000002
> CommHdr->Count     = 0x00000001
> CommHdr->TotalSize = 0x00000014
> 
> CommHdr = 0x7B00003C
> CommHdr->Signature = 0x4F435042
> CommHdr->Id        = 0x00000001
> CommHdr->Count     = 0x00000004
> CommHdr->TotalSize = 0x000000FF
> 
> CommHdr = 0x7B00013B
> CommHdr->Signature = 0x4F435042
> CommHdr->Id        = 0x00000001
> CommHdr->Count     = 0x00000004
> CommHdr->TotalSize = 0x00000028
> 
> CommHdr = 0x7B000163
> CommHdr->Signature = 0x4F435042
> CommHdr->Id        = 0x00000002
> CommHdr->Count     = 0x00000001
> CommHdr->TotalSize = 0x00000014

The debug log of the last boot:

> CommHdr = 0x7B000000
> CommHdr->Signature = 0x4F435042
> CommHdr->Id        = 0x00000001
> CommHdr->Count     = 0x00000004
> CommHdr->TotalSize = 0x00000028
> 
> CommHdr = 0x7B000028
> CommHdr->Signature = 0x4F435042
> CommHdr->Id        = 0x00000002
> CommHdr->Count     = 0x00000001
> CommHdr->TotalSize = 0x00000014
> 
> CommHdr = 0x7B00003C
> CommHdr->Signature = 0x4F435042
> CommHdr->Id        = 0x00000001
> CommHdr->Count     = 0x00000004
> CommHdr->TotalSize = 0x00000000

Thanks

stanleyintel, 2021-08-18:
Hi,

   Please try with this patch: https://github.com/slimbootloader/slimbootloader/pull/1266

PennZheng, 2021-08-19:
> Hi,
> 
> Please try with this patch: #1266

Hi,

I have verified the patch and it works.

Thanks

stanleyintel, 2021-08-20:
> > Hi,
> > Please try with this patch: #1266
> 
> Hi,
> 
> I have verified the patch and it works.
> 
> Thanks

Glad to hear the issue has been resolved.

PennZheng, 2021-08-20:
Thank you for your support.

closed 2021-08-20 by PennZheng