slimbootloader/slimbootloader #1267: [EHL] CallFspSiliconInit() Change PMODE

tdltaken2009, 2021-08-19:
I want to use GP_F05 as GPIO OUTPUT HIGH

Set with ConfigEditor.py

Now, build and boot

stage1A... OK
stage1B... OK
stage 2...
In PreSiliconInit, GpioInit() function set GP_F05 as HIGH (PMODE = 0)
after PreSiliconInit, CallFspSiliconInit() set GP_F05 as LOW (PMODE = 1)

so i change PMODE to 0 after CallFspSiliconInit(), now HIGH

why CallFspSiliconInit() change GPIO PMODE?

FSP binary is latest, default state

koktong-ong, 2021-08-20:
Hi can you share the generated dlt file from the ConfigEditor.py?

tdltaken2009, 2021-08-20:
[CfgData_Ext_kimjishu.zip](https://github.com/slimbootloader/slimbootloader/files/7020627/CfgData_Ext_kimjishu.zip)

dlt file, here it is.

I found solution. Solution is disable eMMC.

Fspscfg->ScsEmmcEnabled      = 0; // original is 1
Fspscfg->ScsEmmcHs400Enabled = 0; // original is 1
Fspscfg->ScsSdCardEnabled    = 0; // original is 1
Fspscfg->ScsEmmcDdr50Enabled = 0; // original is 1

then GP_F05 Config is not changed.
But i want to use eMMC :(

how can i use both GP_F05 and GP_F05?

koktong-ong, 2021-08-23:
Yes the GP_F05 is for EMMC_HIP_MON if eMMC is using (eg. Fspscfg->ScsEmmcEnabled = 1)
I don't think it can be use for both GP_F05 and eMMC.
@gdong1 
Any comment on this?

gdong1, 2021-08-23:
You could check the Signal Description for EMMC_HIP_MON in EDS. If it is an optional signal for EMMC operation you could check if there is a FSP UPD to skip setting it to EMMC signal. If no such FSP UPD, the WA is to set it to GPIO mode in SBL platform code after CallFspSiliconInit().

mauricema, 2021-09-11:
Per design, current EHL FSP will re-init EMMC related GPIO pins to be used as EMMC functionality.  So it will override the GPIO settings set by SBL.  As Guo mentioned, you can re-init the required pin to GPIO mode after FspSiliconInit if desired.  
Can we close this issue ?

tdltaken2009, 2021-09-11:
i will re-init gpio setting
thank you!

closed 2021-09-11 by tdltaken2009