slimbootloader/slimbootloader #1272: [TGL] Capsule image verification fails after triggering firmware update

PennZheng, 2021-08-24:
Hi,

I encounter an issue that the capsule image verification fails after triggering firmware update

The related SBL debug log:

> =================Read Capsule Image==============
> ...
> Capsule Image found, ImageSize=0xA7528C
> ...
> CapsuleImage: 0x7560E010, CapsuleSize: 0xA7528C
> HASH verification for usage (0x00000400) with Hash Alg (0x2): Security Violation
> ...
> Image verification failed, Security Violation!
> AuthenticateCapsule, Status = 0x8000001A
> Firmware update failed with Status = Security Violation
> Reset required to proceed with the firmware update.

It seems that the **MatchHashInStore** function return an error status.
https://github.com/slimbootloader/slimbootloader/blob/6495e7effb46225de4d5606b392279e1cccfcc36/BootloaderCommonPkg/Library/SecureBootLib/SecureBootHash.c#L199

Do you have any idea about this issue?

The following is the Slim Bootloader and UEFI Payload source code version :

Slim Bootloader Project

> Git Repository: https://github.com/slimbootloader/slimbootloader.git
> Download Date: 2021/07/16
> Git Commit ID: ec20bf3

EDK II Project (UEFI Payload)

> Git Repository: https://github.com/tianocore/edk2.git
> Download Date: 2021/07/09
> Git Commit ID: b560e9d9b67450d3baea56d0e211066722643c57

Thanks

gdong1, 2021-08-24:
from the boot log, the private key used to signing the capsule image is not expected.
Please refer below wiki page on how to sign capsule image.
https://slimbootloader.github.io/security/firmware-update.html

PennZheng, 2021-08-25:
I have followed the instruction to generate capsule image for Tiger Lake platform.
I have tried it with the FirmwareUpdateTestKey_Priv_RSA2048.pem and FirmwareUpdateTestKey_Priv_RSA3072.pem.

> **Slimbootloader binary for capsule image**
> slimbootloader binary generated in outputs\tgl\slimbootloader.bin can be used for generating capsule image
> 
> **Capsule image for Tiger Lake**
> To generate capsule image for Tiger Lake platform:
> python ./BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS Outputs/tgl/SlimBootloader.bin -k <Keys> -o FwuImage.bin

gdong1, 2021-08-25:
During SBL build, it will generate/use SBL key Sblkeys/FirmwareUpdateTestKey_Priv_RSA3072.pem by default to get the public key hash and save to SBL. During capsule generation, please use the same key Sblkeys/FirmwareUpdateTestKey_Priv_RSA3072.pem. Also please check if it is expected capsule image (size, A7528C) or capsule from others. Please add more details (repo commitid, build from windows or Linux, etc.) so we could try if we could reproduce this issue.

PennZheng, 2021-08-30:
> Hi,
> I cannot reproduce the issue from our side with Intel RVP. Some suggestions:
> 
> 1. May you check which boot device/partition/fs is used to carry the FwuImage.bin? You can find the info in Platform/TigerlakeBoardPkg/CfgData/CfgData_CapsuleInformation.yaml
> 2. In EDK2 Shell, before triggering fw update, please make sure the FwuImage.bin is stored in the same boot device specified in (1). We would like to rule out loading a wrong FwuImage.bin (and results in image verification failed).
> 3. In OS Loader, maybe you can try changing the following code around line 361 in PayloadPkg/FirmwareUpdate/FirmwareUpdate.c
>    ResetSystem (EfiResetCold); => ResetSystem (EfiResetWarm);
>    Then, clean build and flash SBL (make sure the new FwUpdate payload in your SPI). And last, try the fw update again.

1. In the EDK2 Shell case, I have verified the file size and file content of capsule image get by SBL built-in Firmware Update Payload and the image restore in the USB flash drive. I also have checked the path of FwuImage.bin defined in the Platform/TigerlakeBoardPkg/CfgData/CfgData_CapsuleInformation.yaml.

2. In the SBL Shell case, SBL hangs at the follow code line.
https://github.com/slimbootloader/slimbootloader/blob/76b1539645d9c0ace8118287d17375540c1913c1/PayloadPkg/FirmwareUpdate/FirmwareUpdate.c#L410
And SBL does not initiate a reset at the follow code line.
https://github.com/slimbootloader/slimbootloader/blob/76b1539645d9c0ace8118287d17375540c1913c1/PayloadPkg/FirmwareUpdate/FirmwareUpdate.c#L361
So your third suggestion does not affects the result.

> Hi,
> One more question: For OS Loader case, Did you modify some settings in Platform/TigerlakeBoardPkg/BoardConfig.py in-between the running SBL and FW update?

No, I didn't.

stanleyintel, 2021-08-30:
> > Hi,
> > I cannot reproduce the issue from our side with Intel RVP. Some suggestions:
> > 
> > 1. May you check which boot device/partition/fs is used to carry the FwuImage.bin? You can find the info in Platform/TigerlakeBoardPkg/CfgData/CfgData_CapsuleInformation.yaml
> > 2. In EDK2 Shell, before triggering fw update, please make sure the FwuImage.bin is stored in the same boot device specified in (1). We would like to rule out loading a wrong FwuImage.bin (and results in image verification failed).
> > 3. In OS Loader, maybe you can try changing the following code around line 361 in PayloadPkg/FirmwareUpdate/FirmwareUpdate.c
> >    ResetSystem (EfiResetCold); => ResetSystem (EfiResetWarm);
> >    Then, clean build and flash SBL (make sure the new FwUpdate payload in your SPI). And last, try the fw update again.
> 
> 1. In the EDK2 Shell case, I have verified the file size and file content of capsule image get by SBL built-in Firmware Update Payload and the image restore in the USB flash drive. I also have checked the path of FwuImage.bin defined in the Platform/TigerlakeBoardPkg/CfgData/CfgData_CapsuleInformation.yaml.
> 2. In the SBL Shell case, SBL hangs at the follow code line.
>    https://github.com/slimbootloader/slimbootloader/blob/76b1539645d9c0ace8118287d17375540c1913c1/PayloadPkg/FirmwareUpdate/FirmwareUpdate.c#L410
>    
>    And SBL does not initiate a reset at the follow code line.
>    https://github.com/slimbootloader/slimbootloader/blob/76b1539645d9c0ace8118287d17375540c1913c1/PayloadPkg/FirmwareUpdate/FirmwareUpdate.c#L361
>    
>    So your third suggestion does not affect the result.
> 
> > Hi,
> > One more question: For OS Loader case, Did you modify some settings in Platform/TigerlakeBoardPkg/BoardConfig.py in-between the running SBL and FW update?
> 
> No, I didn't.

Hi,

   Have you tried with a fresh SBL? I.e., (1) download and integrate your change -> Build -> Stitch -> Flash -> And then (2) Gen capsule firmware -> test fw update?  Note: no code change between (1) and (2).

PennZheng, 2021-08-30:
Hi,

I use a fresh SBL capsule image each time.

stanleyintel, 2021-08-31:
> Hi,
> 
> I use a fresh SBL capsule image each time.

Hi,

   May you try with this patch: https://github.com/slimbootloader/slimbootloader/pull/1278/files ?

   Note: with this change, your SBL must be rebuilt -> stitch -> flash IFWI to your board first. And then based on the new running SBL, please try FW update.

   If it still does not work, may you share us (1) full fw update logs and (2) your BoardConfig.py and CfgData in patch files (please remove any project-specific confidential info)? For any concerns, feel free to contact your BIOS vendor or Intel representative to upload these data via IPS  (Intel Premier Support).

closed 2021-09-06 by PennZheng