slimbootloader/slimbootloader #1302: QEMU SMP and KVM flags cause internal error

mehmetakbulut, 2021-09-24:
Running QEMU with both `-smp` and `--accel kvm` doesn't work. Two variations:
- If SBL is built with `-a ia32`: `KVM internal error. Suberror: 1`
- If SBL is built with `-a x64`: `KVM internal error. Suberror: 3`

Removing either `-smp` or `--accel kvm` mitigates the issue but then you are stuck with either a single core or no KVM acceleration.

SBL bin is produced from a fresh clone of repo in these examples. `EXC.svm` warning may not be significant as I'm on an Intel CPU and not AMD. (cpuinfo for one core is pasted at the end as well)

```
$ qemu-system-x86_64 -smp 2 -m 1G --machine q35 --accel kvm -serial mon:stdio -pflash Outputs/qemu/SlimBootloader.bin
WARNING: Image format was not specified for 'Outputs/qemu/SlimBootloader.bin' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_QEMU 
ISVN: 001
IVER: 001.000.001.001.03467
SVER: 7F461C59E0D513D9
FDBG: BLD(D IA32) FSP(R)
FSPV: ID($QEMFSP$) REV(00001000)
Loader global data @ 0x00001D3C
Run  STAGE1A @ 0x00070000
Load STAGE1B @ 0x00040000
HASH verification for usage (0x00000001) with Hash Alg (0x2): Success

============= Intel Slim Bootloader STAGE1B =============
Host Bridge Device ID:0x29C0
Board ID:0x1 - Loading QEMU!
QEMU Flash: Attempting flash detection at FFC00000
QemuFlashDetected => FD behaves as FLASH
QemuFlashDetected => Yes
SpiInstance = 0000E470
Variable region: 0xFFCA0000:0x2000
Loading Component KEYH:_HS_
Registering container KEYH
HASH verification for usage (0x00000100) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000100): Success
HASH verification for usage (0x00000000) with Hash Alg (0x2): Success
Append public key hash into store: Success
Load EXT CFG Data @ 0x0000EB54:0x0158 ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000200): Success
BOOT: BP0 
MODE: 0
BoardID: 0x01
PlatformName: QEMU_01
Memory Init
Found Config TAG: 0x180 @ 0x0000ECCC
  MemCfg.Test1=11223344, MemCfg.Test2=11223346
Found Config TAG: 0x200 @ 0x0000ECDC
  SilCfg.Test1=11223347, SilCfg.Test2=04030201
Call FspMemoryInit ... Success
Loader global data @ 0x3EEBFD70
Load page table from memory @ 0x3EB74000
Remapping Stage to 0x3EB77000
    FSP Resource HOB Range        Type       Owner
================================= ==== ====================================
0000000000000000-00000000000A0000  00  00000000-0000-0000-0000-000000000000
00000000000A0000-0000000000100000  05  00000000-0000-0000-0000-000000000000
0000000000100000-000000003EF00000  00  00000000-0000-0000-0000-000000000000
000000003EF00000-000000003F000000  05  69A79759-1373-4367-A6C4-C7F59EFD986E
000000003F000000-0000000040000000  05  D038747C-D00C-4980-B319-490199A47D55

Switch to memory stack @ 0x3EEFFF00
Stage1 stack: 0x2000 (0x1220 used)
[... 253 lines cut ...]
model		: 165
model name	: Intel(R) Core(TM) i9-10885H CPU @ 2.40GHz
stepping	: 2
microcode	: 0xea
cpu MHz		: 2400.000
cache size	: 16384 KB
physical id	: 0
siblings	: 16
core id		: 0
cpu cores	: 8
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust sgx bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp pku ospke sgx_lc md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple shadow_vmcs pml ept_mode_based_exec
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs itlb_multihit
bogomips	: 4800.00
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:
```

mauricema, 2021-09-24:
SBL does not fully support "--accel kvm" option.  
But I looked into this issue, and it seems to be caused by some TSEG register emulation on QEMU in some condition.
To workaround this your current issue, you can  comment out the line below:
https://github.com/slimbootloader/slimbootloader/blob/7f461c59e0d513d928622f2ddfa5573dd1d35181/Platform/QemuBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c#L341
Please give a try.

mehmetakbulut, 2021-09-24:
Thank you @mauricema ! I have been scratching my head on this issue for a few days now. That changed works great. For our specific case, "Load boot images" (ID 4070) went from 200+ seconds to ~40 with this.

closed 2021-09-24 by mehmetakbulut

mauricema, 2021-09-24:
@mehmetakbulut:
Great to know that.  :)
I am not sure why the register emulation behavior is different on QEMU w/ and wo "--accel kvm" option.  It might be something to do with QEMU implementation.
If later on you find some clue on how this can be addressed or fixed, please let us know.