slimbootloader/slimbootloader #1321: Building image for Elkhart Lake does not work

finikorg, 2021-10-04:
I am building Elkhart Lake image according to instructions: https://slimbootloader.github.io/supported-hardware/ehl-crb.html

The resulting image does not boot to slimboot Shell, it goes to reboot loop.

rprangar, 2021-10-05:
Hi

Would you be able to attach the log? At what point does it reboot?

Thanks
Ravi

finikorg, 2021-10-05:
Hi, I used all steps described in the open manual on github and got following output:

```
============= Intel Slim Bootloader STAGE1A =============
SBID: SB_EHLK 
ISVN: 001
IVER: 001.000.001.002.00017
SVER: 26BB66CEDFF786E4
FDBG: BLD(D IA32) FSP(R)
FSPV: ID($EHLFSP$) REV(09032734)
Loader global data @ 0xFEF1FD3C
Run  STAGE1A @ 0xFFFD8000
Load STAGE1B @ 0xFFDE0000

============= Intel Slim Bootloader STAGE1B =============
[Boot Guard] AcmStatus : 0x00000000
[Boot Guard] BootStatus: 0x00000000
[Boot Guard] Boot Guard Failed or is Disabled!
SpiInstance = FEF303D0
PchSpiBase at 0x00001F05
ScSpiBar0 at 0xF0000000
Loading Component KEYH:_HS_
Registering container KEYH
HASH verification for usage (0x00000100) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000100): Success
HASH verification for usage (0x00000000) with Hash Alg (0x2): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Append public key hash into store: Success
Load EXT CFG Data @ 0xFEF30B54:0x0EDC ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000200): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Initially set to default BoardId 0x3
Raw data from Input Port 0 is 0x6
Raw data from Input Port 1 is 0x5
Fields.BoardId from Smbus Io expander is 0x5
Fields.BomId from Smbus Io expander is 0x0
Fields.FabId from Smbus Io expander is 0x3
Fields.SpdPresent from Smbus Io expander is 0
Processor does not support Boot Guard.
Boot Guard Support status: 0
GetPlatformDataPtr is copied 0xFEF30B34 
BOOT: BP0 
MODE: 0
BoardID: 0x05
PlatformName: EHL_CRB
GpioInit(0xFFDF2A40:7) Done
Memory Init
Load Security Cfg Data
FSPM CfgData assignment
Load Memory Cfg Data
PlatformDebugConsent = 0
DciEn = 0
DciModphyPg = 0
DciUsb3TypecUfpDbg = 2
DciDbcMode = 4
DebugInterfaceEnable = 2
CpuTraceHubMode = 0
CpuTraceHubMemReg0Size = 2
CpuTraceHubMemReg1Size = 2
PchTraceHubMode = 0
PchTraceHubMemReg0Size = 2
PchTraceHubMemReg1Size = 2
Load Graphics Cfg Data
HDA Policy: SSP/I2S Link Mode
S0ix Status = 0
Call FspMemoryInit ... Success
Loader global data @ 0x7CC0FD70
[... 245 lines cut ...]
Failed to get GFX HOB from FSP
MEM: 0000000000000000 00000000000A0000 00 1
MEM: 00000000000A0000 0000000000060000 00 2
MEM: 0000000000100000 000000007C0E0000 00 1
MEM: 000000007C1E0000 0000000000500000 01 2
MEM: 000000007C6E0000 0000000000068000 00 3
MEM: 000000007C748000 0000000000008000 00 4
MEM: 000000007C750000 0000000000500000 00 2
MEM: 000000007CC50000 00000000003B0000 00 2
MEM: 000000007D000000 0000000002000000 00 2
MEM: 000000007F800000 0000000000800000 00 2
MEM: 00000000FED20000 0000000000060000 00 2
MEM: 00000000FF300000 0000000000D00000 00 2
MEM: 0000000100000000 000000016C000000 00 1
MP Init (Wakeup)
SMRR Base: 0x7F800006  Mask: 0xFF800000
MP Init (Run)
Detected 2 CPU threads
 CPU  0 APIC ID: 0
 CPU  1 APIC ID: 2
SMM rebase done on 2 CPUs
PCI Enum
Call FspNotifyPhase(20) ... 40000003
FSP Requested Reboot ...
```

ywho79, 2021-10-05:
> I am building Elkhart Lake image according to instructions: https://slimbootloader.github.io/supported-hardware/ehl-crb.html
> 
> The resulting image does not boot to slimboot Shell, it goes to reboot loop.

Hi,
Just to be sure are you booting on the EHL CRB or a custom board?

koktong-ong, 2021-10-06:
Hi,

Can you provide the ingredients list that you used to build the IFWI?

finikorg, 2021-10-06:
> > I am building Elkhart Lake image according to instructions: https://slimbootloader.github.io/supported-hardware/ehl-crb.html
> > The resulting image does not boot to slimboot Shell, it goes to reboot loop.
> 
> Hi, Just to be sure are you booting on the EHL CRB or a custom board?

Yes, I am using EHL CRB board

finikorg, 2021-10-06:
> Hi,
> 
> Can you provide the ingredients list that you used to build the IFWI?

I am using steps described in your manual: https://slimbootloader.github.io/supported-hardware/ehl-crb.html

```
python BuildLoader.py build ehl

python Platform/ElkhartlakeBoardPkg/Script/StitchLoader.py -i xxxx.bin -s Outputs/ehl/SlimBootloader.bin -o sbl_ehl_ifwi.bin
```

jinjhuli, 2021-10-07:
Hi, may I know where did you get your xxxx.bin (BIOS Image)?

finikorg, 2021-10-07:
> Hi, may I know where did you get your xxxx.bin (BIOS Image)?

I have got the IFWI images from internal repos, they are working if I flash them separately.

jinjhuli, 2021-10-07:
> > Hi, may I know where did you get your xxxx.bin (BIOS Image)?
> 
> I have got the IFWI images from internal repos, they are working if I flash them separately.

Hi, do you have the version information of the IFWI image?

finikorg, 2021-11-01:
> Hi,
> 
> The issue fixed in commit: [9807395](https://github.com/slimbootloader/slimbootloader/commit/9807395a57bc518d87fcffdd25c691ced0d94150)
> 
> You can try to pull the commit and build the SBL

I made pull and tested the fix. The issue is not fixed, I still have reboot on EHL.