slimbootloader/slimbootloader #1389: fsp memory init time variation how do I get debugging output message during memory training flow.

erik-kline, 2021-11-03:
I am seeing large variation on FSP memory init between IDENTICAL kontron boards.

at https://slimbootloader.github.io/how-tos/configure-memory-down.html?highlight=memory%20down

it is stated
Otherwise, SBL may hang after calling FSP-M and never returns. In this case, you have two troubleshooting options:

   1.  Re-examine the configuration values, make changes and repeat.
    2. Object FSP debug build and get debugging output message during memory training flow.

I have compiled slimboot for a debug fsp (-fd)  but I do not see any debug messages to get a idea of what is going on.

since this step is a large percentage of slimboot boot time. I need to optimize this.

erik-kline, 2021-11-03:
large variation means, from .4 seconds to over 5.  This  is big

erik-kline, 2021-11-03:
this is for kontron tigerlake com express platform. 
the answer is we need to contact our intel representative to get the debug fsp.

mauricema, 2021-12-13:
Since this was followed through Intel IPS support,  close it for now.

closed 2021-12-13 by mauricema