slimbootloader/slimbootloader #1396: Buffer overrun triggered by filesystem size

mehmetakbulut, 2021-11-04:
I have a disk image with a MBR table that contains a single partition with an EXT2 filesystem. The image file is always ~200MB. If the filesystem is <=1525MB in size, then everything works as expected. If the filesystem is >=1530MB in size, then I get the following error:
```
======== Try Booting with Boot Option 1 ========
BootMediumPciBase(0x1F02)
Getting boot image from SATA
Init AHCI controller E00FA000
AHCI port [0] has a [harddisk]
AHCI port [2] has a [cdrom]
Try to find boot partition
AhciDeviceRead Status = Device Error
Partition type: MBR  (1 logical partitions)
Find partition success
Init File system
Detected EXT on StartBlock 1 Part 0
BootSlot = 0x0
ASSERT [OsLoader] /home/mehmet.akbulut/code/av-sbl/BootloaderCommonPkg/Library/FullMemoryAllocationLib/Pool.c(546): Tail->Signature == ((('p') | ('t' << 8)) | ((('a') | ('l' << 8)) << 16))

STAGE_PAYLOAD: System halted!
```

This is built for and running on QEMU with:
`qemu-system-x86_64 -pflash ./SlimBootloader.bin -machine pc-q35-4.2,accel=kvm,usb=off,vmport=on,dump-guest-core=off -cpu qemu64 -m 8196 -smp 4 -hda ./disk.img --serial mon:stdio`

Following are the differences I have from master.
```
BuildLoader.py
-        self.PLD_HEAP_SIZE         = 0x02000000
+        self.PLD_HEAP_SIZE         = 0x0F424000

Platform/QemuBoardPkg/BoardConfig.py
-        self.HAVE_VERIFIED_BOOT       = 1
+        self.HAVE_VERIFIED_BOOT       = 0

Platform/QemuBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c
-    PciOr8 (PCI_LIB_ADDRESS(0, 0, 0, MCH_ESMRAMC), MCH_ESMRAMC_T_EN);
+    //PciOr8 (PCI_LIB_ADDRESS(0, 0, 0, MCH_ESMRAMC), MCH_ESMRAMC_T_EN);
```

Issue occurs on both ia32 and x64 builds. If I make a filesystem image (i.e. no partitions) and pass that to qemu, then it seems to work which is strange. Same disk image can be used to boot EFI applications under UEFI just fine.

mauricema, 2021-11-04:
This assertion is caused by memory deallocation.  Looks like some memory buffer overrun as you indicated. 
Can you post the full SBL log for successful boot and a failing boot ?

mehmetakbulut, 2021-11-04:
Thank you for prompt response Maurice!

A few things to note:
- SBL X64 behavior appears to be same.
- Increasing PLD_HEAP_SIZE doesn't seem to play a role.
- There seems to be something nondeterministic or dirty about my environment. I can't replicate the results for success with 1525M vs failure with 1530M. So I have 300M vs 1000M below.

Success with SBL IA32, 300M Partition
```
$ qemu-system-x86_64 -pflash ./ws/firmware/SlimBootloader.bin -machine pc-q35-4.2,accel=kvm -cpu qemu64 -m 8196 -smp 4 -hda ./ws/disk/disk-300M.img --serial mon:stdio
WARNING: Image format was not specified for './ws/firmware/SlimBootloader.bin' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
WARNING: Image format was not specified for './ws/disk/disk-300M.img' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_QEMU 
ISVN: 001
IVER: 001.000.001.001.04451
SVER: ACCCAEA8539976A0-dirty
FDBG: BLD(D IA32) FSP(R)
FSPV: ID($QEMFSP$) REV(00001000)
Loader global data @ 0x00001D18
Run  STAGE1A @ 0x00070000
Load STAGE1B @ 0x00040000

============= Intel Slim Bootloader STAGE1B =============
Host Bridge Device ID:0x29C0
Board ID:0x1 - Loading QEMU!
QEMU Flash: Attempting flash detection at FFC00000
QemuFlashDetected => FD behaves as FLASH
QemuFlashDetected => Yes
SpiInstance = 0000E480
Variable region: 0xFFCA0000:0x2000
Loading Component KEYH:_HS_
Registering container KEYH
Append public key hash into store: Success
Load EXT CFG Data @ 0x0000EB54:0x0158 ... Success
BOOT: BP0 
MODE: 0
BoardID: 0x01
PlatformName: QEMU_01
Memory Init
Found Config TAG: 0x180 @ 0x0000ECCC
  MemCfg.Test1=11223344, MemCfg.Test2=11223346
Found Config TAG: 0x200 @ 0x0000ECDC
  SilCfg.Test1=11223347, SilCfg.Test2=04030201
Call FspMemoryInit ... Success
Loader global data @ 0x7EEBFD50
Load page table from memory @ 0x7EB74000
Remapping Stage to 0x7EB77000
    FSP Resource HOB Range        Type       Owner
================================= ==== ====================================
0000000000000000-00000000000A0000  00  00000000-0000-0000-0000-000000000000
00000000000A0000-0000000000100000  05  00000000-0000-0000-0000-000000000000
0000000000100000-000000007EF00000  00  00000000-0000-0000-0000-000000000000
000000007EF00000-000000007F000000  05  69A79759-1373-4367-A6C4-C7F59EFD986E
0000000100000000-0000000280400000  00  00000000-0000-0000-0000-000000000000
000000007F000000-0000000080000000  05  D038747C-D00C-4980-B319-490199A47D55

Memory Tolum @ 0x7EF00000
Memory Touum @ 0x280400000
Memory Tom   @ 0x1FF300000
Switch to memory stack @ 0x7EEFFF00
[... 395 lines cut ...]
SdMmcHcReset: reset done with Time out
SdMmcHcReset Fail Status = 0x80000012
Failed to init media - Time out
Failed to Initialize Boot Device - Type 1, Instance 0
Payload normal heap: 0x5F424000 (0xB000 used)
Payload reserved heap: 0x4000 (0x0 used)
Payload stack: 0x10000 (0x548 used)

======== Try Booting with Boot Option 1 ========
BootMediumPciBase(0x1F02)
Getting boot image from SATA
Init AHCI controller E00FA000
AHCI port [0] has a [harddisk]
AHCI port [2] has a [cdrom]
Try to find boot partition
AhciDeviceRead Status = Device Error
Partition type: MBR  (1 logical partitions)
Find partition success
Init File system
Detected EXT on StartBlock 1 Part 0
BootSlot = 0x0
ASSERT [OsLoader] /home/mehmet.akbulut/code/av-sbl/BootloaderCommonPkg/Library/FullMemoryAllocationLib/Pool.c(546): Tail->Signature == ((('p') | ('t' << 8)) | ((('a') | ('l' << 8)) << 16))

STAGE_PAYLOAD: System halted!
```

mauricema, 2021-11-04:
I tried a similar config as yours with SBL, and I did not see same issue.
I have a single image with 1 MBR partition formatted as EXT4.  Then I have one single file IasImage.bin around 80MB on the root. With this config, I did not see any issue.  

======== Try Booting with Boot Option 1 ========
BootMediumPciBase(0x1F02)
Getting boot image from SATA
Init AHCI controller E00FA000
AHCI port [0] has a [harddisk]
AHCI port [2] has a [cdrom]
Try to find boot partition
AhciDeviceRead Status = Device Error
Partition type: MBR  (1 logical partitions)
Find partition success
Init File system
Detected EXT on StartBlock 2048 Part 0
BootSlot = 0x0
File 'iasimage.bin' size 78755776
Get file 'iasimage.bin' (size:0x4B1B7C0) success.
LoadBootImage ImageType-0 Success
ParseBootImage ImageType-0
IAS image verification is skipped!
IAS size = 0x4B1B7C0, file number: 3
IAS Image Type = 0x3
SetupBootImage ImageType-0
Assume BzImage...
Found bzimage Signature

Dump normal boot image info:

mauricema, 2021-11-04:
Forget to mention, the disk partition is 2GB.

mehmetakbulut, 2021-11-04:
Very strange! Would you mind sharing details about your build environment? I'm building SBL natively on Ubuntu 20.04. I don't have gcc-5 installed so I think `gcc` is defaulting gcc-9.

I produce the disk images using [genimage](https://github.com/pengutronix/genimage) but I assume that's fine since it works with SBL if small enough.

mauricema, 2021-11-04:
Here is what I did:
-  I used latest SBL from github.  
- Applied the changes you made.
- Build SBL image using Ubuntu Ubuntu 20.04.2 LTS with GCC 9.3

I generated disk image by:
-  Use "dd" to create zero filled image file with size of 2GB.
-  Use "fdisk" to create MBR partition and create one primary partition.
-  Use  "kpartx" to map the disk image into loop device
-  Use  "mkfs.ext4" to format the partition to EXT4.
-  Use  "mount" to mount the new partition
-  Finally, copy the IasImage.bin into the root, and then "unmount"

BTW,  some tips here to boot from SATA directly without trying MMC first,  you can add "-boot order=d" in QEMU command line.

mehmetakbulut, 2021-11-04:
Just feeding `-boot order=d` seems enough to "fix" this issue for 1000M which was otherwise failing. Doesn't seem to be enough for 1500M+ though. Logs are attached if you are curious.

I'll try a fresh build using your steps in a Ubuntu 20.04 container instead of natively to see if that helps.

Failure with `-boot order=d` and 1500M Partition:
```
$ qemu-system-x86_64 -pflash ./ws/firmware/SlimBootloader.bin -machine pc-q35-4.2,accel=kvm -cpu qemu64 -m 8196 -smp 4 -hda ./ws/disk/disk-1500M.img --serial mon:stdio --boot order=d
WARNING: Image format was not specified for './ws/firmware/SlimBootloader.bin' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
WARNING: Image format was not specified for './ws/disk/disk-1500M.img' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.80000001H:ECX.svm [bit 2]

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_QEMU 
ISVN: 001
IVER: 001.000.001.001.04451
SVER: ACCCAEA8539976A0-dirty
FDBG: BLD(D IA32) FSP(R)
FSPV: ID($QEMFSP$) REV(00001000)
Loader global data @ 0x00001D18
Run  STAGE1A @ 0x00070000
Load STAGE1B @ 0x00040000

============= Intel Slim Bootloader STAGE1B =============
Host Bridge Device ID:0x29C0
Board ID:0x1 - Loading QEMU!
QEMU Flash: Attempting flash detection at FFC00000
QemuFlashDetected => FD behaves as FLASH
QemuFlashDetected => Yes
SpiInstance = 0000E480
Variable region: 0xFFCA0000:0x2000
Loading Component KEYH:_HS_
Registering container KEYH
Append public key hash into store: Success
Load EXT CFG Data @ 0x0000EB54:0x0158 ... Success
BOOT: BP0 
MODE: 0
BoardID: 0x01
PlatformName: QEMU_01
Memory Init
Found Config TAG: 0x180 @ 0x0000ECCC
  MemCfg.Test1=11223344, MemCfg.Test2=11223346
Found Config TAG: 0x200 @ 0x0000ECDC
  SilCfg.Test1=11223347, SilCfg.Test2=04030201
Call FspMemoryInit ... Success
Loader global data @ 0x7EEBFD50
Load page table from memory @ 0x7EB74000
Remapping Stage to 0x7EB77000
    FSP Resource HOB Range        Type       Owner
================================= ==== ====================================
0000000000000000-00000000000A0000  00  00000000-0000-0000-0000-000000000000
00000000000A0000-0000000000100000  05  00000000-0000-0000-0000-000000000000
0000000000100000-000000007EF00000  00  00000000-0000-0000-0000-000000000000
000000007EF00000-000000007F000000  05  69A79759-1373-4367-A6C4-C7F59EFD986E
0000000100000000-0000000280400000  00  00000000-0000-0000-0000-000000000000
000000007F000000-0000000080000000  05  D038747C-D00C-4980-B319-490199A47D55

Memory Tolum @ 0x7EF00000
Memory Touum @ 0x280400000
Memory Tom   @ 0x1FF300000
Switch to memory stack @ 0x7EEFFF00
Stage1 stack: 0x2000 (0xBA4 used)
Stage1 heap: 0xE000 (0x1F10 used, 0x1F40 max used)
Call FspTempRamExit ... Success
[... 369 lines cut ...]
 3150 |     403 ms |       0 ms | Extend payload hash
 31A0 |     404 ms |       1 ms | Board PostPayloadLoading hook
 31B0 |     405 ms |       1 ms | Decode payload format
 31F0 |     405 ms |       0 ms | End of stage2
 4000 |     410 ms |       5 ms | Payload entry point
 4010 |     413 ms |       3 ms | OS loader main entry
 4020 |    6416 ms |    6003 ms | Shell exit
 4040 |    6424 ms |       8 ms | Load image entry
 4050 |    6612 ms |     188 ms | Boot device init
 4055 |    6612 ms |       0 ms | Boot device tuning
 4060 |    6618 ms |       6 ms | Parse partition info
 4070 |  172715 ms |  166097 ms | Load boot images
 4080 |  172717 ms |       2 ms | Verify IAS image
 40A0 |  172718 ms |       1 ms | Process IAS type
 40B0 |  172837 ms |     119 ms | Process ELF/MultiBoot
 40F0 |  172866 ms |      29 ms | FSP ReadyToBoot/EndOfFirmware notify
 4100 |  172866 ms |       0 ms | TPM IndicateReadyToBoot
------+------------+------------+----------------------------------

Starting Kernel ...

Switch to LongMode and jump to 64-bit kernel entrypoint ...
QEMU 4.2.1 monitor - type 'help' for more information
(qemu) quit
```

mauricema, 2021-11-04:
Frankly, I don't know how boot order will be related to this issue, and how partition size makes a difference.
We might have poetical issue in code, but without reproducing, it is going to be hard to debug.  

Do you think you can provide a test disk disk for us to debug ?  Or if it is difficult, can you help put more debug msg to figure out which high level function triggered this issue?  I guess something inside LoadBootImages () / GetBootImageFromFs ().  But if you can help provide more info, it might be helpful for us to analyze.

mehmetakbulut, 2021-11-04:
I repeated your steps to produce disk images and that was it. Seems to be caused by a difference in the partition's start/end sectors. `genimage-style.img` does not boot but `maurice-style.img` does.

```
$ fdisk -l /tmp/genimage-style.img
Disk /tmp/genimage-style.img: 1.98 GiB, 2097152512 bytes, 4096001 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device                   Boot Start     End Sectors Size Id Type
/tmp/genimage-style.img1          1 4096000 4096000   2G 83 Linux

$ fdisk -l /tmp/maurice-style.img
Disk /tmp/maurice-style.img: 2 GiB, 2147483648 bytes, 4194304 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2f6fdf38

Device                  Boot Start     End Sectors Size Id Type
/tmp/maurice-style.img1       2048 4194303 4192256   2G 83 Linux
```

It seems either genimage (+ my config) is producing bad images or SBL is making some assumptions about partition layout. This is good enough to unblock me though. If you would like to look more into this, I am happy to try to get more verbose logs as you suggested, produce some disk images with vanilla Ubuntu kernel in it as an example or just provide more complete reproduction instructions.

mauricema, 2021-11-04:
Thanks for the updates. Glad to know you are unblocked.
I want to try genimage.  It needs a cfg file,  will you be able to provide the cfg that you used ?

mehmetakbulut, 2021-11-04:
Sure thing! 
```
image boot.ext2 {
  ext2 {
  }
  size = 5G
  mountpoint = "${rootpath} plus ${mountpoint} should point to dir containing boot.kpi"
}

image disk.img {
  hdimage {
    gpt = false
    fill = true
  }

  partition boot {
    image = "boot.ext2"
    partition-type = 0x83
  }
}
```
^ Technically my cfg file looks a bit different because I call genimage via buildroot but this should be a close approximation.

Also just for reference, I make the IAS image as below:
```
iasimage create -o path/to/your/boot.kpi -i 0x30000 path/to/your/cmdline.txt path/to/your/bzImage
```

mehmetakbulut, 2021-11-04:
I also tried to offset my partition with `offset = 1M` in `partition boot` section of genimage cfg which does produce images looking more like yours but apparently that wasn't enough to boot successfully. So there is a bit more than meets the eye. I'll do more digging.

```
$ fdisk -l ./disk.img
Disk ./disk.img: 2 GiB, 2148532224 bytes, 4196352 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device         Boot Start End Sectors Size Id Type
./disk.img1       2048 4196351 4194304   2G 83 Linux
```

mauricema, 2021-11-04:
Thanks,  I'll give a try.

mauricema, 2021-11-04:
image boot.ext2 {
  ext2 {
    use-mke2fs = true
  }
  size = 2G
}

image disk.img {
  hdimage {
    gpt = false
    fill = true
  }

  partition boot {
    image = "boot.ext2"
    partition-type = 0x83
  }
}

Can you try this ?   Added "use-mke2fs = true".
It might be some FS compatibility issue.

mehmetakbulut, 2021-11-05:
`use-mke2fs = true` did it! It seems good with any partition size now. I see you already have a MR to address it!

mauricema, 2021-11-05:
😄   Glad to know that.
Yes,  I tried genimage, but it gave me different error though.  What I got was an exception.  By default, genimage will use genext2fs to create ext2 filesystem. And it will use old revision 0.  With rev 0, many fields are all 0 in the superblock, and some of the code cannot handle it correctly.  Tried to resolve the issue I had with the PR.  I think we should just move to rev 1.  mke2fs will generate rev 1 filesystem.