slimbootloader/slimbootloader #1400: Consider using LF line endings in the SlimBootloader Git repository

mikecrowe, 2021-11-05:
The SlimBootloader Git repository uses DOS-style CRLF line endings on all its text files in the repository. It appears that this is intentional since I see various commits fixing files that had different line endings to use CRLF. The requirement may come from the [EDK2 coding standards)[https://edk2-docs.gitbook.io/edk-ii-c-coding-standards-specification/5_source_files] but no rationale is given there. It looks like it [might be for historic reasons](https://www.mail-archive.com/edk2-devel@lists.01.org/msg00865.html) that no longer apply.

Text files with CRLF line endings are annoying to use from Linux. Whilst most editors can be configured to accept and preserve them, the line endings will often be changed when running files through other tools. Even running the Python scripts included in SlimBootloader rewrites files to use only LF line endings.

Modern Git's support for CRLF line endings in work trees when checking out on Windows but LF line endings on Linux and MacOS is generally good. However, this requires that the files are stored in the repository with LF line endings. In my research I've been unable to find a standard way[1] for standard Git to use LF line endings in the work tree when the repository blobs use CRLF. If the blobs in the repository use LF then it is straightforward for clients on Linux and MacOS to use LF in the work tree whilst clients on Windows will use CRLF in the work tree. See https://docs.github.com/en/get-started/getting-started-with-git/configuring-git-to-handle-line-endings. If any files need to have CRLF line endings even on Linux then this is possible using `gitattributes` directives.

SlimBootloader is a great way to boot Linux on Intel hardware. Developers using Linux for their products will most likely be building their software on Linux and will want to edit and compile SlimBootloader from Linux.

It's clear that files with only LF endings are sneaking into the repository (there are currently at least six), and using Git's standard mechanisms ought to avoid inconsistent line endings being introduced in the future.

Please can you consider switching to using LF line endings in the repository?

Thanks.

Mike.

[1] It is possible to use filters to smudge files to change the line endings as they are checked out. However, this requires extra configuration after cloning the repository. It doesn't happen automatically.

mauricema, 2021-11-05:
Thank you for the suggestion.  The primary reason that we used CRLF is just to be in sync with EDK2 since we leverage some of the code direct from EDK2 project, and we don't want to cause unnecessary issue to us.  What you mentioned is reasonable to me.  Let me bring this topic back to the team and then update here on the results.

mikecrowe, 2021-11-07:
Thanks for the quick reply. Let me know if you'd like me to open an issue somewhere else.

mikecrowe, 2021-11-08:
Just in case anyone else is interested, here's the `.gitattributes` file I'm currently using to work around this:
```
# When this file is changed run
#  git add --renormalize .
# to determine whether any files need more work.

# By default, all files in the repository use DOS line endings, but we
# want to use LF only in the work tree.
*	filter=fix-line-endings

# But binary files should not be modified
*.dat filter=
*.bin filter=
*.bmp filter=
*.jpg filter=

# These files have LF line endings even in the upstream repository.
BaseTools/BinWrappers/PosixLike/* filter=
BaseTools/Bin/CYGWIN_NT-5.1-i686/* filter=
BaseTools/BinWrappers/WindowsLike/GenerateCapsule.bat filter=
BaseTools/BuildEnv filter=
BaseTools/Scripts/GccBase.lds filter=
BaseTools/Source/C/Common/PcdValueCommon.h filter=
BaseTools/Source/C/Include/IndustryStandard/Bluetooth.h filter=
BaseTools/Source/C/Include/Protocol/DevicePathUtilities.h filter=
BaseTools/Tests/GNUmakefile filter=
BaseTools/Tests/PythonTest.py filter=
BaseTools/Tests/RunTests.py filter=
MdePkg/Library/BaseIoLibIntrinsic/Ia32/IoFifo.nasm filter=
MdePkg/Library/BaseLib/Ia32/Lfence.nasm filter=
MdePkg/Library/BaseLib/Ia32/WriteTr.nasm filter=
Platform/ApollolakeBoardPkg/ApollolakeBoardPkg.dec filter=
Silicon/QemuSocPkg/FspBin/Patches/0001-Build-QEMU-FSP-2.0-binaries.patch filter=
```
Along with:
```
git config --local filter.fix-line-endings.smudge dos2unix
git config --local filter.fix-line-endings.clean unix2dos
git add --renormalize .
```

mauricema, 2021-11-10:
We had some wider discussions with the team.  And the team seems to be OK with this recommendation.
Since you have done a lot to get it sorted out,  are you willing to submit a PR to fix it ?

closed 2021-11-10 by mauricema