slimbootloader/slimbootloader #1422: SBL does not recognize USB stick

finikorg, 2021-11-25:
I built latest SBL for EHL. For some reason I cannot boot from USB. When I try to init fs I have following errors:

```
Shell> fs init 5:0
Init USB XHCI - Success
XhcPeiCheckUrbResult: TRANSACTION_ERROR! Completecode = 4
Not supported EPType 0x0!
XhcPeiCreateUrb: XhcPeiCreateTransferTrb Failed, Status = Unsupported
XhcPeiControlTransfer: failed to create URBXhcPeiControlTransfer: error - Out of Resources, transfer - 100
Not supported EPType 0x0!
XhcPeiCreateUrb: XhcPeiCreateTransferTrb Failed, Status = Unsupported
XhcPeiControlTransfer: failed to create URBXhcPeiControlTransfer: error - Out of Resources, transfer - 100
Not supported EPType 0x0!
XhcPeiCreateUrb: XhcPeiCreateTransferTrb Failed, Status = Unsupported
XhcPeiControlTransfer: failed to create URBXhcPeiControlTransfer: error - Out of Resources, transfer - 100
PeiUsbGet Device Descriptor fail: 3 Out of Resources
Not supported EPType 0x0!
XhcPeiCreateUrb: XhcPeiCreateTransferTrb Failed, Status = Unsupported
XhcPeiControlTransfer: failed to create URBXhcPeiControlTransfer: error - Out of Resources, transfer - 100
Not supported EPType 0x0!
XhcPeiCreateUrb: XhcPeiCreateTransferTrb Failed, Status = Unsupported
XhcPeiControlTransfer: failed to create URBXhcPeiControlTransfer: error - Out of Resources, transfer - 100
Not supported EPType 0x0!
XhcPeiCreateUrb: XhcPeiCreateTransferTrb Failed, Status = Unsupported
XhcPeiControlTransfer: failed to create URBXhcPeiControlTransfer: error - Out of Resources, transfer - 100
PeiUsbGet Device Descriptor fail: 3 Out of Resources
Enumerate Bus - Success
Found 0 USB devices on bus
Media(USB) Init Fail!
Deinit USB controller
```
I tried USB formatted as EXT2 and FAT32

finikorg, 2021-11-30:
It looks like it does not recognize USB stick if it is connected through USB splitter (Cleware).

mauricema, 2022-01-11:
At this moment, the team does not have this special USB device to debug this issue.  And we tested several different USB hubs, and seemed to work fine.  So there is no update so far on this issue.
Since it might be USB device compatibility issue, we would recommend to use a different USB hub/switch in this case.   We might not be able to look into the issue unless we get the proper hardware.

mauricema, 2022-01-24:
Since we don't have the hardware to debug further, plan to close this for now.  When we have device to reproduce this issue,  can reopen it.

closed 2022-01-27 by mauricema