slimbootloader/slimbootloader #1442: Build error for EHL

shawnhsiao97, 2021-12-22:
Hello ,
I built slimbootloader for EHL, I just used follow cmd to build:
`sudo python3 BuildLoader.py build ehl`
But it is encounter follow error :
```
 GCC5/FV/SIIPFW.bin
Patching STAGE1A
Patched offset 0x00027FFC:[90909090] with value 0xFFFDF000  # Patch BFV
ERROR: Unknown symbol Stage1A:__ModuleEntryPoint !
Error in running process:
  /usr/bin/python3 BootloaderCorePkg/Tools/PatchFv.py /home/shawn/ElkhartPSE/intel_SLB/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV STAGE1A:STAGE1A 0xFFFFFFFC,            _BASE_STAGE1A_,                    @Patch BFV _OFFS_STAGE1A_,        Stage1A:__ModuleEntryPoint,        @Patch Stage1A Entry _OFFS_STAGE1A_+4,      Stage1A:BASE,                      @Patch Module Base <Stage1A:__gPcd_BinaryPatch_PcdVerInfoBase>,  {3473A022-C3C2-4964-B309-22B3DFB0B6CA:0x1C}, @Patch VerInfo <Stage1A:__gPcd_BinaryPatch_PcdFileDataBase>, {EFAC3859-B680-4232-A159-F886F2AE0B83:0x1C}, @Patch PcdBase 0xFFFFFFF8, {3CEA8EF3-95FC-476F-ABA5-7EC5DFA1D77B:0x1C}, @Patch FlashMap 0xFFFFFFC0,        ({CD17FF5E-7731-4D16-8441-FC7A113C392F:0x1C} + 0x3F) & ~0x3F,  @FIT table <[0xFFFFFFC0]>+0,  0x5449465F,                                                    @FIT Signature Low <[0xFFFFFFC0]>+4,  0x2020205F,                                                    @FIT Signature High <[0xFFFFFFC0]>+8,  [CD17FF5E-7731-4D16-8441-FC7A113C392F:0x18] & 0xFFFFFF ,       @FIT FFS section length <[0xFFFFFFC0]>+8,  {CD17FF5E-7731-4D16-8441-FC7A113C392F:0x1C} + [[0xFFFFFFC0] + 8] - [0xFFFFFFC0],  @FIT table max length <Stage1A:__gPcd_BinaryPatch_PcdHashStoreBase>, {18EDB1DF-1DBE-4EC5-8E26-C44808B546E1:0x1C}, @Patch HashStore
```
Here is my toolchain:
```
Checking Toolchain Versions...
- /usr/bin/python3: Version 3.8.10 (>= 3.6.0) [PASS]
- /usr/bin/openssl: Version 1.1.1f (>= 1.1.0g) [PASS]
- /usr/bin/nasm: Version 2.15.04 (>= 2.12.02) [PASS]
- /usr/bin/iasl: Version 20190509 (>= 20160422) [PASS]
- /usr/bin/git: Version 2.27.0 (>= 2.20.0) [PASS]
- /usr/bin/gcc: Version 10.3.0 (>= 7.3) [PASS]
...Done!

```

Please help for me. Thank!

jgutbub, 2021-12-22:
With my toolchain the same build is able to pass in Linux build.

Checking Toolchain Versions...
- /usr/bin/python3.7: Version 3.7.5 (>= 3.6.0) [PASS]
- /usr/bin/openssl: Version 1.1.1 (>= 1.1.0g) [PASS]
- /usr/bin/nasm: Version 2.13.02 (>= 2.12.02) [PASS]
- /usr/bin/iasl: Version 20190509 (>= 20160422) [PASS]
- /usr/bin/git: Version 2.17.1 (>= 2.20.0) [RECOMMEND]
- /usr/bin/gcc: Version 7.5.0 (>= 7.3) [PASS]
...Done!

Also seeing that in recent Azure build this environment is working for EHL Linux build.

Checking Toolchain Versions...
- /opt/hostedtoolcache/Python/3.6.15/x64/bin/python: Version 3.6.15 (>= 3.6.0) [PASS]
- /usr/bin/openssl: Version 1.1.1f (>= 1.1.0g) [PASS]
- /usr/bin/nasm: Version 2.14.02 (>= 2.12.02) [PASS]
- /usr/bin/iasl: Version 20190509 (>= 20160422) [PASS]
- /usr/bin/git: Version 2.34.1 (>= 2.20.0) [PASS]
- /usr/bin/gcc: Version 9.3.0 (>= 7.3) [PASS]
...Done!

I know it does not use the same toolchains you are using but it does confirm latest code is able to build. I would probably guess your gcc compiler version might be related to the issue.

mauricema, 2021-12-22:
Since you are using GCC 10.3,  it might have not been tested and supported yet.    The error was caused by map file parsing.  Maybe the generated map file format changed a little bit with the new GCC tool.
Maybe you can try GCC 9.3 instead.

shawnhsiao97, 2021-12-22:
Hi @mauricema  and @jgutbub 
Thank you for help and reply.
I changed my gcc version to 9.3 ,now as mentioned above issue is solved.
However I encountered new error while building again. 

error log:

```
make: 對「tbuild」無需做任何事。
Building ... /home/shawn/ElkhartPSE/intel_SLB/slimbootloader/Silicon/ElkhartlakePkg/Library/Stage1BSocInitLib/Stage1BSocInitLib.inf [IA32]
make: 對「tbuild」無需做任何事。
Building ... /home/shawn/ElkhartPSE/intel_SLB/slimbootloader/Silicon/CommonSocPkg/Library/SpiFlashLib/SpiFlashLib.inf [IA32]
lto1: internal compiler error: compressed stream: data error
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-9/README.Bugs> for instructions.
lto-wrapper: fatal error: gcc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [GNUmakefile:296：/home/shawn/ElkhartPSE/intel_SLB/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/BootloaderCorePkg/PcdData/PcdData/DEBUG/PcdData.dll] 錯誤 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/shawn/ElkhartPSE/intel_SLB/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/BootloaderCorePkg/PcdData/PcdData]

build.py...
 : error F002: Failed to build module
	/home/shawn/ElkhartPSE/intel_SLB/slimbootloader/BootloaderCorePkg/PcdData/PcdData.inf [IA32, GCC5, DEBUG]

- Failed -
Build end time: 16:21:38, Dec.22 2021
Build total time: 00:00:04

Error in running process:
  build --platform BootloaderCorePkg/BootloaderCorePkg.dsc -b DEBUG --arch IA32 --tagname GCC5 -n 40 -y Report.log -Y PCD -Y FLASH -Y LIBRARY
```
Here is toolchain:
```

Checking Toolchain Versions...
- /usr/bin/python3: Version 3.8.10 (>= 3.6.0) [PASS]
- /usr/bin/openssl: Version 1.1.1f (>= 1.1.0g) [PASS]
- /usr/bin/nasm: Version 2.15.04 (>= 2.12.02) [PASS]
- /usr/bin/iasl: Version 20190509 (>= 20160422) [PASS]
- /usr/bin/git: Version 2.27.0 (>= 2.20.0) [PASS]
- /usr/bin/gcc: Version 9.3.0 (>= 7.3) [PASS]
...Done!
```

mauricema, 2021-12-22:
Can you make sure you rebuild your BaseTools by removing files under "BaseTools/Source/C/bin" ?
Or can you try a new "git clone" and then build ?   
Since azure pipeline worked fine,  I don't know what else could be different in your environment.

shawnhsiao97, 2021-12-23:
Hi @mauricema Thank for reply
I tried to both a new git cloen and` sudo python3 BuildLoader.py clean ` ,and then build.
Unfortunately both of failure.

mauricema, 2021-12-23:
I tried it on my Linux box as well.  And I was able to build SBL EHL without any issue with Ubuntu 20.04 (GCC 9.3).
At this moment, we think it is your build environment issue.  Please check it.
You can also try the build on a different machine to see if you still have this issue.

closed 2021-12-27 by shawnhsiao97