slimbootloader/slimbootloader #1464: The igfx can not display on Tiger Lake UP3 CRB.

Advgcipc, 2022-01-19:
Tiger Lake UP3 frame buffer can not initial successfully.

I also do the same thing on Tiger Lake RVP DDR4 CRB. It work fine.

Did any suggestion to us? Attach debug log.

>Tiger Lake UP3 DDR4 debug log

============= Intel Slim Bootloader STAGE2 =============
PCH gen (0x5), series (0x2), stepping (0x10)
LPC DeviceId (0xA088)
Boot-time Pch Dts: 0x136
Variable region: 0xFF60E000:0x2000
SaveNvsData  Length=0xFAA8
Matched with saved NVS data
Save MRC Training Data (0x79C96110 0x00FAA8) ... Already started
Silicon Init
PchLp is detected
BIOS SPI region will be protected
IAS  SPI region will be protected
Mike PcdFramebufferInitEnabled :GraphicsConfigPtr=0
Select VBT ImageId 0x00000001
Mike PcdFramebufferInitEnabled :GraphicsConfigPtr=79A5701C
Load Silicon Cfg Data
Call FspSiliconInit ... 
Success
>FspGfxHob is not available
Boot-time Cpu DTS: 0x12C
>Failed to get GFX HOB from FSP
Found a gap in memory map: 7E000000 - 00000000
MEM: 0000000000000000 00000000000A0000 00 1
MEM: 00000000000A0000 0000000000060000 00 2
MEM: 0000000000100000 00000000790B8000 00 1
MEM: 00000000791B8000 0000000000500000 01 2
MEM: 00000000796B8000 0000000000090000 00 3
MEM: 0000000079748000 0000000000008000 00 4
MEM: 0000000079750000 0000000000500000 00 2
MEM: 0000000079C50000 00000000003B0000 00 2
MEM: 000000007A000000 0000000004000000 00 2
MEM: 000000007E000000 0000000000800000 00 2
MEM: 000000007E800000 0000000000800000 00 2
MEM: 000000007F000000 0000000000800000 00 2
MEM: 000000007F800000 0000000000800000 00 2
MEM: 00000000FED20000 0000000000060000 00 2
MEM: 00000000FF530000 0000000000AD0000 00 2
MEM: 0000000100000000 0000000380000000 00 1
MP Init (Wakeup)
SMRR Base: 0x7E800006  Mask: 0xFF800000
MP Init (Run)
Detected 8 CPU threads
 CPU  0 APIC ID: 0
 CPU  1 APIC ID: 4
 CPU  2 APIC ID: 2
 CPU  3 APIC ID: 6
 CPU  4 APIC ID: 1
 CPU  5 APIC ID: 3
 CPU  6 APIC ID: 5
 CPU  7 APIC ID: 7
PCI Enum
>Failed to set GFX framebuffer as WC
Call FspNotifyPhase(20) ... Success
ACPI NVS, LowPowerS0IdleConstraint(Micro-pep constraints) = 0x00000000614A86F7 
Update Cpu Nvs Done

>Tiger Lake RVP DDR4 debug log

============= Intel Slim Bootloader STAGE2 =============
PCH gen (0x5), series (0x1), stepping (0x11)
LPC DeviceId (0x438A)
Boot-time Pch Dts: 0x190
Update PayloadId to UEFI
Variable region: 0xFF60E000:0x2000
SaveNvsData  Length=0xFAA8
Matched with saved NVS data
Save MRC Training Data (0x45C96110 0x00FAA8) ... Already started
Silicon Init
PchH is detected
BIOS SPI region will be protected
IAS  SPI region will be protected
Mike PcdFramebufferInitEnabled :GraphicsConfigPtr=0
Select VBT ImageId 0x00000001
Mike PcdFramebufferInitEnabled :GraphicsConfigPtr=45A5701C
Load Silicon Cfg Data
Call FspSiliconInit ... 
Success
FspGfxHob->FrameBufferBase = 0xB0000000
Boot-time Cpu DTS: 0x10E
Graphics Info: 1280 x 1024 x 32 @ 0xB0000000
Found a gap in memory map: 4A000000 - 00000000
Found a gap in memory map: 4AC00000 - 00000000
MEM: 0000000000000000 00000000000A0000 00 1
MEM: 00000000000A0000 0000000000060000 00 2
MEM: 0000000000100000 00000000450B8000 00 1
MEM: 00000000451B8000 0000000000500000 01 2
MEM: 00000000456B8000 0000000000090000 00 3
MEM: 0000000045748000 0000000000008000 00 4
MEM: 0000000045750000 0000000000500000 00 2
MEM: 0000000045C50000 00000000003B0000 00 2
MEM: 0000000046000000 0000000004000000 00 2
MEM: 000000004A000000 0000000000B00000 00 2
MEM: 000000004AB00000 0000000000100000 00 2
MEM: 000000004AC00000 0000000000400000 00 2
MEM: 000000004B000000 0000000000800000 00 2
MEM: 000000004B800000 0000000000800000 00 2
MEM: 000000004C000000 0000000003C00000 00 2
MEM: 00000000FED20000 0000000000060000 00 2
MEM: 00000000FF530000 0000000000AD0000 00 2
MEM: 0000000100000000 00000007B0400000 00 1
MP Init (Wakeup)
SMRR Base: 0x4B000006  Mask: 0xFF800000
MP Init (Run)
Detected 16 CPU threads
 CPU  0 APIC ID: 0
 CPU  1 APIC ID: 8
 CPU  2 APIC ID: 4
 CPU  3 APIC ID: 12
 CPU  4 APIC ID: 2
 CPU  5 APIC ID: 6
 CPU  6 APIC ID: 10
 CPU  7 APIC ID: 14
 CPU  8 APIC ID: 1
 CPU  9 APIC ID: 3
 CPU 10 APIC ID: 5
 CPU 11 APIC ID: 7
 CPU 12 APIC ID: 9
 CPU 13 APIC ID: 11
 CPU 14 APIC ID: 13
 CPU 15 APIC ID: 15
PCI Enum
GFX framebuffer WC is enabled @ 0x90000000:0x10000000
Call FspNotifyPhase(20) ... Success
ACPI NVS, LowPowerS0IdleConstraint(Micro-pep constraints) = 0x0000000061EA06F7 
Update Cpu Nvs Done

randylintw, 2022-01-20:
Can you check the vbt.bin ? 
Do you use the same as RVP?
FSP update includes this file updating. 
https://github.com/intel/FSP/commit/04ad3cd76f87f44a8e632db60cecd82e881ca081

Advgcipc, 2022-01-20:
I had found the root case for this issue. Thanks your reply.
I used the different RVP for Up3 and H. 
The Up3 CRB used USB type c alt mode. The "mTglUDdr4RowDisplayDdiConfig" had to update.
Ours designed as the same UP3 but does not USB type c connector.
The FSP can not find DDI. The GFX will be init.
Anyway, thanks! Our board can boot up into Windows 10.

closed 2022-01-20 by Advgcipc