slimbootloader/slimbootloader #1467: [EHL] GpioDataConvert not working or not documented

bmilosevic, 2022-01-21:
I've successfully build and deployed slimbootloader on the EHL-CRB. I'm now customizing it for another target board and I need to adjust the configuration of some GPIO pads for the new board. 

If I'm correct, slimbootloader/Platform/ElkhartlakeBoardPkg/CfgData/CfgData_Gpio.yaml holds the config for all pads, in SBL format. 
the doc https://slimbootloader.github.io/tools/GpioTool.html specifies that slimbootloader/Platform/CommonBoardPkg/Tools/GpioDataConvert.py can be used to convert gpio config to/from different formats. I would like to convert yaml to csv, which is easier to interpret and modify, then to convert back the modified csv to a new yaml.

Now, the problem is that GpioDataConvert asks for 3 inputs: input data file, output format and a "Config file". I did not find anywhere what config file should be used... I've tried ifferent combinations of files to be passed but without luck.

Can you check if there is a bug and provide me with an example on how to convert gpio config for EHL?

Thanks!

mauricema, 2022-01-21:
Hi,
I just looked into this and found the GPIO template has changed a little bit for EHL.  As a result, some of the script needs to be adjusted.  Also as you noticed, it needs a platform specific GPIO configuration python file.  It needs to be created by the user.  For convenience, I added this file for EHL.   Please check the following PR:
https://github.com/slimbootloader/slimbootloader/pull/1468
Please let us know if it works for you.  After the conversion, the data might have minor difference.  It is because of the skipped GPIO init in the table.  It should be fine.

On the other side, if you just have very few GPIOs to be modified.  I would recommend you to use ConfigEditor to change the DLT file instead.  It will be much easier.  Please refer to https://slimbootloader.github.io/how-tos/configure_gpio.html.

bmilosevic, 2022-01-24:
Thank you, I've successfully tested converting back and forth from yaml and csv. 
I did use the Config Editor as well, but having lots of gpios to change and verify, a scripted solution will be handy.

mauricema, 2022-01-24:
Glad to know it worked for you.   Close it for now.

closed 2022-01-24 by mauricema