slimbootloader/slimbootloader #1469: QEMU error on Ubuntu 20.04 running in VirtualBox - KVM internal error. Suberror 1

Lewons, 2022-01-25:
Hi there, 
I'm having and issue executing QEMU on Ubuntu 20.04 which is running inside VirtualBox on a Windows 10 host. This is the output when I try to run it, it crashes right at the start:

```
$ /usr/bin/qemu-system-x86_64 -nodefaults -nographic -vga none -machine q35 -cpu host -smp 2,maxcpus=2,cores=2 -m 4G -accel kvm -pflash ./qemu_workspace/slim_bootloader.img -object rng-random,filename=/dev/random,id=rng0 -device virtio-rng-pci,rng=rng0 -drive file=./qemu_workspace/disk.img,format=qcow2,if=ide,id=disk0,file.locking=off -drive file=./qemu_workspace/disk.img,format=qcow2,if=virtio,file.locking=off  -netdev tap,id=t1,ifname=tap0,script=no,downscript=no -device virtio-net-pci,netdev=t1,id=nic1,mac=52:54:11:22:33:01,guest_csum=off  -netdev tap,id=t2,ifname=tap1,script=no,downscript=no -device virtio-net-pci,netdev=t2,id=nic2,mac=52:54:11:22:33:02,guest_csum=off -serial mon:stdio
WARNING: Image format was not specified for './qemu_workspace/slim_bootloader.img' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
KVM internal error. Suberror: 1
emulation failure
EAX=00000000 EBX=00000000 ECX=00000000 EDX=000a0652
ESI=00000000 EDI=00005042 EBP=00000000 ESP=00000000
EIP=0000ffb2 EFL=00000002 [-------] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0000 00000000 0000ffff 00009300
CS =f000 ffff0000 0000ffff 00009b00
SS =0000 00000000 0000ffff 00009300
DS =0000 00000000 0000ffff 00009300
FS =0000 00000000 0000ffff 00009300
GS =0000 00000000 0000ffff 00009300
LDT=0000 00000000 0000ffff 00008200
TR =0000 00000000 0000ffff 00008b00
GDT=     00000000 0000ffff
IDT=     00000000 0000ffff
CR0=60000010 CR2=00000000 CR3=00000000 CR4=00000000
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000000
Code=8b 00 8b 30 0f 7e c0 ff e6 bf 42 50 eb e9 bf 41 50 eb e4 fa <0f> 6e c0 eb e0 90 90 90 90 90 90 90 90 90 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

These are the relevant specs:
Guest system: Ubuntu 20.04 (kernel: 5.13.0-27-generic)
VirtualBox ver. 6.1.30, nested virtualization activated, Paravirtualization: default, 6 cores
Host system: Windows 10 20H2

I am really not sure what could be the issue here. Any help would be appreciated. cpuinfo for one core is as follows:

```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 165
model name	: Intel(R) Core(TM) i7-10850H CPU @ 2.70GHz
stepping	: 2
cpu MHz		: 2712.000
cache size	: 12288 KB
physical id	: 0
siblings	: 6
core id		: 0
cpu cores	: 6
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid tsc_known_freq pni pclmulqdq vmx ssse3 cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx rdrand hypervisor lahf_lm abm 3dnowprefetch invpcid_single tpr_shadow vnmi flexpriority vpid fsgsbase avx2 invpcid rdseed clflushopt md_clear flush_l1d arch_capabilities
vmx flags	: vnmi flexpriority tsc_offset vtpr vapic
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs itlb_multihit
bogomips	: 5424.00
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:
```

Thanks!

mauricema, 2022-01-26:
First of all, have  you tried on a native Ubuntu ?  Do you see issue ?   
In your case, it stopped pretty early at f000:ffb2.  So it is still in real mode.  Not sure what went wrong.

I did try similar command in ubuntu box,  it can boot with one SBL change as discussed in
https://github.com/slimbootloader/slimbootloader/issues/1302

Lewons, 2022-01-26:
I did not try it yet on native Ubuntu since I would need to set it up first. But I will try to set it up and test it out.
Still, it's necessary for me to work on virtualized Ubuntu since my host machine will always be Windows (and running the project on Windows is also not an option).

I already included the patch from #1302 and I still have this issue. I also tried removing the -smp flags and tweaking some of the parameters but without any success.

LittleHuba, 2022-01-26:
> First of all, have you tried on a native Ubuntu ? Do you see issue ?

I was able to try it on a machine with native Ubuntu 20.04. The issue does not appear there. It seems to be an issue that is connected to the virtualization.

Lewons, 2022-01-26:
After removing the "-cpu host" and "-accel kvm" flags, QEMU is starting again. But removing the acceleration is not feasible unfortunately, due to a significant drop in performance.

mauricema, 2022-01-26:
If so, I guess it will need to be debugged from virtualization side to see what was wrong.  From SBL side, we don't have any visibility on the KVM behavior.

mauricema, 2022-01-26:
I did some further debug on this issue.  I was able to reproduce this in a nested virtualization environment.  I have submitted a fix for this
https://github.com/slimbootloader/slimbootloader/pull/1471.
Could you please try it out to see if it works for you ?

Lewons, 2022-01-27:
The fix works! Thanks a lot for that :)

closed 2022-01-27 by Lewons

mauricema, 2022-01-27:
You are welcome!  
Glad to know it worked for you !  😃

commit e75725ef (2022-01-27): [QEMU] Fix QEMU boot issue in nested virtualization env

Is is reported that SBL QEMU target crashes when running in nested
virtualization env. For example, run QEMU with KVM flag inside
Ubuntu running on Windows Virtual Box.
After debug, it turned out somehow the KVM generated exception for
instruction "movd mm0, eax" at the very beginning. This was used to
save the CPU BIST info. If it is moved to be after switching to FLAT32
mode, then everything goes well.  This patch moved the BIST saving
into later stage in the boot in order to address the KVM crash issue.

This fixed #1469.