slimbootloader/slimbootloader #1502: memory conflicts on tigerlake kontron board

erik-kline, 2022-02-25:
On our Kontron tigerlake system In dmesg. Why is there a bar 6. This is a strange bar number. If I install additional cards in the systems. I get more of these kinds of error. With -realloc as a kernel parameter helps. I need to track down the source of these conflicts. They do **NOT** appear in uefi

[    3.611552] pci 0000:05:00.0: BAR 6: no space for [mem size 0x00100000 pref]
[    3.614224] pci 0000:05:00.0: BAR 6: failed to assign [mem size 0x00100000 pr
-[0000:00]-+-00.0  Intel Corporation 11th Gen Core Processor Host Bridge/DRAM Registers
           +-02.0  Intel Corporation TigerLake GT2 [Iris Xe Graphics]
           +-05.0  Intel Corporation Device 9a19
           +-07.0-[01]--
           +-07.1-[02]--
           +-07.2-[03]--
           +-07.3-[04]--
           +-08.0  Intel Corporation Device 9a11
           +-0a.0  Intel Corporation Device 9a0d
           +-0d.0  Intel Corporation Tiger Lake-LP Thunderbolt 4 USB Controller
           +-0d.1  Intel Corporation Device 9a15
           +-0d.2  Intel Corporation Tiger Lake-LP Thunderbolt 4 NHI #0
           +-0d.3  Intel Corporation Tiger Lake-LP Thunderbolt 4 NHI #1
           +-10.0  Intel Corporation Device a0d8
           +-10.5  Intel Corporation Device a0af
           +-14.0  Intel Corporation Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller
           +-14.1  Intel Corporation Device a0ee
           +-14.2  Intel Corporation Tiger Lake-LP Shared SRAM
           +-15.0  Intel Corporation Tiger Lake-LP Serial IO I2C Controller #0
           +-15.1  Intel Corporation Tiger Lake-LP Serial IO I2C Controller #1
           +-17.0  Intel Corporation Device a0d3
           +-1d.0-[05]----00.0  Intel Corporation Ethernet Controller I225-LM
           +-1e.0  Intel Corporation Tiger Lake-LP Serial IO UART Controller #0
           +-1e.3  Intel Corporation Tiger Lake-LP Serial IO SPI Controller #1
           +-1f.0  Intel Corporation Tiger Lake-LP LPC Controller
           +-1f.4  Intel Corporation Tiger Lake-LP SMBus Controller
           \-1f.5  Intel Corporation Tiger Lake-LP SPI Controller

     PCI(00,1D,00)
       PPB BAR0/1
       PPB APPERTURE
       BAR[2].TYP = 3
       BAR[2].OFF = 0x100
       BAR[2].BAS = 0000000082400000  BAR[2].LEN = 0000000000200000  BAR[2].ALN
     PCI(05,00,00)
       BAR[0].TYP = 3
       BAR[0].OFF = 0x10
       BAR[0].BAS = 0000000082400000  BAR[0].LEN = 0000000000100000  BAR[0].ALN
       BAR[3].TYP = 3
       BAR[3].OFF = 0x1C
       BAR[3].BAS = 0000000082500000  BAR[3].LEN = 0000000000004000  BAR[3].ALN

**Dump kernel startup below**

Starting Kernel ...

Switch to LongMode and jump to 64-bit kernel entrypoint ...
[    0.000000] Linux version 4.18.0-240.el8.x86_64 (mockbuild@kbuilder.bsys.cen0
[    0.000000] Command line: root=/dev/mapper/cl-root loglevel=7 console=ttyS0,1
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point regi
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x020: 'AVX-512 opmask'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x040: 'AVX-512 Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x080: 'AVX-512 ZMM_Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User re
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[5]:  832, xstate_sizes[5]:   64
[    0.000000] x86/fpu: xstate_offset[6]:  896, xstate_sizes[6]:  512
[    0.000000] x86/fpu: xstate_offset[7]: 1408, xstate_sizes[7]: 1024
[    0.000000] x86/fpu: xstate_offset[9]: 2432, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x2e7, context size is 2440 byte
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000777b7fff] usable
[... 825 lines cut ...]
[   66.557441] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[   66.563897] iTCO_wdt: Found a Intel PCH TCO device (Version=6, TCOBASE=0x0400
[   66.572133] iTCO_wdt: initialized. heartbeat=30 sec (nowayout=0)
[   66.598108] pxa2xx-spi pxa2xx-spi.4: missing platform data
[   67.090659] XFS (sda2): Mounting V5 Filesystem
[   67.287143] XFS (dm-1): Mounting V5 Filesystem
[   67.452111] XFS (sda2): Ending clean mount
[   67.899213] XFS (dm-1): Ending clean mount
[   68.620824] RPC: Registered named UNIX socket transport module.
[   68.627458] RPC: Registered udp transport module.
[   68.632727] RPC: Registered tcp transport module.
[   68.638024] RPC: Registered tcp NFSv4.1 backchannel transport module.
[   84.672338] IPv6: ADDRCONF(NETDEV_UP): enp5s0: link is not ready
[   84.784574] IPv6: ADDRCONF(NETDEV_UP): enp5s0: link is not ready
[   86.898706] IPv6: ADDRCONF(NETDEV_UP): enp5s0: link is not ready
[   88.069167] igc 0000:05:00.0 enp5s0: igc: enp5s0 NIC Link is Up 1000 Mbps Ful
[   88.184249] IPv6: ADDRCONF(NETDEV_CHANGE): enp5s0: link becomes ready
[   91.719132] bridge: filtering via arp/ip/ip6tables is no longer available by
[   91.766927] tun: Universal TUN/TAP device driver, 1.6
[   91.773500] virbr0: port 1(virbr0-nic) entered blocking state
[   91.779959] virbr0: port 1(virbr0-nic) entered disabled state
[   91.786424] device virbr0-nic entered promiscuous mode
[   92.363732] virbr0: port 1(virbr0-nic) entered blocking state
[   92.370172] virbr0: port 1(virbr0-nic) entered listening state
[   92.501589] virbr0: port 1(virbr0-nic) entered disabled state

erik-kline, 2022-02-25:
uefi dmesg

[erik@localhost ~]$ dmesg
[    0.000000] microcode: microcode updated early to revision 0x88, date = 2021-03-31
[    0.000000] Linux version 4.18.0-305.19.1.el8_4.x86_64 (mockbuild@kbuilder.bsys.centos.org) (gcc version 8.4.1 20200928 (Red Hat 8.4.1-1) (GCC)) #1 SMP Wed Sep 15 15:39:39 UTC 2021
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/vmlinuz-4.18.0-305.19.1.el8_4.x86_64 root=/dev/mapper/cl-root ro crashkernel=auto resume=/dev/mapper/cl-swap rd.lvm.lv=cl/root rd.lvm.lv=cl/swap rhgb quiet
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x020: 'AVX-512 opmask'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x040: 'AVX-512 Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x080: 'AVX-512 ZMM_Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[5]:  832, xstate_sizes[5]:   64
[    0.000000] x86/fpu: xstate_offset[6]:  896, xstate_sizes[6]:  512
[    0.000000] x86/fpu: xstate_offset[7]: 1408, xstate_sizes[7]: 1024
[    0.000000] x86/fpu: xstate_offset[9]: 2432, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x2e7, context size is 2440 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003d630fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003d631000-0x000000003d631fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003d632000-0x000000003f19ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003f1a0000-0x000000003f1a0fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003f1a1000-0x00000000702d2fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000702d3000-0x00000000733d2fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000733d3000-0x000000007349efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007349f000-0x0000000073559fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007355a000-0x0000000073efefff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000073eff000-0x0000000073efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000073f00000-0x0000000079ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007ab00000-0x000000007abfffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x00000000803fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000047fbfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x69c1b018-0x69c4c657] usable ==> usable
[    0.000000] e820: update [mem 0x69c1b018-0x69c4c657] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000003d630fff] usable
[    0.000000] reserve setup_data: [mem 0x000000003d631000-0x000000003d631fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000003d632000-0x000000003f19ffff] usable
[    0.000000] reserve setup_data: [mem 0x000000003f1a0000-0x000000003f1a0fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000003f1a1000-0x0000000069c1b017] usable
[    0.000000] reserve setup_data: [mem 0x0000000069c1b018-0x0000000069c4c657] usable
[    0.000000] reserve setup_data: [mem 0x0000000069c4c658-0x00000000702d2fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000702d3000-0x00000000733d2fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000733d3000-0x000000007349efff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007349f000-0x0000000073559fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007355a000-0x0000000073efefff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000073eff000-0x0000000073efffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000073f00000-0x0000000079ffffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007ab00000-0x000000007abfffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007b000000-0x00000000803fffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[... cut ...]
[   49.837737] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   49.929078] tun: Universal TUN/TAP device driver, 1.6
[   49.929980] virbr0: port 1(virbr0-nic) entered blocking state
[   49.929981] virbr0: port 1(virbr0-nic) entered disabled state
[   49.930028] device virbr0-nic entered promiscuous mode
[   51.254231] virbr0: port 1(virbr0-nic) entered blocking state
[   51.254233] virbr0: port 1(virbr0-nic) entered listening state
[   51.384611] virbr0: port 1(virbr0-nic) entered disabled state
[erik@localhost ~]$

erik-kline, 2022-02-25:
slimboot output

============= Intel Slim Bootloader STAGE1A =============
SBID: SBL_TGL
ISVN: 001
IVER: 001.000.001.004.00012
SVER: D880AC8760185051
FDBG: BLD(D IA32) FSP(D)
FSPV: ID(TGLI-FSP) REV(0A005D41)
Loader global data @ 0xFEF3FD10
Run  STAGE1A @ 0xFFFEA000
Load STAGE1B @ 0xFFD80000

============= Intel Slim Bootloader STAGE1B =============
[Boot Guard] AcmStatus : 0x00000000
[Boot Guard] BootStatus: 0x00000000
[Boot Guard] Boot Guard Failed or is Disabled!
SpiInstance = FEF4C330
PchSpiBase at 0x00001F05
ScSpiBar0 at 0xF0000000
Loading Component KEYH:_HS_
Registering container KEYH
Append public key hash into store: Success
Load EXT CFG Data @ 0xFEF4CB54:0x0F50 ... Success
Processor does not support Boot Guard.
Boot Guard Support status: 0
GetPlatformDataPtr is copied 0xFEF4CB34
BOOT: BP0
MODE: 0
BoardID: 0x01
PlatformName: TGLU_DDR
Memory Init
FSPM CfgData assignment
Load Memory Cfg Data
SkipMbpHob = 0x0
Load Graphics Cfg Data
board Id = 1 .....
Call FspMemoryInit ... Success
Loader global data @ 0x78C0FD50
Memory Tolum @ 0x7FC00000
Memory Touum @ 0x480400000
Memory Tom   @ 0x400000000
Switch to memory stack @ 0x78C4FF00
Stage1 stack: 0x40000 (0x26414 used)
Stage1 heap: 0x10000 (0x4360 used, 0x4360 max used)
Call FspTempRamExit ... Success
Memory FSP @ 0x78C50000
Memory TOP @ 0x78750000
Loading Component FLMP:SG02
Loaded STAGE2 @ 0x78A03000

============= Intel Slim Bootloader STAGE2 =============
PCH gen (0x5), series (0x2), stepping (0x10)
LPC DeviceId (0xA082)
Boot-time Pch Dts: 0x10E
Update PayloadId to OS Loader
Variable region: 0xFF7AC000:0x2000
SaveNvsData  Length=0xFAA8
Matched with saved NVS data
Save MRC Training Data (0x78C96110 0x00FAA8) ... Already started
Silicon Init
PchLp is detected
BIOS SPI region will be protected
IAS  SPI region will be protected
Load Silicon Cfg Data
Call FspSiliconInit ...
Success
FspGfxHob->FrameBufferBase = 0xB0000000
Boot-time Cpu DTS: 0x122
Graphics Info: 1920 x 1080 x 32 @ 0xB0000000
[... 606 lines cut ...]
 31F0 |    3102 ms |       0 ms | End of stage2
 4000 |    3138 ms |      36 ms | Payload entry point
 4010 |    3151 ms |      13 ms | OS loader main entry
 4020 |    5166 ms |    2015 ms | Shell exit
 4040 |    5221 ms |      55 ms | Load image entry
 4040 |    6263 ms |    1042 ms | Load image entry
 4040 |    7305 ms |    1042 ms | Load image entry
 4050 |    7340 ms |      35 ms | Boot device init
 4055 |    7340 ms |       0 ms | Boot device tuning
 4060 |    7461 ms |     121 ms | Parse partition info
 4070 |    7490 ms |      29 ms | Load boot images
 4040 |    7513 ms |      23 ms | Load image entry
 4040 |    7558 ms |      45 ms | Load image entry
 4050 |    7569 ms |      11 ms | Boot device init
 4055 |    7569 ms |       0 ms | Boot device tuning
 4070 |    7569 ms |       0 ms | Load boot images
 4040 |    7593 ms |      24 ms | Load image entry
 4050 |    7604 ms |      11 ms | Boot device init
 4055 |    7604 ms |       0 ms | Boot device tuning
 4060 |    7641 ms |      37 ms | Parse partition info
 4070 |    8515 ms |     874 ms | Load boot images
 40B0 |    8528 ms |      13 ms | Process ELF/MultiBoot
 40F0 |    8639 ms |     111 ms | FSP ReadyToBoot/EndOfFirmware notify
 4100 |    8639 ms |       0 ms | TPM IndicateReadyToBoot
------+------------+------------+----------------------------------

mauricema, 2022-02-25:
BAR6 listed above is PCI option ROM BAR.  SBL does not allocate any option ROM bar at this moment since SBL does not deal with any option ROM at all.
You can find some detailed discussion in kernel at:
https://lore.kernel.org/lkml/CAL-B5D3FgF+o7BcLBe7O_gcxy2jf3w43bXeDJuRH2VS2ES0A7Q@mail.gmail.com/T/

We will have some discussion to see if we need to add PCI ROM BAR resource allocation in SBL.

erik-kline, 2022-02-26:
There is a lot of stuff to sort through with this and it is complicated.

It seems that there is a bit that tells the linux kernel if a option rom is on the pci device. if this bit is zero. the linux kernel still attempts to allocate memory for a option rom. this part seems to be a bug in the linux kernel. strange that is has not been fixed. this appears to be what is happening here.

The kernel option suggested only works if the enable bit has been set to zero. This suggests the kernel people are aware of the problem and have not fixed it for some reason. That is odd. There seems to be no reason possible for a disable bit to be ignored.

It seems that other pci devices have done something to not have this error NOT triggered. most likely the memory ranges are filed in  better. What is it that these other devices have done. Since this a intel chip does this indicate a problem with the 225 device. if the 225 device actually has option rom. is this actually needed. the ethernet does work on the box. but is there some functionality that does not work.

What is the solution to this problem?

I have other third party card that exhibit the same behavior I took these out to the system to simplify the problem. I will have to get them to work also.

mauricema, 2022-02-27:
Hi,

I believe this error will only be triggered when the upstream PCI bridge has no additional resource to be allocated to the ROM bar.  In many cases due to alignments PCI bridge will have extra spaces beyond what is actually required by all the PCI devices underneath.  These extra space can be used by kernel to assign the ROM bar.  But if this extra space is not big enough,  kernel will complain.  To fully resolve this,  I guess we will need to allocate PCI ROM bar resource in SBL to make Linux happy.  

I have submitted a PR to enable PCI ROM bar resource allocation for SBL.
You need to add "self._PCI_ENUM_FLAG_ALLOC_ROM_BAR = 1" in your BoardConfig.py to enable this feature.  
Please try the PR  https://github.com/slimbootloader/slimbootloader/pull/1507,  and let us know if it can address the PCI ROM bar resource issue.

Regards,
Maurice

erik-kline, 2022-02-28:
this change seems to address  the issue with BAR6. 
unfortunately when I put a pci switch card into the board. I still get errors as shown below. If I use the kernal parameter. -realloc these error still occur but the kernel gives it another shot on goal and the system works. I do need to resolve the kind of  issues

 pci 0000:02:00.0: BAR 13: no space for [io  size 0x1000]
[    2.374151] pci 0000:02:00.0: BAR 13: failed to assign [io  size 0x1000]
[    2.376683] pci 0000:02:01.0: BAR 13: no space for [io  size 0x1000]
[    2.379087] pci 0000:02:01.0: BAR 13: failed to assign [io  size 0x1000]
[    2.381619] pci 0000:02:02.0: BAR 13: no space for [io  size 0x1000]
[    2.384023] pci 0000:02:02.0: BAR 13: failed to assign [io  size 0x1000]
[    2.386555] pci 0000:02:03.0: BAR 13: no space for [io  size 0x1000]
[    2.388960] pci 0000:02:03.0: BAR 13: failed to assign [io  size 0x1000]
[    2.391501] pci 0000:02:04.0: BAR 13: no space for [io  size 0x1000]
[    2.393903] pci 0000:02:04.0: BAR 13: failed to assign [io  size 0x1000]
[    2.396434] pci 0000:02:05.0: BAR 13: no space for [io  size 0x1000]

erik-kline, 2022-02-28:
kernel messages below

Starting Kernel ...

Switch to LongMode and jump to 64-bit kernel entrypoint ...
[    0.000000] Linux version 4.18.0-240.el8.x86_64 (mockbuild@kbuilder.bsys.centos.org) (gcc version 8.3.1 20191121 (Red Hat 8.3.1-5) (GCC)) #1 SMP Fri Sep 25 19:40
[    0.000000] Command line: root=/dev/mapper/cl-root loglevel=7 console=ttyS0,115200n8 acpi_rsdp=0x786B8000
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x020: 'AVX-512 opmask'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x040: 'AVX-512 Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x080: 'AVX-512 ZMM_Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[5]:  832, xstate_sizes[5]:   64
[    0.000000] x86/fpu: xstate_offset[6]:  896, xstate_sizes[6]:  512
[    0.000000] x86/fpu: xstate_offset[7]: 1408, xstate_sizes[7]: 1024
[    0.000000] x86/fpu: xstate_offset[9]: 2432, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x2e7, context size is 2440 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000777b7fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000777b8000-0x00000000786b7fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000786b8000-0x0000000078747fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000078748000-0x000000007874ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000078750000-0x000000007fbfffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff530000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000004803fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.3 present.
[    0.000000] DMI: Intel Corporation TigerLake Client Platform/TigerLake U DDR4 SODIMM RVP, BIOS SB_TGL.001.001.000.001.004.00007.D-31AA4F7DC3035C
[    0.000000] tsc: Detected 2800.000 MHz processor
[    0.000000] tsc: Detected 2803.200 MHz TSC
[    0.000000] last_pfn = 0x480400 max_arch_pfn = 0x400000000
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] last_pfn = 0x777b8 max_arch_pfn = 0x400000000
[    0.000000] Using GB pages for direct mapping
[    0.000000] RAMDISK: [mem 0x73562000-0x76cc9fff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000786B8000 000024 (v02 OEMID )
[    0.000000] ACPI: XSDT 0x00000000786B80E0 000104 (v01 OEMID  OEMTABLE 00000005 CREA 0100000D)
[    0.000000] ACPI: BDAT 0x00000000786B8210 000030 (v01 OEMID  OEMTABLE 00000005 CREA 0100000D)
[    0.000000] ACPI: DMAR 0x00000000786B8240 000100 (v01 INTEL  EDK2     00000002      01000013)
[    0.000000] ACPI: FACP 0x00000000786B8380 000114 (v06 OEMID  OEMTABLE 00000005 CREA 0100000D)
[    0.000000] ACPI: DSDT 0x00000000786BE8B0 021547 (v02 INTEL  SKL      00000000 INTL 20190509)
[    0.000000] ACPI: FACS 0x00000000786B8340 000040
[    0.000000] ACPI: FACS 0x00000000786B8340 000040
[    0.000000] ACPI: FWST 0x00000000786B84A0 0000AC (v01 INTEL  FWUPDSTS 00001000 INTL 20090903)
[    0.000000] ACPI: HPET 0x00000000786B8550 000038 (v01                 00000001      00000000)
[    0.000000] ACPI: LPIT 0x00000000786B8590 0000CC (v01                 00000000      00000000)
[    0.000000] ACPI: APIC 0x00000000786B8660 0000EC (v04                 00000001      00000000)
[    0.000000] ACPI: MCFG 0x00000000786B8750 00003C (v01                 00000001      00000000)
[    0.000000] ACPI: FPDT 0x00000000786B8790 000044 (v01 INTEL  OEMTABLE 00000005 CREA 0100000D)
[    0.000000] ACPI: TPM2 0x00000000786B8830 00004C (v04 INTEL  OEMTABLE 00000005 CREA 0100000D)
[    0.000000] ACPI: PSDS 0x00000000786B8880 000052 (v01 INTEL  EDK2     00000005 INTL 0100000D)
[    0.000000] ACPI: SSDT 0x00000000786B88E0 00048A (v02 PmRef  ApCst    00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786B8D70 00048A (v02 PmRef  ApHwp    00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786B9200 0008E7 (v02 PmRef  ApIst    00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786B9AF0 0004D4 (v02 PmRef  ApPsd    00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786B9FD0 000B37 (v02 PmRef  ApTst    00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786BAB10 000394 (v02 PmRef  Cpu0Cst  00003001 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786BAEB0 00013C (v02 PmRef  Cpu0Hwp  00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786BAFF0 000815 (v02 PmRef  Cpu0Ist  00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786BB810 0001CB (v02 PmRef  Cpu0Psd  00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786BB9E0 0003E0 (v02 PmRef  Cpu0Tst  00003000 INTL 20190509)
[    0.000000] ACPI: SSDT 0x00000000786BBDC0 002481 (v02 CpuRef CpuSsdt  00003000 INTL 20190509)
[... cut ...]
[   96.988274] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   97.023847] tun: Universal TUN/TAP device driver, 1.6
[   97.032976] virbr0: port 1(virbr0-nic) entered blocking state
[   97.039414] virbr0: port 1(virbr0-nic) entered disabled state
[   97.045900] device virbr0-nic entered promiscuous mode
[   97.798988] virbr0: port 1(virbr0-nic) entered blocking state
[   97.805437] virbr0: port 1(virbr0-nic) entered listening state
[   97.903155] virbr0: port 1(virbr0-nic) entered disabled state

CentOS Linux 8

mauricema, 2022-02-28:
I guess you might have two many devices requesting I/O spaces. The whole system can only have up to 64KB I/O space, and we are running out of space.

Also if you check dmesg "pci 0000:00:06.0: BAR 13: no space for [io size 0x11000]", it indicates your PCI card might have compatibility issue since IO BAR size should not be greater than 64KB.

erik-kline, 2022-03-04:
the manufacture of the devices claims 
"Your root complex is trying to set up IO space which our switch has nothing to with us."
how would slimboot affect the root complex configuring io space

mauricema, 2022-03-04:
SBL PCI enumeration is aligned with EDK2 PCI bus enumeration.  The IO space required by the root complex is the sum of the IO space required by the devices underneath.  So the devices behind the root port will definitely have impacts on the system IO space resource.

Do you see the same resource issue with UEFI BIOS ? If so I would suggest you follow up with BIOS vendor and device vendor to get it work in UEFI BIOS first.  Once that works, then we can see how to make it work in SBL if there is any issue.   If not,  please attach both BIOS boot log and SBL boot log for our analysis.  BTW, it might be better to attach the log file since the logs are pretty long.

erik-kline, 2022-03-15:
it has taken a while but apparently both uefi and slimboot are writing bit 0 of the command register of the pci bridge that is causing the problems. per the specification this indicates IO is enabled on the bridge. It also appears that this bit is supposed to be 0 and read only for devices that do not support IO space.  This is a non-compliance of the hardware device.

can I NOT set this bit to 1 in slimboot to work around the non-compliance of this piece of hardware?

gdong1, 2022-03-23:
I think you could do some work around in SBL for a specific device (e.g. not enable IO space if there is no IO required, or give a hardcoded IO spaced size instead of required IO space size by the device, etc.) based on the PCI device actual requirement.

randylintw, 2022-06-30:
Hi, is this issue still valid? Can we close it?

closed 2022-10-18 by rprangar