slimbootloader/slimbootloader #1503: disable multimedia controler

erik-kline, 2022-02-25:
In slimboot on a kontron tiger lake board. lspci shows
00:05.0 Multimedia controller: Intel Corporation Device 9a19 (rev 01)

This device does not show up in uefi

What is this multimedia controller. And how do I disable it. If I disable graphics. In the configuration file the video disappears from lspci 
- InternalGfx  :
        name         : Internal Graphics
        type         : Combo
        option       : $EN_DIS
        help         : >
                       Enable/disable internal graphics.
        length       : 0x01
        value        : 0x01

mauricema, 2022-02-25:
This is Image Processing Unit (IPU).  And it has different from the internal GFX.
I don't believe TGL FSP provides option to disable IPU, so currently SBL has no way to disable it.

erik-kline, 2022-02-26:
ok. it something that is not needed but it taking up space.

mauricema, 2022-02-27:
Can we close this ?

erik-kline, 2022-03-02:
yes we can close this. sorry about the delay

mauricema, 2022-03-02:
Thanks,  no problem!

closed 2022-03-02 by mauricema