slimbootloader/slimbootloader #1515: There is a desire the limit the power of the tiger lake processor to PL1 level.

erik-kline, 2022-03-02:
There is a desire the limit the power of the tiger lake processor to PL1 level. I see PsysPowerLimit1 in the FSP. In alder lake I see a configure option for this. But I do not see this for tiger lake.
Apparently, Tiger Lake has four different power settings that are controlled in the BIOS – PL1, PL2, PL3, PL4. we desire to not exceed the 15W TDP power limit. So, PL2 should be set to PL1 = 15W. I’m assuming that PL3, PL4 are also set to 15W.
How would one go about doing this.

gdong1, 2022-03-03:
There are so many FSP UPDs and TGL only map some of the import UPDs into SBL configuration data. yes currently TGL didn't provide these power settings. Could you please have a try to directly update these FSP UPDs in UpdateFspConfig () in file https://github.com/slimbootloader/slimbootloader/blob/master/Platform/TigerlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c?

erik-kline, 2022-03-07:
I understand that there are more setting in the FSP than  in the configuration data.
I am trying to understand what PsysPowerLimit1 , PsysPowerLimit1Time: and PsysPowerLimit1Power work. the FSP documentation is confusing.
I want to limit the processor to 15W all the time and never above this. it should be a simple case.

gdong1, 2022-03-09:
yes. Some of the FSP UPD descriptions are not very clear. I got some input from FSP folks and hope it could help.
-	PowerLimit:
o	CPU package level power limit
	PL1 = Long duration level
	PL2 = Short duration level
	PL3 = PL2 + ~2W (might be difference depends on CPU Gen and Sku)
	PL4 = Peak power limit in very short time (might only 10ms) 

-	Psys Power Limit:
o	Platform Level power limit

-	Custom PowerLimit:
o	Custom should be for cTDP CPU part !!
	Custom1 = override cTDP Nominal
	Custom2 = override cTDP Level 1 (Down)
	Custom3 = override cTDP Level 2 (UP)

so if you want to know “how to configure TGL FSP UPDs to limit CPU to 15W TDP”?
-	To limit CPU should be CPU package level, not Platform level (PSys)
-	Expected the request should be long duration power limit (PL1)
-	So we can just simply update new value of PowerLimit1 (for non-cTDP) to be 15000 (15.0W) which should be below or equal to Fused TDP !
-	If the CPU support cTDP, then can simply change to “cTDP down” or update “Custom1 PowerLimit (default is Nominal) = 15W”

erik-kline, 2022-03-15:
I have been distracted on other issues.
how do I know if the device support cTDP. it is a tigerlake processor.

it seems that i would set PowerLimit1 = 15000;
OR
 PowerLimit1 = 1; // 1 is TDP down

are there more settings in play here is it just the one that I have to deal with for this simple case

erik-kline, 2022-03-16:
I added 

 FspsConfig->PowerLimit1 = 200; 
 FspsConfig->PowerLimit2 = 0; // this should disable this
  FspsConfig->PowerLimit2Power=200;
 FspsConfig->PowerLimit3 = 200; 
 FspsConfig->PowerLimit4 = 200;
 FspsConfig->Custom1PowerLimit1 = 200;
 FspsConfig->Custom1PowerLimit2 = 200; 
 FspsConfig->Custom2PowerLimit1 = 200;
 FspsConfig->Custom2PowerLimit2 = 200; 
 FspsConfig->Custom3PowerLimit1 = 200; 
 FspsConfig->Custom3PowerLimit2 = 200; 

This generates a maximum power as shown in turbstat --show power of 15.14 watts

It is unclear why 200 or if there are other settings that can affect this.

gdong1, 2022-03-17:
On how to check if CPU supports cTDP, you could check the MSR 0xCE[34:33] (you could read MSR value from SBL Osloader shell), If the value is 0 means doesn’t support cTDP.
Sorry, for the FSP UPD PowerLimit value setting, need correct as below.
-	PowerLimit1 = 15000 -> should be 120 (= 15W * 8)
o	15000 is used for BIOS setup, not for UPD

randylintw, 2022-06-30:
Hi, is this issue still valid? Can we close it?

closed 2022-10-18 by rprangar