slimbootloader/slimbootloader #1556: [EHL] Unable to see both PSE-GbE for PXE boot

BCilEng, 2022-04-22:
Dear slimbootloader community,
I am  trying to  implement the PXE boot on a custom board with EHL. I built the UEFI payload to include the pxe undi driver released by intel as indicated in the slimbootloader documentation.
On the custom board I am  using the two PSE-GbE, the ethernet PHYs are supported, SGMII interface.
My problem is that I cannot see the PSE-GbE0 as PXE-boot option, even though both the hardware and the settings on the slimbootloader are identical to the ones for the PSE-GbE1 which is shown at boot manager. Both the GbEs works once that the OS starts. I suspect I am missing something on the slimbootloader side, since with the same UEFI payload I am able to see all the GbEs when moving to the reference board CRB.
Do you have any hint on where to start investigate? 
Thanks in advance,
Barbara

mauricema, 2022-04-28:
Here are things that maybe you can check:
- Ensure PciIo protocol is installed on PSE-GbE0 device
- Ensure PCI resource bars are allocated properly for PSE-GbE0 device
- Ensure you can access the PSE-GbE0 MMIO using memory dump after enabling the PCI memory decoding in CMD register.
- If possible, develop some test code to verify you have access to the SGMII phy registers for PSE-GbE0.
- Ensure driver binding occurs for PSE-GbE0 device and check the supported()/start() status.  If it returns error, then I guess you will have to get support for the UNDI to help you understand what caused the binding failure.

Also maybe you can try to disable PSE-GbE1 and only leave PSE-GbE0 on to see if it changes anything.

koktong-ong, 2022-05-05:
Hi Barbara,

Possible to check the difference of the Gbe layout in terms of the schematic design?
The only different is on the different board and got different outcome.

mauricema, 2022-05-05:
BTW, this has been discussed in the mailing list already. 
https://groups.io/g/slimbootloader/topic/90749046#1153

randylintw, 2022-05-06:
Hi Barbara, 
I  had EHL CRB and porting the network portion and using the [IntelgbeUndiDxe] driver, I can see three network cards in UEFI shell: ifconfig -l, also can see three PXE option in boot manager.

randylintw, 2022-06-30:
Hi, is this issue still valid? Can we close it?

BCilEng, 2022-08-24:
Hello,
Thank you for the support. The difference in behavior was given by the PHYAD that was different between the two ethernet controllers. Intel confirmed that the version of the IntelgbeUndiDxe driver I was using was coded to work only with PHYAD = 1. Changing the HW configuration of the controllers to match such requirement solved the issue. 
I will close the issue, thank you
Barbara

closed 2022-08-24 by BCilEng