slimbootloader/slimbootloader #1561: I2C Signal not propigating on TGL-UP3 Reference Board

evanblough, 2022-04-27:
I have the latest slimbootloader from master programmed onto a TGL-UP3 Reference board. The BIOS passes all stages and boots into CentOS 8.4. I have i2c-tools installed on the OS image. Using i2c-detect -l shows 3 i2c busses and one smbus. I have an oscilloscope plugged into the i2c-2 SDA/SCL pin connections labeled  on the silkscreen. On an earlier version of SBL I was seeing dmesg errors being generated about these i2c busses. After updating the dmesg errors went away. When i run the command i2c-detect 2 to scan for devices on the second i2c bus, I see no change in voltage level on my oscilloscope to indicate that the i2c bus is being scanned. I have been able to pass i2c on a TGL processor in a Kontron board, but the Intel supplied TGL-UP3 board doesn't pass signal. Is there additional configuration to be made in SBL to enable these pins, or are there hardware reworks that need to be applied to my board to pass signal ?

gdong1, 2022-05-03:
TGL-U PCH LPSS includes 6 I2C ports. Please check the TGL UP3 reference board documents on their connections (e.g., By default I2C1 used for Touch Panel 1, rework is required for 1x3 SV header). 
There are several FSP UPDs related with I2C. The most important one is SerialIoI2cMode. I think SBL just used the FSP default values. You could update its default value in Stage2BoardInitLib.c if needed.

/** Offset 0x018C - I2Cn Device Mode
  Selects I2c operation mode. N represents controller index: I2c0, I2c1, ... Available
  modes: 0:SerialIoI2cDisabled, 1:SerialIoI2cPci, 2:SerialIoI2cHidden
**/
  UINT8                       SerialIoI2cMode[8];

randylintw, 2022-06-30:
Hi, is this issue still valid? Can we close it?

closed 2022-10-18 by rprangar