slimbootloader/slimbootloader #1593: tgl fwupdate fail

yongyumijiebengji, 2022-05-31:
Dear slimbootloader community,

I'm developing on tiger lake platform and I have a problem when I use "fwupdate" command. When I type "fwupdate" under the shell, the system resets and starts the upgrade operation, and then the system hangs. The log is in the attachment.
I think there is something wrong with my FwuImage.bin that causes the firmware in the flash to be corrupted.But I don't know what is the cause of the problem? Do you have any ideas?

I use this script to compile:
	python BuildLoader.py build tgl -a x64
	
I use this script to generate iwfi file:
	python Platform\TigerlakeBoardPkg\Script\StitchIfwi.py -b fvme -s Outputs\tgl\Stitch_Components.zip -c Platform\TigerlakeBoardPkg\Script\StitchIfwiConfig_tglu.py -w C:\Stitching -p tglu_b0
	  
I use this script to generate FwuImage.bin:
	python ./BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS Outputs/tgl/SlimBootloader.bin -k ../SblKeys/FirmwareUpdateTestKey_Priv_RSA3072.pem -o FwuImage.bin

Thanks.

[fwupdate log.txt](https://github.com/slimbootloader/slimbootloader/files/8803821/fwupdate.log.txt)

yongyumijiebengji, 2022-06-01:
I did not stitch.Every time I use the StitchIfwi.py script to generate a new iwfi file.
Yes, power on the system after updated and it goes to off soon.
How can I fix it?

r0tw, 2022-06-01:
The Outputs/tgl/SlimBootloader.bin isn't signed by oem key, If you want to use fvme profile, it is required to use the signed SBL image.  That file you can use the fit tool to decompress the stitched IFWI binary and get the **BIOSRegion.bin** as **-P BIOS input file.**

yongyumijiebengji, 2022-06-01:
Thanks for your suggestion, it works fine.

closed 2022-06-01 by yongyumijiebengji