slimbootloader/slimbootloader #1609: Two PCIe slots not functioning on TGL RVP

tatokis, 2022-06-28:
With SBL, Linux does not recognise x4 cards installed in the x4 PCIe slot (J7B1), nor PCIe M.2 modules installed in the WWAN slot (J7D2) of a TGL DDR4 RVP.

With the standard BIOS/UEFI firmware, the PCIe x4 slot works out of the box, and the WWAN slot requires enabling in the firmware settings. I am unable to locate any setting that affects these two slots specifically in SBL.

If a x1 card is installed in the x4 slot, it is recognised and functions correctly. Specifically, the card that doesn't function under SBL is an ASUS [XG-C100C](https://www.asus.com/Networking-IoT-Servers/Wired-Networking/All-series/XG-C100C/).

Any ideas as to what might be happening? I assume at least the x4 slot should function by default without any additional configuration.
I verified that the correct IFWI image for this board has been stitched, and it is configured correctly.

The cards don't show up in lspci at all, and there aren't any relevant messages in dmesg.

randylintw, 2022-07-20:
Can you try to disable PM function on TGL boardConfig.py ? 
Platform/TigerlakeBoardPkg/BoardConfig.py
-->self.ENABLE_PCIE_PM       = 1  ->  change to 0 ?

tatokis, 2022-07-25:
@randylintw Thank you for the suggestion. I disabled PCIe PM, however it didn't seem to make much of a difference.

After some more investigation, it seems that it's random which cards are detected on every boot. In general PCH PCIe seems less reliable than when using the original firmware. After a while, we did get the cards recognised on all slots except WWAN.
Is there anything else that can be changed to improve PCH PCIe stability under SBL?

As for the WWAN port, in the original firmware we have to set the WWAN Device setting to 5G-M80. I am unable to find the equivalent setting in slim bootloader. How could that setting be changed?

randylintw, 2022-07-26:
Hi, 
1. On IFWI BIOS, is it work normal? 
2. If BIOS work well and you have debug version, maybe can compare the log to see what's the different on FSP parameter.

tatokis, 2022-10-28:
@randylintw Apologies for the super late reply. I still haven't had time to look into this, but yes the normal BIOS works fine. I believe I have acquired a debug version and I'll compare the FSP parameters when I get the chance, however more important things have come up so it'll be a while. Thanks!

rprangar, 2023-01-10:
Closing it as the issue is inactive.

closed 2023-01-10 by rprangar

tatokis, 2023-01-10:
For what it's worth, the issues were not resolved last time I checked, but I haven't had the resources to look into it. I'll update this when I have more information and if any of the issues were resolved.