slimbootloader/slimbootloader #1703: Slim Bootloader failed to boot on UP2 6000

MiaoMKung, 2022-09-30:
Hello, 

I am using the slim bootloader on the board of UP2 6000. can not boot.

it can be boot by the up2 6000 factory bin file of  UNEHAM20.bin.

but failed by follow the instruction of :
https://slimbootloader.github.io/supported-hardware/up2-6000.html

python Platform/ElkhartlakeBoardPkg/Script/StitchLoader.py -i <BIOS_IMAGE_NAME> -s Outputs/ehl/SlimBootloader.bin -o <SBL_IFWI_IMAGE_NAME> -p 0xAA000206

the UART log information is:
============= Intel Slim Bootloader STAGE1A =============
SBID: SB_EHLK 
ISVN: 001
IVER: 001.000.001.004.00020
SVER: C3E42632BAB695EC-dirty
FDBG: BLD(D IA32) FSP(R)
FSPV: ID($EHLFSP$) REV(09042511)
CPUV: ID(90661) UCODE(16)
Loader global data @ 0xFEF1FD10
Run  STAGE1A @ 0xFFFD8000
Load STAGE1B @ 0xFFDE0000

============= Intel Slim Bootloader STAGE1B =============
[Boot Guard] AcmStatus : 0x00000000
[Boot Guard] BootStatus: 0x00000000
[Boot Guard] Boot Guard Failed or is Disabled!
SpiInstance = FEF2F3D0
PchSpiBase at 0x00001F05
ScSpiBar0 at 0xF0000000
Loading Component KEYH:_HS_
Registering container KEYH
HASH verification for usage (0x00000100) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000100): Success
HASH verification for usage (0x00000000) with Hash Alg (0x2): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Append public key hash into store: Success
Load EXT CFG Data @ 0xFEF2FB54:0x1B4C ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000200): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Processor does not support Boot Guard.
Boot Guard Support status: 0
GetPlatformDataPtr is copied 0xFEF2FB34 
BOOT: BP0 
MODE: 0
BoardID: 0x06
PlatformName: UP6000
Memory Init
Load Security Cfg Data
FSPM CfgData assignment
Load Memory Cfg Data
PlatformDebugConsent = 0
DciEn = 0
DciModphyPg = 0
DciUsb3TypecUfpDbg = 2
DciDbcMode = 4
DebugInterfaceEnable = 2
CpuTraceHubMode = 0
CpuTraceHubMemReg0Size = 2
CpuTraceHubMemReg1Size = 2
PchTraceHubMode = 0
PchTraceHubMemReg0Size = 2
PchTraceHubMemReg1Size = 2
FiaLaneReversalEnable = 0
Load Graphics Cfg Data
HDA Policy: SSP/I2S Link Mode
S0ix Status = 0
Call FspMemoryInit ...

randylintw, 2022-10-12:
Hi, 
Is it possible to build with ver and test again? 
  commit bf4a56033fffe9629ecbfbf778adc9cf1c64a6ac (tpm)
  Author: Guo Dong <guo.dong@intel.com>
  Date:   Thu Dec 23 16:11:03 2021 -0700

MiaoMKung, 2022-10-12:
> Hi, Is it possible to build with ver and test again? commit [bf4a560](https://github.com/slimbootloader/slimbootloader/commit/bf4a56033fffe9629ecbfbf778adc9cf1c64a6ac) (tpm) Author: Guo Dong [guo.dong@intel.com](mailto:guo.dong@intel.com) Date: Thu Dec 23 16:11:03 2021 -0700

thanks your reply. 
sorry, what's the ver? build with ver?

randylintw, 2022-10-13:
Yes, could you git reset --hard  bf4a560 & build the SBL and test again?

randylintw, 2022-10-14:
Hi, 
  Maybe the SPI connection is not stable?

MiaoMKung, 2022-10-14:
> Hi, Maybe the SPI connection is not stable?

a) yes, sorry for that. today, I just borrow another Dediprog tools to try. Both tools become works now.  

b) after I download the sbl_up2_6000_ifwi_1013. bin file to board.  the log information is same.

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_EHLK 
ISVN: 001
IVER: 001.000.001.002.00020
SVER: BF4A56033FFFE962
FDBG: BLD(D IA32) FSP(R)
FSPV: ID($EHLFSP$) REV(09042511)
Loader global data @ 0xFEF1FD08
Run  STAGE1A @ 0xFFFD8000
Load STAGE1B @ 0xFFDE0000

============= Intel Slim Bootloader STAGE1B =============
[Boot Guard] AcmStatus : 0x00000000
[Boot Guard] BootStatus: 0x00000000
[Boot Guard] Boot Guard Failed or is Disabled!
SpiInstance = FEF303D0
PchSpiBase at 0x00001F05
ScSpiBar0 at 0xF0000000
Loading Component KEYH:_HS_
Registering container KEYH
HASH verification for usage (0x00000100) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000100): Success
HASH verification for usage (0x00000000) with Hash Alg (0x2): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Append public key hash into store: Success
Load EXT CFG Data @ 0xFEF30B54:0x1B48 ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000200): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Processor does not support Boot Guard.
Boot Guard Support status: 0
GetPlatformDataPtr is copied 0xFEF30B34 
BOOT: BP0 
MODE: 0
BoardID: 0x06
PlatformName: UP6000
Memory Init
Load Security Cfg Data
FSPM CfgData assignment
Load Memory Cfg Data
PlatformDebugConsent = 0
DciEn = 0
DciModphyPg = 0
DciUsb3TypecUfpDbg = 2
DciDbcMode = 4
DebugInterfaceEnable = 2
CpuTraceHubMode = 0
CpuTraceHubMemReg0Size = 2
CpuTraceHubMemReg1Size = 2
PchTraceHubMode = 0
PchTraceHubMemReg0Size = 2
PchTraceHubMemReg1Size = 2
Load Graphics Cfg Data
HDA Policy: SSP/I2S Link Mode
S0ix Status = 0
Call FspMemoryInit ...

randylintw, 2022-10-17:
Hi MaioM, 
1. Is this your first try boot on SBL with up2 6000 board? 
2. From log hang on FspMemoryInit, it looks hang on FSP. I try to force crb to boot with board id 6 on SBL plus UP2 6000 IFWI, it can enter the OsLoader.

MiaoMKung, 2022-10-17:
Hello randylintw,
> 1. Is this your first try boot on SBL with up2 6000 board?
->  yes, first time. never success
> 2. From log hang on FspMemoryInit, it looks hang on FSP. I try to force crb to boot with board id 6 on SBL plus UP2 6000 IFWI, it can enter the OsLoader.
-> Have you try to use the UP2 6000 IFWI bin file which I have attached in previous message?
    or just the IFWI file you build by yourself?

randylintw, 2022-10-18:
The IFWI binary is from https://www.aaeon.com/en/p/up-boards-up-squared-6000

MiaoMKung, 2022-10-18:
> The IFWI binary is from https://www.aaeon.com/en/p/up-boards-up-squared-6000

can you share your builded bin file to me？i will
verify it on my Up2 6000 board?

stanleyintel, 2022-10-20:
Hi MiaoMKung,

    Because your system hangs in FSP, please try with debug FSP (get it from your Intel rep, and then build SBL with the "-fd" flag, e.g.,  python BuildLoader.py .... -fd ), and share us console logs.

MiaoMKung, 2022-10-20:
> Hi MiaoMKung, Because your system hangs in FSP, please try with debug FSP (get it from your Intel rep, and then build SBL with the "-fd" flag, e.g., python BuildLoader.py .... -fd ), and share us console logs.

sorry, is FSP not automatically download when i run the script ? I saw a folder named "Download" contain the FSP when i run the build script.

MiaoMKung, 2022-10-20:
> > The IFWI binary is from https://www.aaeon.com/en/p/up-boards-up-squared-6000
> 
> can you share your builded bin file to me？i will verify it on my Up2 6000 board?
> 
> i used the UNEHAM20.bin ((v2.0)) as the IFWI input to build.

I think we use same factory BIOS： UNEHAM20.bin
same build scripts with option is same: python BuildLoader.py build ehl
same stitch scripts：python Platform/ElkhartlakeBoardPkg/Script/StitchLoader.py -i <BIOS_IMAGE_NAME> -s Outputs/ehl/SlimBootloader.bin -o <SBL_IFWI_IMAGE_NAME> -p 0xAA000206

in theory, these script will use the same FSP. right?

Only different is the hardware. 

it can be booted successfully which is forced on CRB. 

so i want to double check my problem:
- environment setting
- build steps
- UP6000 Hardware mismatch for UP6000.DLT files

thank you

MiaoMKung, 2022-10-20:
> > > > The IFWI binary is from https://www.aaeon.com/en/p/up-boards-up-squared-6000
> > > 
> > > 
> > > can you share your builded bin file to me？i will verify it on my Up2 6000 board?
> > > i used the UNEHAM20.bin ((v2.0)) as the IFWI input to build.
> > 
> > 
> > I think we use same factory BIOS： UNEHAM20.bin same build scripts with option: python BuildLoader.py build ehl same stitch scripts：python Platform/ElkhartlakeBoardPkg/Script/StitchLoader.py -i <BIOS_IMAGE_NAME> -s Outputs/ehl/SlimBootloader.bin -o <SBL_IFWI_IMAGE_NAME> -p 0xAA000206
> > in theory, these script will use the same FSP. right?
> > Only difference is the hardware.
> > it can be booted successfully,  it is forced on CRB for UP6000  HW description.
> > so i want to double check my problem:
> > 
> > * environment setting
> > * build steps
> > * UP6000 Hardware mismatch for UP6000.DLT files
> > 
> > thank you

stanleyintel, 2022-10-20:
> sorry, is FSP not automatically download when i run the script ? I saw a folder named "Download" contain the FSP when i run the build script.

Hi MiaoMKhung,
You need to get the whole EHL SBL release package (says, EHL MR4) and its Release Notes from Intel rep. In the Release Notes, you can find where the FspBin is, please replace your FspBin (auto-downloaded and under Silicon/ElkhartlakePkg/FspBin/) with the FspBin from the release package.

MiaoMKung, 2022-10-20:
> Hi MiaoMKhung, You need to get the whole EHL SBL release package (says, EHL MR4) and its Release Notes from Intel rep. In the Release Notes, you can find where the FspBin is, please replace your [...]

that sounds good. I will check that. thank you

stanleyintel, 2022-10-20:
BTW, we have EHL MR5 released early this month, so you may also try to get the EHL MR5 from Intel rep.  (your previous log shows its FSP is based on EHL MR4, but it is okay. Your issue so far looks like not related to MR4 or MR5)

MiaoMKung, 2022-10-25:
Dear stanleyintel,

is this intel file number 616714 for MR4 package?
I got the package, there is no fspbin file in this folder.

[image]

but I have checked by folder of FSP, I saw a fspDbg.bin file. 
[image]

i tried to build with: python BuildLoader.py build ehl -fd

and get the sbl_up2_6000_ifwi_1025.bin as in attachment.

[sbl_up2_6000_ifwi_1025.zip](https://github.com/slimbootloader/slimbootloader/files/9860449/sbl_up2_6000_ifwi_1025.zip)

the log is below: same place to hangs in FSP:

�
============= Intel Slim Bootloader STAGE1A =============
SBID: SB_EHLK 
ISVN: 001
IVER: 001.000.001.002.00020
SVER: BF4A56033FFFE962
FDBG: BLD(D IA32) FSP(D)
FSPV: ID($EHLFSP$) REV(09042511)
Loader global data @ 0xFEF1FD08
Run  STAGE1A @ 0xFFFD8000
Load STAGE1B @ 0xFFDE0000

============= Intel Slim Bootloader STAGE1B =============
[Boot Guard] AcmStatus : 0x00000000
[Boot Guard] BootStatus: 0x00000000
[Boot Guard] Boot Guard Failed or is Disabled!
SpiInstance = FEF303D0
PchSpiBase at 0x00001F05
ScSpiBar0 at 0xF0000000
Loading Component KEYH:_HS_
Registering container KEYH
HASH verification for usage (0x00000100) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000100): Success
HASH verification for usage (0x00000000) with Hash Alg (0x2): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Append public key hash into store: Success
Load EXT CFG Data @ 0xFEF30B54:0x1B48 ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000200): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Processor does not support Boot Guard.
Boot Guard Support status: 0
GetPlatformDataPtr is copied 0xFEF30B34 
BOOT: BP0 
MODE: 0
BoardID: 0x06
PlatformName: UP6000
Memory Init
Load Security Cfg Data
FSPM CfgData assignment
Load Memory Cfg Data
PlatformDebugConsent = 0
DciEn = 0
DciModphyPg = 0
DciUsb3TypecUfpDbg = 2
DciDbcMode = 4
DebugInterfaceEnable = 2
CpuTraceHubMode = 0
CpuTraceHubMemReg0Size = 2
CpuTraceHubMemReg1Size = 2
PchTraceHubMode = 0
PchTraceHubMemReg0Size = 2
PchTraceHubMemReg1Size = 2
Load Graphics Cfg Data
HDA Policy: SSP/I2S Link Mode
S0ix Status = 0
Call FspMemoryInit ...

stanleyintel, 2022-10-26:
> Dear stanleyintel,
> 
> a) is this intel file for MR4 package? I got the package, is this one? ![image](https://user-

Hello, 

   for (a) It is not SBL MRx release package. please get doc (#616714) from your Intel rep.
   for (b), please replace the FspDbg.bin (or the whole folder) with the one(s) from (a).
   The output log for (b) is huge, so you may want to compress it first before attaching it here.

MiaoMKung, 2022-10-26:
Dear stanley,

thanks for providing the file number. It's great to see more information.
attached please see the debug log:
[Debug log_1026.txt](https://github.com/slimbootloader/slimbootloader/files/9870252/Debug.log_1026.txt)

regards

stanleyintel, 2022-10-27:
> Dear stanley,
> 
> thanks for providing the file number. It's great to see more information. attached please see the debug log: [Debug log_1026.txt](https://github.com/slimbootloader/slimbootloader/files/9870252/Debug.log_1026.txt)
> 
> regards

Hello MiaoMKung,

The hang happens in MRC (memory reference code), which means the configuration in your SPD does not match your memory module and/or does not tune for HW circuit layout. According to the log, the SBL is using a hardcoded SPD table. And the SPD table is the same as the reference code (i.e., Spd_Lpddr4_8G.bin), e.g., your DLT may specify something like:

  MEMORY_CFG_DATA.MemorySpdPtr00    | {FILE: Spd_Lpddr4_8G.bin}
  MEMORY_CFG_DATA.MemorySpdPtr10    | {FILE: Spd_Lpddr4_8G.bin}

Please work with your memory vendor to get the correct SPD table, and replace the default one under Platform/ElkhartlakeBoardPkg/CfgData. After updating the SPD file, if the memory issue is still present, you can get help from our memory expert by filing an IPS (Intel Premier Support) with the Debug.log_1026.txt attached.

MiaoMKung, 2022-11-02:
Hello,  
come back for updated SPD. 
it seems no RAM problem. but still report a error in config block.
and the log:
[Debug log_1102_3.txt](https://github.com/slimbootloader/slimbootloader/files/9920807/Debug.log_1102_3.txt)

stanleyintel, 2022-11-03:
> Hello, come back for updated SPD. it seems no RAM problem. but still report a error in config block. and the log: [Debug log_1102_3.txt](https://github.com/slimbootloader/slimbootloader/files/9920807/Debug.log_1102_3.txt)

Yes, the memory issue is fixed. Great!
Now the SBL hangs in FSP-s (stage2). According to the log, two things should be fixed first:

1. SBL source is out of date. The FSP is based on MR4, but the source and microcode is based on older release (MR2). There were significant update during/after MR3 (e.g., FSP UPD), If the source is not aligned with FSP binary, some unexpected error will happen. Also, I am not sure about other FW ingredients (e.g., CSEM, PMC....). All of them should be aligned. Suggest to upgrade to MR5.

2. No PSE FW, but it is enabled. Please either adding PSE FW (IPFW/PseFw.bin) or disable PSE by setting PchPseEnable = 0 in Platform/ElkhartlakeBoardPkg/CfgData/CfgData_Memory.yaml. If you want to disable it, please note that there were several patches addressing it in MR3 and MR4. Migrating your code to MR5 might be easier.

MiaoMKung, 2022-11-29:
Hello,

sorry for late updating.

updated to latest SBL. but still reboot happen, and the log :
[Debug log_1129_1.txt](https://github.com/slimbootloader/slimbootloader/files/10111924/Debug.log_1129_1.txt)

thank you

stanleyintel, 2022-11-29:
Hello,

The reboot is normal, because of missing or corrupted PSE FW. In log, we can find "PSE is enabled but missing PSE FW !! Bad Buffer Size"

GongWeijia, 2022-12-14:
Hello,

progress: can boot successfully, 

I want to install debian by instruction of: https://slimbootloader.github.io/how-tos/boot-ubuntu.html

but fail on OS image installation.

attached please see the debug log.
[Debug log_1214_1.txt](https://github.com/slimbootloader/slimbootloader/files/10224314/Debug.log_1214_1.txt)

how to enable HECI 2, 3, 4?

thank you

MiaoMKung, 2022-12-14:
Hello,

progress: can boot successfully now.

I want to install debian by follow the instruction of: https://slimbootloader.github.io/how-tos/boot-ubuntu.html

but failed on OS image installation.

attached please see the debug log.
[Debug log_1214_1.txt](https://github.com/slimbootloader/slimbootloader/files/10224314/Debug.log_1214_1.txt)

how to enable HECI 2, 3, 4?

thank you

stanleyintel, 2022-12-14:
Hello,
What does it mean "failed on OS image installation"? The log shows SBL can boot (jump) to Linux kernel.

MiaoMKung, 2022-12-14:
Hello stanley, 

a) "failed on OS image installation", i mean
OS image installation hangs on below logs. 
Starting Kernel ...

Switch to LongMode and jump to 64-bit kernel entrypoint ..._

b) do you know what's the reason? I saw the log showing that HECI function is disable.

stanleyintel, 2022-12-14:
Hi,
It is not always "hang". If the kernel command line does not have proper console setting, the kernel boot up message will not be shown via serial port. Also, if the getty does not have proper setting, it will not be able to login Linux via serial console. As I know, most Linux installation disks (LiveCD) do not have such settings. You can change grub.conf to enable limited kernel dmesg, e.g., see similar [post ](https://github.com/slimbootloader/slimbootloader/issues/1739). The messages are limited so not very useful and meaningful unless for debugging Linux kernel. The You should try to attach an external display and usb keyboard to control your board directly to install Linux.

For external display, we can find the "GraphicsPpiInit failed" messages in your log. You may need to customize and integrate a correct Vbt file. Please check Intel doc: 631572 to learn how to customize your VBT file. 

The log for b is normal.

stanleyintel, 2022-12-14:
There are another 2 things you may resolve:
1. Your FSP is out of date. The latest Intel SBL release is MR5 which is also supported by open source SBL. However, your FSP is still MR4. 
2. The chipsetinit fw (input\ChipInitBinary.bin) is incorrect. It does not match either MR5 or MR4 release.

Please find the MR5 FW ingredients in doc # 749311 and try to align with them.

rprangar, 2023-01-10:
Closing it as the issue is inactive for about a month. Please reopen if more information is needed.

closed 2023-01-10 by rprangar

WhiteNite11, 2023-04-21:
> Hello,
> 
> sorry for late updating.
> 
> updated to latest SBL. but still reboot happen, and the log : [Debug log_1129_1.txt](https://github.com/slimbootloader/slimbootloader/files/10111924/Debug.log_1129_1.txt)
> 
> thank you

Hi MiaoMKung,

I am having the same issue as you were with getting a UP squared 6000 board booting. It hangs on FSPMemoryInit and I believe it's due to bad SPD file. We are using the 2GB RAM board (https://www.aaeon.com/en/p/up-boards-up-squared-6000). I was hoping you could share how you came to a working SPD file. Did you contact the manufacturer or somehow extract the correct SPD information from your trip. If you could share your final SPD bin file that would be a great data point for our team to see what it should look like.

Thanks and congratulations on getting past your issue.

stanleyintel, 2023-04-24:
Hi WhiteNite,

   As I know, most customers get the SPD from memory vendor, while some customers get the SPD from their UEFI BIOS source and the other customers develop the SPD by themselves. If you want to develop the SPD, the first step is to get/read the SPD spec from [JEDEC ](https://www.jedec.org/), e.g., search "SPD LPDDR4" for LPDDR4.