slimbootloader/slimbootloader #1711: ElkhartLake EcRegion area capsule update

GabrieleGiannini-Seco, 2022-10-06:
We are working on a custom project with SBL MR3 for Elkhart Lake
We are using the EcRegion->InputFile are to store the firmware for our embedded controller but as we can se there is no container associated to this region that can be used to create a capsule to update this area.
We suppose that we have to add the support by ourselves but are there any specific guide or documentation on this topic? On how to expand capsule update support to other flash areas?

Best Regards
Gabriele

randylintw, 2022-10-12:
Hi, maybe you can refer the implementation of PSE FW, store in the container.
https://github.com/slimbootloader/slimbootloader/blob/ad6da2876a81d2954d7f2d410a523b240d165a0b/Platform/ElkhartlakeBoardPkg/BoardConfig.py#L294

FWupdate doc:  https://slimbootloader.github.io/security/firmware-update.html?highlight=ipfw

randylintw, 2022-10-25:
Hi Gabriele
  I saw there is an IPS for this case, please get support through it, thanks.

randylintw, 2022-12-07:
Hi @GabrieleGiannini-Seco, 
  I saw the IPS have been closed, can you close this, too?

closed 2023-01-10 by rprangar