slimbootloader/slimbootloader #1713: USB descriptor iteration error

maldf, 2022-10-10:
While iterating over a USB descriptor in GetExpectedDescriptor(), the "Len" variable always uses the 1st byte in the buffer, see
https://github.com/slimbootloader/slimbootloader/blob/107fabbebdbc8ebb60dfd038c0896a99cffd1a58/BootloaderCommonPkg/Library/UsbBusLib/UsbPeim.c#L913
This is only true for the 1st iteration, thereafter the variable should advance as the descriptor advances, it should be:
Len = Buffer[parsed]

randylintw, 2022-10-12:
Hi, 
Would you like to submit a patch to fix this issue?

maldf, 2022-10-12:
Patch attached.
[0001-USB-fix-descriptor-parsing.txt](https://github.com/slimbootloader/slimbootloader/files/9764244/0001-USB-fix-descriptor-parsing.txt)

randylintw, 2022-10-13:
Could you create a pull request for this fix?

closed 2022-10-19 by maldf