slimbootloader/slimbootloader #1714: Interrupt packet length is maximum 8, not exclusively 8

maldf, 2022-10-10:
While integrating a HID USB keyboard I found that it had an interrupt packet length of 4.
In
https://github.com/slimbootloader/slimbootloader/blob/107fabbebdbc8ebb60dfd038c0896a99cffd1a58/BootloaderCommonPkg/Library/XhciLib/XhcPeim.c#L920
it requires the interrupt packet length to be only 8.
If it is changed so that:
... (MaximumPacketLength > 8)
similar to all the other tests, the HID USB keyboard works.

randylintw, 2022-10-12:
Hi, 
Would you like to submit a patch to fix this issue?

maldf, 2022-10-12:
Patch attached:
[0001-USB-irq-transfer-size.txt](https://github.com/slimbootloader/slimbootloader/files/9764265/0001-USB-irq-transfer-size.txt)

randylintw, 2022-10-13:
Could you create a pull request for this fix?

randylintw, 2022-10-14:
Thanks, btw, your patch format is incorrect, could you modify ? 
 * Missing Signed-off-by! (Note: this must be added by the code contributor!)
WARNING - Line 3 of commit message is too long (77 >= 76).

randylintw, 2022-10-19:
Hi Mald, thanks for your contribute and you can close this issue, thanks.

closed 2022-10-19 by maldf