slimbootloader/slimbootloader #1739: [EHL] SBL bring up for EHL based board

lutooog, 2022-11-01:
Hi，
  I am working for  a self-developed board based on Elkhart Lake processor. The relevant configuration is **self DEBUG_ PORT_ NUMBER = 0x2** ， **self. ENABLE_ PSEFW_ LOADING = 1**. The pseFW. bin version is **PSE_ FW-0.20.6.2210_ 5**。

  I Used the command ' **python StitchIfwi py -c Platform\ElkhartlakeBoardPkg\Script\StitchIfwiConfig. py -s Outputs\ehl\Stitch_ Components. zip -p ehl -w .\ifwi_ Stitch - k.. SblKeys - o tsn - b vm**’ to generate IFWI.

  After flashing the IFWI to the board, run  then enter **Zephyr OS**. Uart2 port outputs a log about PSE, but no SBL related. Can you tell me why? And How can I do for entering SBL?

thank you.
[Uploading log_20221101.txt…]()

lutooog, 2022-11-01:
LOG:

============ PSE BringUp ============
BringUp Version: 0.9.0.1
Build date-time: Mar  4 2022 - 10:39:28
Build Machine:   0325a19025c0
Compiler ver:    10.3.0
Commit:          39d6a0a - dirty
=====================================
BUP: early init - DRAM adrs:
BUP: sync path by default 0x7A000000
HW revision: 0x1
BUP: cold boot
BUP: vnn req/ack - 0x100000 0x3F
BUP: fusa_ctrl =  0x2 0x2
BUP: RT_NTWK_CTRL -  0x400 0x0
BUP: iccm_lmt & misc_opt -  0x4 0x1 0xF008000 0x50101400 0x0 0x0 0xF007500 0x0 0xF003000 0x8004200 0x8004200 0xF00A600 0x0 0x0 0x7000CC60 0x7A000000 0x4 0x3 0xF00B900 0x50101000 0x1A 0xF00C800 0x50101000 0x1A 0xF00DE00 0x50100398 0x1 0x50100010 0x0 0x50100018 0xD824 0xF00EB00 0x0 0x0 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x7000CC60 0x7A000000 0x4 0x8004200 0x8004200
SRAM ECC Correction Enabled by default! - 0x0
sram_enable done!
SRAM - MPU cfg -  0x308F025 0x1308CF25
BUP: loading cycle: 0x1
BUP: starting DMA...SRAM test 0x12345678 0x8004200 0x8004200
BUP: D3 status:  0x18180018
BUP: Image modules' count: 0x3
BUP: sz/off/adrs/off_lmt: 0x205C 0x27B0 0x70000000 0x27A8 0xF00A600 0x0 0x0 0x70000000 0x7A0027B0 0x205C 0x3 0xF00B900 0x50101000 0x1A 0xF00C800 0x50101000 0x1A 0xF00DE00 0x50100398 0x1 0x50100010 0x0 0x50100018 0xD824 0xF00EB00 0x0 0x0 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x70000000 0x7A0027B0 0x205C
BUP: sz/off/adrs/off_lmt: 0x56674 0x4810 0x60000000 0x27A8 0xF00A600 0x0 0x0 0x60000000 0x7A004810 0x56674 0x3 0xF00B900 0x50101000 0x1A 0xF00C800 0x50101000 0x1A 0xF00DE00 0x50100398 0x1 0x50100010 0x0 0x50100018 0xD824 0xF00EB00 0x0 0x0 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x60000000 0x7A004810 0x10000 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x60010000 0x7A014810 0x10000 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x60020000 0x7A024810 0x10000 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x60030000 0x7A034810 0x10000 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x60040000 0x7A044810 0x10000 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x60050000 0x7A054810 0x6674
BUP: sz/off/adrs/off_lmt: 0xA10 0x5AE84 0x70008800 0x27A8 0xF00A600 0x0 0x0 0x70008800 0x7A05AE84 0xA10 0x3 0xF00B900 0x50101000 0x1A 0xF00C800 0x50101000 0x1A 0xF00DE00 0x50100398 0x1 0x50100010 0x0 0x50100018 0xD824 0xF00EB00 0x0 0x0 0xF010200 0x3F 0xF011300 0x50101400 0x50100000 0x501003A0 0x101 0xF012D00 0x70008800 0x7A05AE84 0xA10
BUP: DMA done 0x8004200 0x8004200 0x60000000
: 0x60050000 0x60019F4D 0x60018F5D 0x60019EF9
BUP: MSP|EntryPnt: 0x60050000 0x60019F4D 0xF007500 0xF004500
BUP: Ready jump into FW!
BUP: vnn req/ack - 0x100000 0x3F

ownership: 5f833b838 8000c4343
PHY conn type= 1/1
*** Booting Zephyr OS build pse_fw_main_z2px_weekly_2210.5_1555  ***

FW VERSION: 0.20.6
FW HW IP: PSE_0_00
FW BUILD: DATE 03042022, TIME 103835, BY lab_bldmstr@0325a19025c0

[1;32muart:~$ [mhlep
[1;31mhlep: command not found
[m[1;32muart:~$ [mhelp
Please press the <Tab> button to see all available commands.
You can also use the <Tab> button to prompt or auto-complete all commands or its subcommands.
You can try to call commands with <-h> or <--help> parameter for more information.

Shell supports following meta-keys:
  Ctrl + (a key from: abcdefklnpuw)
  Alt  + (a key from: bf)
Please refer to shell documentation for more details.

Available commands:
  arm_core_clk         :Display PSE ARM Core clk speed
  clear                :Clear screen.
  device               :Device commands
  devmem               :Read/write physical memory"devmem address [width
[24C[value]]
  fw_version           :Display PSE FW version
  help                 :Prints the help message.
  history              :Command history.
  i2c                  :I2C commands
  kernel               :Kernel commands
  log                  :Commands for controlling logger
  long_sleep           :Sleep for inputted minutes
  ownership            :Display IO's ownership
  pwm                  :PWM shell commands
  resize               :Console gets terminal screen size or assumes default in
[24Ccase the readout fails. It must be executed after each
[24Cterminal width change to ensure correct text display.
  shell                :Useful, not Unix-like shell commands.
  sleep                :Sleep D0ix
  stop_eclite_service  :Stop EClite service
  switch_core_clock    :Switch core clock frequency
[1;32muart:~$ [mfw_version

FW VERSION: 0.20.6
FW HW IP: PSE_0_00
FW BUILD: DATE 03042022, TIME 103835, BY lab_bldmstr@0325a19025c0

[1;32muart:~$ [m

stanleyintel, 2022-11-01:
> 

Hello,
The default EHL SBL shows UART2 log via GPIO C12/13/20/21. Does your board also use them?
If not, please configure **mUartGpioTable** in Platform/ElkhartlakeBoardPkg/Library/Stage1ABoardInitLib/Stage1ABoardInitLib.c.

lutooog, 2022-11-02:
HI,
  Thank you for your reply.
  Yes!  UART outputs SBL log after configured mUartGpioTable.
  But it does not boot to slimboot Shell, it goes to reboot loop. WHY?
 LOG:
```
Failed to get GFX HOB from FSP
MEM: 0000000000000000 00000000000A0000 00 1
MEM: 00000000000A0000 0000000000060000 00 2
MEM: 0000000000100000 00000000780E0000 00 1
MEM: 00000000781E0000 0000000000500000 01 2
MEM: 00000000786E0000 0000000000068000 00 3
MEM: 0000000078748000 0000000000008000 00 4
MEM: 0000000078750000 0000000000500000 00 2
MEM: 0000000078C50000 00000000003B0000 00 2
MEM: 0000000079000000 0000000002000000 00 2
MEM: 000000007B000000 0000000000800000 00 2
MEM: 000000007B800000 0000000000800000 00 2
MEM: 000000007C000000 0000000003C00000 00 2
MEM: 00000000FED20000 0000000000060000 00 2
MEM: 00000000FF300000 0000000000D00000 00 2
MEM: 0000000100000000 000000016C000000 00 1
MP Init (Wakeup)
SMRR Base: 0x7B000006  Mask: 0xFF800000
MP Init (Run)
Detected 4 CPU threads
 CPU  0 APIC ID: 0
 CPU  1 APIC ID: 4
 CPU  2 APIC ID: 2
 CPU  3 APIC ID: 6
SMM rebase done on 4 CPUs
PCI Enum
Call FspNotifyPhase(20) ... 40000003
FSP Requested Reboot ...
```
[log_20221102_1121.txt](https://github.com/slimbootloader/slimbootloader/files/9916959/log_20221102_1121.txt)

stanleyintel, 2022-11-02:
Hi,

Note: one thing to correct in my previous reply, only mUartGpioTable[2] and mUartGpioTable[3] are for UART2.

In addition, I checked again the EHL GPIO design (doc: #613952) and found that only GPIO C20 and C21 are designed for UART2. Thus, the reboot issue might be caused by SBL trying to configure GPIO pins with unsupported functions. May you share us your modified mUartGpioTable ?

Last, when using the default mUartGpioTable, SBL will not request for reboot (but no log), right?

lutooog, 2022-11-02:
Hi,
”Last, when using the default mUartGpioTable, SBL will not request for reboot (but no log), right?“
--No， SBL will still request for reboot（no log）, I can see this solution by current change.

I find a similar issue(#1321) . But I can't fix the reboot also.

After skipping FspResetHandler (Status), I can enter shell successfully.
[log_20221102_1641.txt](https://github.com/slimbootloader/slimbootloader/files/9918061/log_20221102_1641.txt)

stanleyintel, 2022-11-02:
Hi,

I think the previous issue is somehow different. Indeed, there have some unusual logs in your log file, but possibly the wrong configuration of GPIO pins is the main cause of infinite reboot request.  Let's try to correct them one by one.

Issue 1. about GPP_T12/T13. Their native func 2 for UART0, so you shall set DEBUG_PORT_NUMBER = 0. However, ConfigureGpio in EHL Stage1ABoardInitLib.c will not be executed if DEBUG_PORT_NUMBER = 0. You may customize the code snippet around [line 127 in EHL Stage1ABoardInitLib.c](https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ElkhartlakeBoardPkg/Library/Stage1ABoardInitLib/Stage1ABoardInitLib.c#L127).

Issue 2. one unusual log is that "PSE is enabled but missing PSE FW !! Unsupported", which means that your IFWI does not stitch **PseFw.bin**. If you intend to not include it, you should set PchPseEnable = 0 in around [line 866 in EHL CfgData_Memory.yaml](https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ElkhartlakeBoardPkg/CfgData/CfgData_Memory.yaml#L866).

lutooog, 2022-11-02:
HI，
    Issue 1. You are right！I tried to set DEBUG_PORT_NUMBER = 2， and show SBL log via GP_C20 and GP_C21. It can work to enter shell.  But SBL would reboot just once. Is it normal?
   Issue 2. Fix.
[log_20221102_1843.txt](https://github.com/slimbootloader/slimbootloader/files/9919567/log_20221102_1843.txt)

stanleyintel, 2022-11-03:
Hi,
When you said "just once", do you mean it is auto rebooted only one time after a new IFWI is flashed, or it is auto rebooted one time after every cold boot?

You may share the console log of booting SBL with debug FSP (note: need to (1) use SBL source from Intel release package, and (2) add '-fd' when BuildLoader.py), so we can check the auto reboot reason.

lutooog, 2022-11-03:
Hi，
  Sorry for unclear expression.  it is auto rebooted one time after every cold boot.
  I built the lastest version of SBL from github with "-fd".  Here is the log.
[log_20221103_1350.txt](https://github.com/slimbootloader/slimbootloader/files/9925923/log_20221103_1350.txt)

stanleyintel, 2022-11-03:
Hi,
The SBL is not built with debug FSP (you can use the SBL from github, but need to replace the whole folder under Silicon/ElkhartlakePkg/FspBin/ with the one from Intel release). The console log will be huge.

lutooog, 2022-11-04:
Hi，
  I asked intel for debug FSP. unfortunately, they said they didn't have one !
  May I ask whether you have one or not？

stanleyintel, 2022-11-04:
> Hi， I asked intel for debug FSP. unfortunately, they said they didn't have one ! May I ask whether you have one or not？

It is in Intel release package, says, for EHL MR5, it comes with doc: 616714.

lutooog, 2022-11-04:
Hi,
  Thank you for your help! Here is the log with debug FSP.
[log_20221104_1119.txt](https://github.com/slimbootloader/slimbootloader/files/9934746/log_20221104_1119.txt)

lutooog, 2022-11-04:
Hi,
   I am trying to  follow the guide to boot ubuntu from USB media. [https://slimbootloader.github.io/how-tos/boot-ubuntu.html](url). There are some ploblem:
  1.I can't see any graphics of SBL or Ubuntu installer. What should I do for showing graphics? Which interface Should I connect?
  2. system hang in "**Switch to LongMode and jump to 64-bit kernel entrypoint ..**."
  Here is log.
[log_20221104_1526.txt](https://github.com/slimbootloader/slimbootloader/files/9936034/log_20221104_1526.txt)

stanleyintel, 2022-11-04:
> Hi, Thank you for your help! Here is the log with debug FSP. [log_20221104_1119.txt](https://github.com/slimbootloader/slimbootloader/files/9934746/log_20221104_1119.txt)

Hi,
Based on your current configuration (no PSE and PchPseEnable = 0), the reboot is normal: FSP is requesting a reboot to actually disable PSE.

lutooog, 2022-11-04:
> Hi, Based on your current configuration (no PSE FW and PchPseEnable = 0), the reboot is normal: FSP is requesting a reboot to actually disable PSE.
 Thanks For your reply. And  I need your help for booting ubuntu from USB.  T.T

stanleyintel, 2022-11-04:
> Hi, I am trying to follow the guide to boot ubuntu from USB media. [https://slimbootloader.github.io/how-tos/boot-ubuntu.html](url). There are some ploblem: 1.I can't see any graphics of SBL or [...]

Hi,

   It may not hang, just no display to show Ubuntu GUI, and also no proper tty settings for linux kernel to output logs to console (i.e., your kernel command line: Cmd Args: file=/cdrom/preseed/ubuntu.seed boot=casper only-ubiquity quiet splash ---)

  Back to display issue, please check doc: 631572 to learn how to customize your VBT file (under Platform/ElkhartlakeBoardPkg/VbtBin/). In brief, you need to use "DisCon" and Vbt.json to customize Vbt.dat according to your HW configuration. Note that you can get the Vbt.json from Intel rep (in the package of EHL "GOP Driver and Gfx PEIM Build 18.0.1041").

  When customizing the VBT, if you still encounter display issue, please file a case in IPS (Intel premier support) and provide them your debug FSP logs.

lutooog, 2022-11-07:
Hi，
  For tty settings, I tried some different parameters but they didn't work. Do you have some suggest for it .
` Cmd Args: init=/init root=/dev/ram0 rw 3 console=ttyS0,115200 console=ttyS1,115200 console=ttyS2,115200 console=ttyS4,115200`

stanleyintel, 2022-11-08:
Hello,

May I know what's your expectation from the uart console? You are using Ubuntu LiveCD that is designed for installing Ubuntu. Thus, first, enabling its console output to serial port may result in strange behavior. You can try it, but the issue is out of scope of this forum. Please check with your Linux kernel expert and Ubuntu forum. Second, the LiveCD does not enable getty, so you can only get few logs from console.

On Intel CRB, I append the following command line: loglevel=8 earlycon=uart,mmio32,0x8101d000,115200
where the mmio address comes from the BAR0 of "pci 0 19 2"
It only shows the first 1 or 2 seconds of kernel booting.

Last, please note that SBL OS Loader only designs to load grub.cfg [for debugging](https://github.com/slimbootloader/slimbootloader/blob/master/PayloadPkg/OsLoader/LoadImage.c#L868). If your product is to run Ubuntu, please try with "UEFI Payload". Otherwise, in the SBL release build, you will need to either (1) customize OS Loader to load grub.cfg, or (2) create a container image as described in the wiki you mentioned.

lutooog, 2022-11-08:
Hi,
I can see the Ubuntu GUI and finish to install it. But when I reboot, I can't enter the OS that I just installed.
`Last, please note that SBL OS Loader only designs to load grub.cfg [for debugging](https://github.com/slimbootloader/slimbootloader/blob/master/PayloadPkg/OsLoader/LoadImage.c#L868). If your product is to run Ubuntu, please try with "UEFI Payload". Otherwise, in the SBL release build, you will need to either (1) customize OS Loader to load grub.cfg, or (2) create a container image as described in the wiki you mentioned.`
In other words, when I finished to install ubuntu to a disk with SBL OS Loader, I can't run ubuntu form this disk?

lutooog, 2022-11-09:
Hi,
   Anyaway, Thank you very much. You helped me a lot.

lutooog, 2022-11-09:
when I reboot, there is a error. It maybe a disk write error?
```
======== Try Booting with Boot Option 3 ========
BootMediumPciBase(0x1700)
Getting boot image from SATA
Init AHCI controller C00B8000
Skip reinit AHCI controller
Try HwPart 0
Try to find boot partition for HW part 0
Protective MBR does not protect the whole drive!
Part 00:              0x00000800--0x000EE7FF, LBA count: 0xEE000
Part 01:              0x000EE800--0x0E6DAFFF, LBA count: 0xE5EC800
Part 02:              0x0E6DB000--0x0EE7B7FF, LBA count: 0x7A0800
Partition type: GPT  (3 logical partitions)
Find partition success
Init File system
Detected EXT on StartBlock 976896 Part 0
BootSlot = 0x0
Open file '/boot/sbl_os' failed, Status = Not Found
LoadBootImage ImageType-0 Not Found
Try booting Linux from config file ...
Checking config.cfg
Open file 'config.cfg' failed, Status = Not Found
Checking boot/grub/grub.cfg
Load file boot/grub/grub.cfg [size 0x1FD0]: Success

Boot Menu:
  0 - * Ubuntu
  1 -   Ubuntu, with Linux 5.15.0-46-generic
  2 -   Ubuntu, with Linux 5.15.0-46-generic (recovery mode)
  3 -   Memory test (memtest86+)
  4 -   Memory test (memtest86+, serial console 115200)
Load file /boot/vmlinuz-5.15.0-46-generic [size 11431744 bytes]: Success
**Could not find FileBlock #0 in the index extent data!
Load file /boot/initrd.img-5.15.0-46-generic [size 69780277 bytes]: Load Error
Load initrd failed!**
Failed to Load Boot Image
CloseFileSystem: FsType 1
Payload normal heap: 0x10000000 (0x10F000 used)
Payload reserved heap: 0x500000 (0x0 used)
Payload stack: 0x20000 (0xA8C used)
Deinit AHCI controller 0
Not a AHCI controller or Disabled

```

stanleyintel, 2022-11-09:
> 

Hi,
The log shows it fails to find the first node of /boot/initrd.img-5.15.0-46-generic. The error is raised from [here](https://github.com/slimbootloader/slimbootloader/blob/master/BootloaderCommonPkg/Library/Ext23Lib/Ext2Fs.c#L334). It is possible caused by corrupted file system/device or also possible an Ext2Lib implementation error.

Do you know how large your root filesystem is? Or if possible, could you format it as "EXT2" with smaller partition size during installing Ubuntu?

lutooog, 2022-11-09:
Hi,
If format it as "EXT2",  it would report :
```
Detected EXT on StartBlock 976896 Part 0
BootSlot = 0x0
ASSERT [OsLoader] e:\work\projects\x6425e\code\slimbootloader\BootloaderCommonPkg\Library\FullMemoryAllocationLib\Pool.c(533): CR has Bad Signature

STAGE_PAYLOAD: System halted!
```

lutooog, 2022-11-09:
Hi,
  Here is the reboot log if root filesystem part is formated as "EXT4" during installing Ubuntu.
[log_formated_as_EXT4_202211091735.txt](https://github.com/slimbootloader/slimbootloader/files/9969559/log_formated_as_EXT4_202211091735.txt)

stanleyintel, 2022-11-09:
Hello,
You may hit an unknown Ext2Lib bugs. 
1. Which version of Ubuntu release did you install?
2. May you tell me the brand/model of your SATA disk?

Suggestion: please try with UEFI Payload so the bug will not block your project. 
I will debug the Ext2Lib issue from my side, but it may take few days, because so far I cannot reproduce it.

Method to integrate UEFI Payload:
1. Assume you have Intel release package. Decompress the UefiPldPlatform.zip from the release package.
2. In UefiPldPlatform, execute 
*python BuildPayload.py build ehl -r*
Then you can find the Uefi Payload: ./Build/EhlPlatformPkgX64/$compiler_identifier/FV/*UEFIPAYLOAD.fd*. 
3. In SBL source, create a directory: Platform/ElkhartlakeBoardPkg/Binaries
4. Copy the Uefi Payload (*UEFIPAYLOAD.fd*) to Platform/ElkhartlakeBoardPkg/Binaries
5. Modify your DLT file, set the following line 
*GEN_CFG_DATA.PayloadId | 'UEFI'*
6. Build SBL with 
*python BuildLoader.py build tgl -p  "OsLoader.efi:LLDR:Lz4;UEFIPAYLOAD.fd:UEFI:Lzma*

Done

lutooog, 2022-11-09:
Hi,
1.The version of Ubuntu is ubuntu-20.04.5-desktop-amd64.
2.Brand of my SATA disk is Kingston.   Model is ‎SV300S37A/240G.

I just did step by step without no change from [[https://slimbootloader.github.io/how-tos/boot-ubuntu.html]].

Thank you for your suggest!

Attachment is the picture of my disk.
[image]

stanleyintel, 2022-11-10:
Thanks for the info. Will check and update you soon.

stanleyintel, 2022-11-11:
Hi,
The Ext23Lib issue has been fixed. Thanks for finding it out and reporting us.
If possible, please try with the patch in the above link.

lutooog, 2022-11-15:
Hi，
     Thank you!  The patch can work for me!
     Another issue. I try to boot OS through mSATA. Boot option can find the harddisk, but it don't do any boot work. 
     The question： do SBL support to boot OS through mSATA? If supported, How to configure?
```
      ======== Try Booting with Boot Option 0 ========
BootMediumPciBase(0x1700)
Getting boot image from SATA
Init AHCI controller C00B8000
AHCI port [1] has a [harddisk]
Payload normal heap: 0x10000000 (0x10F000 used)
Payload reserved heap: 0x500000 (0x0 used)
Payload stack: 0x20000 (0x5BC used)
Deinit AHCI controller 0
Not a AHCI controller or Disabled
```

lutooog, 2022-11-15:
Thank you! The patch can work for me!

closed 2022-11-29 by lutooog