slimbootloader/slimbootloader #1768: [EHL]How to config to use the two pse-GbEs

lutooog, 2022-11-29:
Hi, 
     I want to use the two pse-GbEs. the ethernet PHYs are supported. But I can't see any internet controller of the two pse-GbEs. How should I config for that.

```
Press any key within 2 second(s) to enter the command shell
Shell>
Shell> pci
00:00.0 8086:452A Bridge Device
00:02.0 8086:4571 Display Controller
00:08.0 8086:4511 Base System Peripherals
00:10.0 8086:4B44 Serial Bus Controllers
00:10.1 8086:4B45 Serial Bus Controllers
00:14.0 8086:4B7D Serial Bus Controllers
00:14.2 8086:4B7F Memory Controller
00:15.0 8086:4B78 Serial Bus Controllers
00:15.2 8086:4B7A Serial Bus Controllers
00:15.3 8086:4B7B Serial Bus Controllers
00:17.0 8086:4B63 Mass Storage Controller
00:19.0 8086:4B4B Serial Bus Controllers
00:19.2 8086:4B4D Simple Communications Controllers
00:1A.0 8086:4B47 Base System Peripherals
00:1A.1 8086:4B48 Base System Peripherals
00:1E.0 8086:4B28 Simple Communications Controllers
00:1E.1 8086:4B29 Simple Communications Controllers
00:1F.0 8086:4B00 Bridge Device
00:1F.4 8086:4B23 Serial Bus Controllers
00:1F.5 8086:4B24 Serial Bus Controller
```s

randylintw, 2022-11-29:
Do you enable PSE ? 
Enabled and you should see 
Ethernet GbE Controllers PCI ID 8086:4BA0 or 8086:4BB0 [Intel® Programmable Services Engine (Intel® PSE):Gigabit Ethernet TSN Controller]

lutooog, 2022-11-29:
HI， 
  If I enabled PSE， PSE would boot Zephyr OS so that Linux cannot be laoded. I want to use pse-GbEs in Linux , how to do that?

randylintw, 2022-11-29:
I use latest SBL source code and enable PSE FW, the ubuntu network setting can see two network devices [4b28/4b29].
OS is booting with osloader and  the two network connections are fine. No change needed.

lutooog, 2022-11-30:
Hi，
    what I do for enabling PSE FW is
  ```
  1.set self.ENABLE_PSEFW_LOADING = 1 in Platform/ElkhartlakeBoardPkg/BoardConfig.py
    2.set PchPseEnable value = 1 in Platform/ElkhartlakeBoardPkg/CfgData/CfgData_Memory.yaml
    3.put a PseFw.bin from intel into ifwi_stitch/IPFW folder I created
    4.bulid SBL with "python BuildLoader.py build ehl"
    5.generate IWFI with "python  Platform\ElkhartlakeBoardPkg\Script\StitchIfwi.py -c Platform\ElkhartlakeBoardPkg\Script\StitchIfwiConfig.py -s Outputs\ehl\Stitch_Components.zip  -p ehl -w .\ifwi_stitch\  -k ..\SblKeys -b  fvme"
```
     
     Is it right?

randylintw, 2022-11-30:
PSE booting looks ok. The Zyphyr is running on PSE.
About the OS side, do you use ubuntu or Yocto?
One thing you can try first is booting on UEFI payload to see if Osloader problem ?

lutooog, 2022-11-30:
I use ubuntu.
yes, I wil have a try.
Anyway, Could you give me your PSE FW file to let me try again?

randylintw, 2022-12-01:
I build with https://github.com/intel/pse-fw
The system still can detect two eths. 
BTW,  in the default ubuntu image, after enter kernel, the uart looks not connect to tty.  The last dmsg is like yours. 
So that I use keyboard/monitor to check the eths status.

randylintw, 2022-12-07:
HI @lutooog , 
  Are you able to detect two Eths on ubuntu?

lutooog, 2022-12-08:
Hi，
   Thank you very much for your help!
    I still can't detect two Eths on ubuntu. 
    When I disable PSE, I can see ubuntu log in uart, but the ubuntu log can't be seen when enbale PSE. 
    I stop debugging the issue since I change  the way to connect internet through PCIe netcard. I will continue to debug this issue at the proper time.

randylintw, 2022-12-13:
You can try to disable PSE log by setting this FSP UPD. 
slimbootloader\Platform\ElkhartlakeBoardPkg\CfgData\CfgData_Silicon.yaml

  - PchPseLogOutputChannel :
      name         : PCH PSE Log Output Channel
      type         : EditNum, HEX, (0x0,0xFFFFFFFF)
      help         : >
                     Set PSE Log Output Channel. 0- internal memory; 1-6- UART channels; Other- shut down
Set to 0 or others to shutdown the logs.

lutooog, 2023-01-05:
HI, randylintw,
  Apologies for the super late reply. I set PSE Log Output Channel to 0 or 0xffffffff, but It doesn't work.
  Here is log with SBL using Debug FSP.
  Do you have some idea?
[log_202301051948.txt](https://github.com/slimbootloader/slimbootloader/files/10351862/log_202301051948.txt)

closed 2023-01-11 by lutooog