slimbootloader/slimbootloader #1780: Included Lzma SDK 19 has a dangling pointer problem

galaxy4public, 2022-12-12:
The included Lzma SDK version 19 has a dangling pointer problem and does not compile with GCC 12.  Updating SDK to the latest version (which is 22 at the moment) and adding "CpuArch.o " to the list of objects in the Makefile in LzmaCompress solves the issue.

rprangar, 2022-12-12:
Thanks for reporting the issue. We will take a look.

tsaikevin, 2022-12-13:
@galaxy4public, there is a code change in tianocore may fix the similar issue. Refer to link below

https://github.com/tianocore/edk2/commit/85021f8cf22d1bd4114803c6c610dea5ef0059f1

would you try and find out if it helps? thanks

galaxy4public, 2022-12-14:
It does address the issue. However, isn't it easier just to synchronise with the upstream? The SDK version went up from 19 to 22 and has several bugs fixed and some enhancements introduced. For my build, I just updated the Lzma SDK and then commented here, so other people would be aware.

rprangar, 2022-12-14:
Yes, we will rebase.

closed 2022-12-19 by tsaikevin