slimbootloader/slimbootloader #1785: 【EHL】How to clear Smbus SPD Write Disable bit?

lutooog, 2022-12-15:
Hi,
     I want to clear Smbus SPD Write Disable bit, how can I clear it in SBL?
```
[    4.886701] i2c_dev: i2c /dev entries driver
[    4.900674] i801_smbus 0000:00:1f.4: SPD Write Disable is set
[    4.916291] i801_smbus 0000:00:1f.4: SMBus using PCI interrupt
```

stanleyintel, 2022-12-15:
Hi,
Please note: For security recommendations, SPD write disable bit must be set.
If you want to not set it, you may try FSP-M UPD: SmbusSpdWriteDisable = 0.

lutooog, 2022-12-15:
I can't find where to modify SmbusSpdWriteDisable. How to config it?
```
Silicon/ElkhartlakePkg/FspBin/Fsp.bsf:        $gPlatformFspPkgTokenSpaceGuid_SmbusSpdWriteDisable            1 bytes    $_DEFAULT_ = 0x01
Silicon/ElkhartlakePkg/FspBin/Fsp.bsf:    Combo $gPlatformFspPkgTokenSpaceGuid_SmbusSpdWriteDisable, "SMBUS SPD Write Disable", &EN_DIS,
Silicon/ElkhartlakePkg/Include/FspmUpd.h:  UINT8                       SmbusSpdWriteDisa
```

stanleyintel, 2022-12-15:
Hi,
Because it is a FSP-m UPD, you can set it in UpdateFspConfig() in Platform/ElkhartlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c.

closed 2022-12-15 by lutooog

sharkblue2009, 2023-10-18:
@stanleyintel 

If there anyway to dynamic enable/disable the SpdWrite, from EDS document, the SPDWD bit in HCFG register is RW,  may we change the bit under Linux?

stanleyintel, 2023-10-18:
Hi,
 How about writing 1? Under Linux, the resource (PCIe config space) is occupied (allocated) by kernel driver, so you will need to find a way to workaround it. The method is not related to SBL, but I think you can modify kernel driver to expose a node for user to update it.