slimbootloader/slimbootloader #1799: [EHL]hang at Call FspMemoryInit...

lutooog, 2023-01-05:
HI,
     SBL hang at Call FspMemoryInit...,Here is the debug log. What issue is it?
```
 C1.R0:  Left    Right   Width   Center  RxDqsPN
  B0:   -31     31      62      0       32
ERROR!! NO EYE found for Channel: 1 Rank: 0 Byte: 1
  B1:   0       0       0       0       32
ERROR!! NO EYE found for Channel: 1 Rank: 0 Byte: 2
  B2:   0       0       0       0       32
  B3:   -31     31      62      0       32
WARNING!! lWidth <= 12 for Channel: 1 Rank: 0 Byte: 4
  B4:   3       13      10      8       40
  B5:   -31     31      62      0       32
ERROR!! NO EYE found for Channel: 1 Rank: 0 Byte: 6
  B6:   0       0       0       0       32
ERROR!! NO EYE found for Channel: 1 Rank: 0 Byte: 7
  B7:   0       0       0       0       32
Final centering at best vref Failed
MRC task Early ReadMPR Timing Centering 2D -- FAILED, Status = 6h. Stack Depth: 55740
<MrcTimer Task="Early ReadMPR Timing Centering 2D" msec="739"/>
MRC timer: Total time for SAGV Low point = 10856 msec.

MRC task -- debug print: 0
MRC timer: Total time to execute tasks = 10856 msec.
Memory initialization has failed

ASSERT_EFI_ERROR (Status = Device Error)
ASSERT c:\source\work7\6c9de795f1e4f2d0\Intel\ClientOneSiliconPkg\IpBlock\MemoryInit\Icl\LibraryPrivate\PeiMemoryInitLib\MemoryInit.c(892): !EFI_ERROR (((RETURN_STATUS)(0x80000000 | (7)))
```)
[log_20230105.txt](https://github.com/slimbootloader/slimbootloader/files/10350027/log_20230105.txt)

lutooog, 2023-01-05:
[log_20230105.txt](https://github.com/slimbootloader/slimbootloader/files/10350038/log_20230105.txt)

stanleyintel, 2023-01-06:
Hi,

Finding: Please contact your memory provider to get the correct SPD file.

Detail:

From the log, your SBL is using the default **hardcoded SPD**, which means your board DLT file has the following configuration:
MEMORY_CFG_DATA.MemorySpdPtr00    | {FILE: Spd_Lpddr4_8G.bin}
MEMORY_CFG_DATA.MemorySpdPtr10    | {FILE: Spd_Lpddr4_8G.bin}
However, Platform/ElkhartlakeBoardPkg/CfgData/Spd_Lpddr4_8G.bin is the memory module for Intel EHL CRB. Please replace it with the memory SPD from your memory provider.

lutooog, 2023-01-06:
Hi,
  the menory module of my board  is the same as  Intel EHL CRB.
   I try to run SBL in three boards. Just one hangs at Call FspMemoryInit...

stanleyintel, 2023-01-06:
Hi,
I see, and then please work with your Intel rep to file an Intel Premier Support (IPS) case to get help from our HW and memory experts.

closed 2023-01-10 by rprangar