slimbootloader/slimbootloader #1800: [EHL]After enabling PSE_FW, SBL can't boot linux OS

lutooog, 2023-01-06:
Hi
    point to issue #1768
    I try to run the newest SBL with enabling PSE_FW, but it can't boot linux OS.  As a reference, when disablig PSE_FW, it can see dmesg of linux.
   Here is the log of SBL with Debug FSP. Do you have some idea?
   
[log_202301051948.txt](https://github.com/slimbootloader/slimbootloader/files/10357616/log_202301051948.txt)

lutooog, 2023-01-06:
Log uart use this port.
[image]

stanleyintel, 2023-01-06:
Hi,

I am not sure if my understanding is correct: 
1. Linux can output log to serial port when PSE_FW is not enabled
2. Linux cannot output log to serial port when PSE_FW is enabled

Does your board have working I/O devices? (e.g., output via a display, and input via kb/mouse; or remote connectable via SSH?)

lutooog, 2023-01-06:
Hi,
   Yes, your understanding is correct!
   I try to output  via a display, it can be seen that OS is running! But how to config to output to serial port when PSE_FW is enabled?

stanleyintel, 2023-01-09:
Hi,
The device number of serial port should be changed, so you need to change the "console" settings in your kernel command line. One method is to check the dmesg.

lutooog, 2023-01-10:
Hi，
  Thank you a lot.
   I can see linux log when I change  hange the "console" to ttyS2.

closed 2023-01-11 by lutooog