slimbootloader/slimbootloader #1815: SBL for Tigerlake

Renuka018, 2023-01-25:
Hi,
I built slimbootloader image and stitched it with IFWI for tigerlake board.
However, when I flashed it on the board, nothing is displayed on the screen and the board is turning on and off periodically.
Additionally, can you explain how to disable boot guard on the board?
Thanking in anticipation.

stanleyintel, 2023-01-25:
Hi Renuka,

You need to get the boot log from the UART port to know what's wrong with the booting. 

About boot guard: it is set during stitch (i.e., the '-b' option for StitchIfwi.py, or the "Boot Guard Profile" if using "fit.exe").

Renuka018, 2023-01-26:
Hi Stanley,
Thank you for this help.
I will try -b option in the stitching command.

closed 2023-01-27 by Renuka018