slimbootloader/slimbootloader #1819: UEFI Payload for TGL

Renuka018, 2023-01-31:
Hi,
I am trying to build UEFI payload for TigerLake UP3 Board.
However, I am facing error while integrating it with SBL.
SBL without UEFI Payload is working on the board.
Can you please elaborate steps to build UEFI payload with SBL for TigerLake UP3 Board?
Thanking you in advance.

stanleyintel, 2023-02-01:
Hi Renuka,

Not sure what kind of error you face.

- For integration/compilation error: You can find the detail build steps in Intel doc#: 759399 (TGL MR7 Release Notes). Please get the file from your Intel representative.

- For runtime error, can you elaborate it more?

Renuka018, 2023-02-01:
Hi Stanley,

Referring the same document you mentioned. (759399).
Able to clone edk2 however not sure from where to find uefipldplatform.zip folder.

stanleyintel, 2023-02-01:
Hi Renuka,

The zip file comes with TGL MR7 SBL release (doc#: 614394).

Renuka018, 2023-02-01:
Hi Stanley,

Thank you for your help.
Lastly, can you please help in telling difference between image built with EDK-II and without EDK-II.
That is, is there any parameter in the built image which would specify if it has EDK-II or not?
Thanking you in anticipation.

stanleyintel, 2023-02-02:
Hi,

I am not sure about the question you asked. Let me try to explain the relation between UefiPldPlatform.zip and EDK-II:

Intel UEFI Payload is based on EDK-II. The package (UefiPldPlatform.zip) only provides platform- and project-specific UEFI modules extending EDK-II. That's why you still need to clone EDK-II to Edk2Open folder before building the Payload. For detail, you can check <UefiPldPlatform.zip>/PldPlatform/TglPlatformPkg/PlatformPkg.dsc.

Renuka018, 2023-02-02:
Hi,

Thanking you for your help.
Understood my doubt.
Able to build SBL.
Board is booting but there is no display through HDMI.

Getting the following output through UART.
============= Intel Slim Bootloader STAGE1A =============
SBID: SBL_TGL 
ISVN: 001
IVER: 001.000.001.007.00017
SVER: F877B736DDFEB4FE-dirty
FDBG: BLD(D IA32) FSP(R)
FSPV: ID(TGLI-FSP) REV(0A007D72)
CPUV: ID(806C2) UCODE(0)
Loader global data @ 0xFEF3FD10
Run  STAGE1A @ 0xFFFEA000
Load STAGE1B @ 0xFFD80000

============= Intel Slim Bootloader STAGE1B =============
[Boot Guard] AcmStatus : 0x00000000
[Boot Guard] BootStatus: 0x00000000
[Boot Guard] Boot Guard Failed or is Disabled!
SpiInstance = FEF4C3D0
PchSpiBase at 0x00001F05
ScSpiBar0 at 0xF0000000
Loading Component KEYH:_HS_
Registering container KEYH
HASH verification for usage (0x00000100) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000100): Success
HASH verification for usage (0x00000000) with Hash Alg (0x2): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Append public key hash into store: Success
Load EXT CFG Data @ 0xFEF4CB54:0x0FD8 ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x2): Success
SignType (0x2) SignSize (0x180)  SignHashAlg (0x2)
RSA verification for usage (0x00000200): Success
Calculate Hash for component Type 0x4 as its not available in Component hash table 
Sending EC Command: 2D
SendEcCommandTimeout sent 2D
Sending EC Command: 0D
SendEcCommandTimeout sent D
ReceiveEcDataTimeout Read 9
Receiving EC Data: F8
ReceiveEcDataTimeout Read 9
Receiving EC Data: 01
Processor does not support Boot Guard.
Boot Guard Support status: 0
GetPlatformDataPtr is copied 0xFEF4CB34 
Sending EC Command: 2D
SendEcCommandTimeout sent 2D
EC: Set FailSafe Critical Temp & Fan
Sending EC Command: 39
SendEcCommandTimeout sent 39
Sending EC Data: 55
SendEcDataTimeout sent 55
Sending EC Data: 41
SendEcDataTimeout sent 41
EC: Set fan pwm
Sending EC Command: 81
SendEcCommandTimeout sent 81
Sending EC Data: 44
SendEcDataTimeout sent 44
Sending EC Data: 64
SendEcDataTimeout sent 64
Sending EC Command: 1A
SendEcCommandTimeout sent 1A
EC: Update Pwm
BOOT: BP0 
MODE: 0
BoardID: 0x01
PlatformName: TGLU_DDR
Memory Init
FSPM CfgData assignment
Load Memory Cfg Data
SkipMbpHob = 0x0
Load Graphics Cfg Data
board Id = 1 .....
Call FspMemoryInit ... Success
Loader global data @ 0x78C0FD50
Memory Tolum @ 0x7FC00000
Memory Touum @ 0x480400000
Memory Tom   @ 0x400000000
Switch to memory stack @ 0x78C4FF00
Stage1 stack: 0x40000 (0x26688 used)
Stage1 heap: 0x10000 (0x4360 used, 0x5274 max used)
Call FspTempRamExit ... Success
Memory FSP @ 0x78C50000
Memory TOP @ 0x78750000
Loading Component FLMP:SG02
HASH verification for usage (0x00000002) with Hash Alg (0x2): Success
Loaded STAGE2 @ 0x78A03000

============= Intel Slim Bootloader STAGE2 =============
PCH gen (0x5), series (0x2), stepping (0x20)
LPC DeviceId (0xA082)
Boot-time Pch Dts: 0xF0
Variable region: 0xFF60E000:0x2000
SaveNvsData  Length=0xFAA8
Matched with saved NVS data
Save MRC Training Data (0x78C96118 0x00FAA8) ... Already started
Silicon Init
PchLp is detected
BIOS SPI region will be protected
IAS  SPI region will be protected
Load Silicon Cfg Data
Call FspSiliconInit ...

Can you please explain where am I going wrong?
Thanks and Regards.

stanleyintel, 2023-02-02:
Hi,

I am not sure which the problem you encounter: SBL hung at stage2 or no HDMI output?

If SBL hung at Stage2 - "FspSiliconInit", please provide the boot log with FSP debug messages. Thus, you will need to

1. (optional if your SBL source is not from SblPlatform.zip in doc 614394) replace Silicon/TigerlakePkg/FspBin/FspDbg.bin with the one from <SblPlatform.zip>/SblOpen/Silicon/TigerlakePkg/FspBin.
2. build SBL with the "-fd" option. e.g., python BuildLoader.py build tgl  -p "OsLoader.efi:LLDR:Lz4;UEFIPAYLOAD.fd:UEFI:Lzma" -fd

If no HDMI output, you may need to 
1. customize your VBT based on your board design. Please check doc: 631572 to learn how to customize your VBT file (under Platform/TigerlakeBoardPkg/VbtBin/). In brief, you need to use "DisCon" and Vbt.json to customize Vbt.dat. 
2. (optional) skip TglUDdr4GopVbtSpecificUpdate()  in Platform/TigerlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c, if the special setting does not match your board design.

Renuka018, 2023-02-02:
Hi Stanley,

Thank you for your replies.
1) Tried by replacing the fspbin folder. Got output through UART ending with the following:

-----------------VmdInit Start ----------------
VmdInfoHob->EfiHobGuidType.Name: CCD0306E-7FA1-4DF5-9999-C1F89A1D1BA9
VmdInfoHob @ 78CB0278
&(VmdInfoHob->VmdPortInfo) @ 78CB0290
VmdHobSize - HobHeaderSize: 174
VmdInfoSize: 174
VMD: VmdDetectPcieStorageDevices Start
VMD: Checking device connection on PEG ports 
Read DEVEN register
PegDevenReg = 2000
VMD: Storage device found  on Bus/device/Func 0/6/0  DevId 0xF1A5   Rp Index 1
VMD: Checking device on PCH PCIe ports 
VMD: Checking the availability of SATA controller(0/23/0) for sata devices 
Dumping port info
RpDev 6 RpFunc 0 DevId 0xF1A5 RpIndex 1

VMD Device has been disabled
Initializing Pavp
POSTCODE=<00009A32>
Initializing PAVP
CpuPcieRpConfigureRpfnMapping () Start!
Initializing IPU device
POSTCODE=<00009A14>
IpuInit Begins
IPU is supported by Silicon
Silicon supports IPU and BIOS has IPU config block
Install IPU HOB
IpuDataHob->EfiHobGuidType.Name: 0061DD66-212B-4DAE-9BC0-30E02E3F40FD
IpuDataHob @ 78CB0408
IpuDataHobSize - HobHeader: 1
IpuDataHobSize: 19
IpuConfigHob->EfiHobGuidType.Name: 446268E5-8C30-4E0A-9B28-A3E7F00431D0
IpuConfigHob @ 78CB0408
IpuConfigHobSize - HobHeader: 11
IpuConfigHobSize: 29
IPU HOB installed
Set IPU bit at DEVEN regiser
IpuDataHob->IpuSupport = 1 
IpuConfigHob->IpuAcpiMode = 1 
[TCC] TccInit() - Start
Cache Prefetecher Tuning: 
 IfuTrainEnable: 0
 ConvertItomToRfo: 0
!!! IA32 Exception Type - 0D(#GP - General Protection) CPU Apic ID - 00000000 !!!!
ExceptionData - 00000000
EIP - 78F327A2, CS - 00000010, EFLAGS - 00010046
EAX - 00000000, ECX - 000000CD, EDX - 00000000, EBX - 78C80000
ESP - 78C8F11C, EBP - 78C8F128, ESI - 00000000, EDI - 78CA81E8
DS - 00000008, ES - 00000008, FS - 00000008, GS - 00000008, SS - 00000008
CR0 - 00000033, CR2 - 00000000, CR3 - 00000000, CR4 - 00040640
DR0 - 00000000, DR1 - 00000000, DR2 - 00000000, DR3 - 00000000
DR6 - FFFF0FF0, DR7 - 00000400
GDTR - 78C0FBF0 00000037, IDTR - 78C8FD10 0000010F
LDTR - 00000000, TR - 00000000
FXSAVE_STATE - 78C8EE60
!!! Find image based on IP(0x78F327A2) C:\Source\work7\d3e1f8e507cdf35f\Build\TigerLakeFspPkg\DEBUG_VS2017\IA32\ClientOneSiliconPkg\Product\TigerLake\SiInit\Pei\SiInitFsp\DEBUG\SiInitFsp.pdb (ImageBase=0000000078F04000, EntryPoint=0000000078F044D4) !!!!

Can this be considered as normal output?
Thanks and Regards.

stanleyintel, 2023-02-02:
Hi,

  From the log, it seems that your CPU sku does not support TCC but you enable it. Please set self.ENABLE_TCC= 0 in Platform/TigerlakeBoardPkg/BoardConfig.py. If it is already disabled but SBL still hung, please attach the "full log".

  For enabling display, please resolve the first issue (hung at stage2) first. After that, if HDMI does not work still, please consult with your Intel rep for further support.

Renuka018, 2023-02-02:
Hi,

self.ENABLE_TCC=0 is already set.
PFA the complete log.

[TGL_SBL_EDK-II_UART.docx](https://github.com/slimbootloader/slimbootloader/files/10569596/TGL_SBL_EDK-II_UART.docx)

stanleyintel, 2023-02-02:
Hi,
 Please attach the full log in text file. Other types of file will be blocked by our system.

Renuka018, 2023-02-03:
B7:	-23	26	49	1
W Mc0.C0 GsmMctRDRDdg: 6
W Mc0.C0 GsmMctRDRDdr: 4
W Mc1.C0 GsmMctRDRDdg: 6
W Mc1.C0 GsmMctRDRDdr: 4
MRC task Write Timing Centering -- SUCCEEDED, Status = 0h.
Post Code: DD4Bh

MRC task -- Write Voltage Centering -- Started.
R Mc0.C0 GsmMctRDRDdg: 6
R Mc0.C0 GsmMctRDRDdr: 4
W Mc0.C0 GsmMctRDRDdg: 32 (0x20)
W Mc0.C0 GsmMctRDRDdr: 32 (0x20)
R Mc1.C0 GsmMctRDRDdg: 6
R Mc1.C0 GsmMctRDRDdr: 4
W Mc1.C0 GsmMctRDRDdg: 32 (0x20)
W Mc1.C0 GsmMctRDRDdr: 32 (0x20)
Rank = 0
Mc		0               1               
Channel		0               0               
Byte		0 1 2 3 4 5 6 7 0 1 2 3 4 5 6 7  Error Count

WriteVoltage
  -36  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -35  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -34  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -33  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -32  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -31  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -30  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -29  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -28  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -27  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -26  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -25  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -24  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -23  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -22  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -21  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -20  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -19  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -18  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -17  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -16  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -15  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -14  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -13  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -12  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -11  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
  -10  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -9  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -8  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -7  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -6  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -5  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -4  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -3  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -2  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   -1  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    0  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    1  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    2  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    3  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    4  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    5  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    6  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    7  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    8  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
    9  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
   10  		. . . . . . . . . . . . . . . .  0x0_0	 0x0_0	
[... 269 lines cut ...]
   25  		# # # # # # # # # # # # # # # #  0x17E6_0	 0x17E6_0
   26  		# # # # # # # # # # # # # # # #  0x17EC_0	 0x17EA_0
   27  		# # # # # # # # # # # # # # # #  0x17EC_0	 0x17EB_0
   28  		# # # # # # # # # # # # # # # #  0x17ED_0	 0x17EB_0
   29  		# # # # # # # # # # # # # # # #  0x17ED_0	 0x17ED_0
   30  		# # # # # # # # # # # # # # # #  0x17ED_0	 0x17ED_0
   31  		# # # # # # # # # # # # # # # #  0x17ED_0	 0x17ED_0
Mc0.C0.R0:	Left	Right	Width	Center
       B0:	-26	13	39	-7
       B1:	-26	8	34	-9
       B2:	-30	11	41	-10
       B3:	-29	9	38	-10
       B4:	-26	13	39	-7
       B5:	-28	9	37	-10
       B6:	-27	11	38	-8
       B7:	-30	6	36	-12
Mc1.C0.R0:	Left	Right	Width	Center
       B0:	-27	12	39	-8
       B1:	-29	13	42	-8
       B2:	-30	10	40	-10
       B3:	-30	10	40	-10
       B4:	-25	12	37	-7
       B5:	-28	10	38	-9
       B6:	-29	11	40	-9
       B7:	-30	12	42	-9

stanleyintel, 2023-02-03:
Hi Renuka,

  Your log is not complete and causes some trouble when people read it directly in this page. If possible, you can delete previous direct-paste log.

  Please save the full log into a .txt or .log file, and attach it as a file here. If you use "putty" to capture log, the "putty" program can automatically save log into file by Session - Logging - Select "All session output".
 
The log should start from "============= Intel Slim Bootloader STAGE1A ============="

Renuka018, 2023-02-03:
Hi Stanley,

I tried attaching .txt file but that extension is not supported.

stanleyintel, 2023-02-03:
Hi Renuka,

   I don't know why, but I saw people in other "Issues" can attach their .txt files.
   If you still have trouble in attaching .txt file, please mail me the .txt file directly. You can find my email by using "git log" in this gihub SBL project.

Renuka018, 2023-02-03:
Hi Stanley,

PFA the log text file.
[TGL_SBL_EDK-II_UART.txt](https://github.com/slimbootloader/slimbootloader/files/10575996/TGL_SBL_EDK-II_UART.txt)

stanleyintel, 2023-02-03:
Hi Renuka,

  Unfortunately, your CPU is "TGL UP3 Refresh," still not supported by Intel SBL release. There are many uncertainties if you want to try it. I can do my best to help you to enable it, but cannot guarantee it works within your project/product deadline.

   If you still want to try it, you will need to work tightly with your Intel rep. The first thing is to get the microcode (000806c2) for TGL UP3 Refresh. SBL uses the binary file as Linux does, so your Intel rep needs to use a tool to covert the *.inc to *.pdb for you. Then, you can add the filename into Silicon/TigerlakePkg/Microcode/Microcode.inf and put the Microcode file under Silicon/TigerlakePkg/Microcode. For further discussion, I suggest moving to https://groups.io/g/slimbootloader/topics and close the issue here.

Renuka018, 2023-02-03:
Hi Stanley,

Thank you for your time and help.
I will move my query to the discussion platform you suggested.
Thanks and Regards.

closed 2023-02-03 by Renuka018