slimbootloader/slimbootloader #1838: [EHL] Elkhart Lake PSETSN SGMII 2.5G fail

sharkblue2009, 2023-03-07:
We are using Slimbootloader with Intel FSP MR5,  the PseTSN1 is using SGMII to connect a switch ASIC and we want the speed to 2.5G, the FSP setting "PseTsnGbePhyInterfaceType[1]" is set to 0x3 which means SGMII+,  but we only get the SGMII bus running on 1G, and the PCI_DeviceID for PseTsn1 remains 0x4bb1 (We should get the device ID is 0x4bb2 if the SGMII is running in 2.5G mode).

We also set the “PseTsnGbeLinkSpeed[1]” to 0x2 for 38.4Mhz 2.5Gbps link speed, but seems its not affect the SGMII. The device ID still 0x4bb1.

As PseTsn0 is using RGMII mode and PchTsn is disabled in our use case, and the Serdes Lane 9 is used for PseTsn1 SGMII, if there are other constrains to get the 2.5G SGMII work.

stanleyintel, 2023-03-07:
Hi,

Have you set PseTsnGbeSgmiiEnable[1] = 1?

In addition to the above setting, please also try EHL MR6 (just released, doc id: #616714) with the new FSP UPD: TsnPllConflictPrevention = 1.

stanleyintel, 2023-03-07:
Add: 

   After enabling TsnPllConflictPrevention, please build SBL with FSP debug binary (i.e., add "-fd" option when executing BuildLoader.py) and check any abnormal in the boot log around the log block: "TsnPllLaneConflictPrevention() Start" to "TsnPllLaneConflictPrevention() End".

sharkblue2009, 2023-03-07:
Hi Stanley,

Yes, I had set PseTsnGbeSgmiiEnable[1] = 1, and the SGMII is working, but it's in 1GB mode rather than 2.5G, the device ID is 0x4bb1.

I only had MR4 debug binary,  from the log I did not find the text "TsnPllConflictPrevention"

Partial logs:
============================
------------------ TSN Sub Region Config ------------------
 Tsn Mac Address Sub Region   - Base Address = 0
 Tsn Mac Address Sub Region   - Size = 0
 Pse Tsn Ip Config Sub Region - Base Address = 0
 Pse Tsn Ip Config Sub Region - Size = 0
 Tsn Config Sub Region        - Base Address = 0
 Tsn Config Sub Region        - Size = 0
 PchTsnEnable = 0
 PchTsnGbeMultiVcEnable    = 0
 PseTsnGbeMultiVcEnable[0] = 0
 PseTsnGbeMultiVcEnable[1] = 0
 PchTsnGbeLinkSpeed = 3
 PseTsnGbeLinkSpeed[0] = 2
 PseTsnGbeLinkSpeed[1] = 2
 PchTsnGbeSgmiiEnable = 1
 PseTsnGbeSgmiiEnable[0] = 0
 PseTsnGbeSgmiiEnable[1] = 1
 PseTsnGbePhyInterfaceType[0] = 1
 PseTsnGbePhyInterfaceType[1] = 3
 PseGbeDllOverride[0] = 0
 PseGbeTxDelay[0] = 10
 PseGbeDllOverride[1] = 0
 PseGbeTxDelay[1] = 10

==============================

PseTsnBiosProg () End
PSE TSN Device: Device Id: 0x4BB1, Vendor Id: 0x8086
PseTsnBiosProg () for Device 29, Function 2 Start
PseTsnBase: 0xEA000
PSE IO 0xEA000: PME_CTRL State : 0x0
PseTsnBar0: 0xFE600000
PSE TSN owned by host, start PSE TSN DMA VC/TC programming
TSN MDIO Read PhyAdd: 0x15, RegAdd: 0x0 , Value: 0x826, Raw: 0x2A04B0D
TSN MDIO Write PhyAdd: 0x15, RegAdd: 0x0, Value : 0x827, Raw: 0x2A04B05
PSE IO 0xEA000: PME_CTRL State : 0x0
PseTsnBiosProg () End
Tsn ModPhy Prog: Configure Lane 9, LinkSpeed 2
Tsn ModPhy Prog: TableEntry 5
Tsn ModPhy Prog: PortId: 0xAA, Offset: 0x8180, Before write value: 0x00000026, After Write Value: 0x58555551
Tsn ModPhy Prog: PortId: 0xAA, Offset: 0x8188, Before write value: 0x00000000, After Write Value: 0x0000012D

stanleyintel, 2023-03-07:
Hi,

1. The log "Tsn ModPhy Prog: Configure Lane 9, LinkSpeed 2" shows it is to be configured as 2.5G 34Mhz.
2. Can you find something like "MAC Phy Interface Type" in the log? 
3. TsnPllConflictPrevention is new in EHL MR6, which may help you to clarify this issue. If possible, please get MR6 from your Intel rep. It is always good to use newer MR release, especially there might have FSP UPD alignment issue between "code" and "FSP binary" in use, not to mention bugs fixed in newer release.

sharkblue2009, 2023-03-07:
I had just downloaed the EHL MR6 package, will try as you mentioned.

for the ""MAC Phy Interface Type" log:
-----------------------------------------------------
Is BIOS the verifier? : 1
        PSE FwSize: 5A364
PSE GBE0 is Enabled. Load TSN GBE PHY IF data to IMR
PSE GBE0 MAC Phy Interface Type 1
PSE GBE1 is Enabled. Load TSN GBE PHY IF data to IMR
PSE GBE1 MAC Phy Interface Type 3
        After TSN data appended: PSE FwSize: 5AD74, FwSizeInKb: 16C
        IMR10 at 0x7A000000 has size 0x00400000 in b, 0x1000 in Kb
        IMR10 offset 0x7A40 is 0x1E8000
        IMR10 offset 0x7A44 is 0x9FFFF000
        IMR10 offset 0x7A48 is 0x1010200
        IMR10 offset 0x7A50 is 0x5000200
        IMR10 offset 0x7A58 is 0x5000200
Loading firmware through IMR.
        Copied from FwBuffer 0x78AA4000 to IMR address 0x7A000000 with size 0x5AD74

 (DDR) Pse zephyr firmware dump: from 78AA4000

stanleyintel, 2023-03-07:
Thanks. 
Above logs from MR4 shows PSE GBE1 is programmed as 2.5G mode.
Wait for newer logs from MR6. If possible, please attach a txt file containing either logs between "TsnInit() Start" and "TsnInit() End", or the full SBL boot log.

sharkblue2009, 2023-03-08:
Thanks, 

Please check attach file for SBL MR6 debug log, seems the SGMII remains 1GB.
[cup_mr6_r01_20230308_095406.txt](https://github.com/slimbootloader/slimbootloader/files/10915812/cup_mr6_r01_20230308_095406.txt)

stanleyintel, 2023-03-08:
Hi,

I have checked the log and confirmed that most settings (used by TsnInit) are correct, except the TSN-related sub regions are zero (e.g., Pse Tsn Ip Config Sub Region). However, the empty sub regions should not cause the issue you encounter.

I am trying to clarify something with our GbE experts and will get back to you once I got the reply.

In the meanwhile, may you help
1. check the GPIO configuration (mEhlPseTsn1DeviceGpioTable in Platform/ElkhartlakeBoardPkg/Library/Stage2BoardInitLib/GpioTableEhl.h) - must match your board design
2. how do you stitch SBL IFWI? Manually execute fit.exe, or by executing the python command described in section 7.4.2.2 in release note (doc: #766113)? For later case, I believe you already modified the 'tsn' related settings in Platform/ElkhartlakeBoardPkg/Script/StitchIfwiConfig.py

sharkblue2009, 2023-03-09:
Thanks,

1. I had checked "mEhlPseTsn1DeviceGpioTable" and seems to be used for RGMII mode so I think it may not affact our case which is using SGMII. and previous SBL MR5 code does not have this table.
2. we are using fit.exe to generate the SBL IFWI, the related setting(Flex I/O and Networking) is changed as the board design. the CSE binary we using is MR5.

stanleyintel, 2023-03-09:
Hi,

One thing you may try is to load the UNDI device driver in UEFI Payload, and then check the pci DID again.
Knowing that you are using OS Loader, so you may need:
1. Build a UEFI Payload, e.g., python BuildPayload.py build ehl  --userphysicalpresence -r, and says the executable is UEFIPAYLOAD.fd
2. Copy UEFIPAYLOAD.fd to (SBL)/Platform/ElkhartlakeBoardPkg/Binaries
3. Set GEN_CFG_DATA.PayloadId | 'UEFI'  in your board DLT file
4. Build SBL with UEFI Payload, e.g., python BuildLoader.py build ehl -p "OsLoader.efi:LLDR:Lz4;UEFIPAYLOAD.fd:UEFI:Lzma"
5. Stitch the SBL IFWI, flash it, and boot

6. Download EHL UNDI driver from: doc # 733646
7. Unzip it to get "IntelgbeUndiDxe.efi", and save it in a USB pen drive.
8. Under "UEFI Shell", execute: "load IntelgbeUndiDxe.efi"
9. Check the PCI DID again

stanleyintel, 2023-03-09:
Correct step 6: the UNDI driver for EHL MR6 should be in doc # 768031

stanleyintel, 2023-03-09:
Thanks for the update. Please work with your Intel rep to file an IPS (Intel® Premier Support (IPS) ticket and get help from our ethernet expert.

Note that if everything works after our ethernet expert figures out the problem, the 2.5G mode can only be shown with the UNDI driver loaded. Because Intel UNDI driver comes with EFI executable only, eventually you need to integrate the IntelgbeUndiDxe.efi into UEFI Payload and boot with UEFI Payload. An example is to add the following code snippet in <UEFI Payload Src>/PldPlatform/EhlPlatformPkg/PlatformPkg.fdf (which equivalent to "load IntelgbeUndiDxe.efi")

FILE DRIVER = 720E6687-43C4-84EA-E1B5-23AAED359840 {
    SECTION PE32 = EhlPlatformPkg/Undi/IntelgbeUndiDxe.efi
    SECTION UI = "IntelGbeUndi"
  }

sharkblue2009, 2023-03-15:
If the issue related the PMC? shall we try to disable the PMC?

stanleyintel, 2023-03-15:
Hi,

I didn't see a clear relation between your issue and PMC.

Maybe you can try with my test configuration:
1. UEFI Payload + IntelgbeUndiDxe.efi  (integrated with the method above)
2. self.ENABLE_TSN = 1, in Platform/ElkhartlakeBoardPkg/BoardConfig.py
3. PseTsnGbe1SgmiiEnable = 1, in Platform/ElkhartlakeBoardPkg/CfgData/CfgData_Silicon.yaml
4. Fspscfg->PseTsnGbeLinkSpeed[1] = 2, in Platform/ElkhartlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c
5. Override the "tsn" stitch option, in Platform/ElkhartlakeBoardPkg/Script/StitchIfwiConfig.py
+            ('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/SgmiPse1',  'Lane 9'),
+            ('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/Psegbe1PhyIfMode',  'SGMII'),
+            ('./FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont3Conf',                 'Disabled'),
   Also, comment out the "FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont3Conf" in common stitch option list 

6. DLT file: to boot with UEFI Payload
GEN_CFG_DATA.PayloadId | 'UEFI'

7. Build SBL: must with UEFI Payload
8. Stitch:
python Platform/ElkhartlakeBoardPkg/Script/StitchIfwi.py -b legacy -s Outputs/ehl/Stitch_Components.zip -c Platform/ElkhartlakeBoardPkg/Script/StitchIfwiConfig.py -w ../<ehl_fw_ingredients>/ -p ehl -k ../<SblKeys> -o tsn

stanleyintel, 2023-03-15:
Last step: ".... -k ../" should be "-k $sblkey_path -o tsn".   Not sure why some string is missing...

stanleyintel, 2023-03-15:
Also, PseTsnGbe1PhyInterfaceType = 3 in PseTsnGbe1PhyInterfaceType.

sharkblue2009, 2023-03-24:
Hi Stanley,

By help from Intel expert, the issue was solved,

The device ID of PSE TSN1 can be dynamic swtiched from 0x4BB1 to 0x4BB2 when loading the "IntelgbeUndiDxe.efi" driver, but the original code of IntelgbeUndiDxe.efi does not support our PHY, so by default it will not init the SGMII serdes lane rightly.

The device ID switch is happend when "intelgbe_modphy_init" called if the PHY is identified to support 2.5G SGMII mode.

closed 2023-03-24 by sharkblue2009