slimbootloader/slimbootloader #1842: [EHL] "support auto negotiation for Intel Gbe“ cause network(PseTsn) fail in ubuntu

sharkblue2009, 2023-03-08:
I had update the SBL code to MR6, but it cause the network fail in ubuntu(kernel version:  linux-intel-iotg_5.15.0-1023.28), from the dmesg log, the ethernet continues fail at DMA engine initialize.

the SBL code was working fine for networks before the commit "support auto negotiation for Intel Gbe“(commit ID: f877b736ddfeb4fef82b937e889488f9fc4da510).  seems  the commit affact the network.

dmesg log:
=======================================
[   22.007921] intel-eth-pci 0000:00:1d.1 enp0s29f1: PHY [stmmac-2:01] driver [Marvell 88E1510] (irq=POLL)
[   22.008637] intel-eth-pci 0000:00:1d.1 enp0s29f1: Register MEM_TYPE_PAGE_POOL RxQ-0
[   22.009177] intel-eth-pci 0000:00:1d.1 enp0s29f1: Register MEM_TYPE_PAGE_POOL RxQ-1
[   22.009681] intel-eth-pci 0000:00:1d.1 enp0s29f1: Register MEM_TYPE_PAGE_POOL RxQ-2
[   22.010264] intel-eth-pci 0000:00:1d.1 enp0s29f1: Register MEM_TYPE_PAGE_POOL RxQ-3
[   22.010909] intel-eth-pci 0000:00:1d.1 enp0s29f1: Register MEM_TYPE_PAGE_POOL RxQ-4
[   22.011468] intel-eth-pci 0000:00:1d.1 enp0s29f1: Register MEM_TYPE_PAGE_POOL RxQ-5
[   22.011998] intel-eth-pci 0000:00:1d.1 enp0s29f1: Register MEM_TYPE_PAGE_POOL RxQ-6
[   22.012564] intel-eth-pci 0000:00:1d.1 enp0s29f1: Register MEM_TYPE_PAGE_POOL RxQ-7
[   23.014135] intel-eth-pci 0000:00:1d.1: Failed to reset the dma
[   23.014153] intel-eth-pci 0000:00:1d.1 enp0s29f1: stmmac_hw_setup: DMA engine initialization failed
[   23.014159] intel-eth-pci 0000:00:1d.1 enp0s29f1: stmmac_open: Hw setup failed

stanleyintel, 2023-03-08:
Hi,
EHL MR6 supports kernel v5.15.71 (doc # 732903). I noticed there are many changes related to stmmac kernel driver, and your kernel might have no these patches included. Upgrade kernel is one method to solve the problem. Please check doc # 732903 for detail.

If kernel is not upgradable, fortunately, we may workaround it. The simplest method is to revert the commit on your side. Or, may you help to check if the error is caused by new ACPI code (Platform/ElkhartlakeBoardPkg/AcpiTables/Dsdt/PchPse.asl) or new GPIO config (mEhlPseTsn0DeviceGpioTable (PSE TSN0) in Platform/ElkhartlakeBoardPkg/Library/Stage2BoardInitLib/GpioTableEhl.h) ?

sharkblue2009, 2023-03-10:
Hi,

I found this issue caused by other IO setting(SILICON_CFG_DATA.PchPseUartEnable[2] = 0x2 will cause the fail but = 0x1 is ok), so please close the issue.

stanleyintel, 2023-03-10:
Great! Thanks for the update.

stanleyintel, 2023-03-10:
Hi,

It is possible to set PchPseUartEnable[2] owned by host. However, PSE UART2, by default, is used in two places:
1. PSE log console: the default value is 3, which means PSE UART2. It can be changed or disabled by configuring PchPseLogOutputChannel in Platform/ElkhartlakeBoardPkg/CfgData/CfgData_Silicon.yaml (or DLT file).
2. Some PSE services use "console" or "shell-uart", while the default settings are to use uart2. You will need to change the default Device Tree settings (in PSE source) and then rebuild PSE FW.

If setting PchPseUartEnable[2] =2 without the above changes, PSE FW may block at unexpected point. As the result, Linux driver using PSE will experience timeout.

gdong1, 2023-03-13:
Great to see the issue was solved. so just close it.

closed 2023-03-13 by gdong1