slimbootloader/slimbootloader #1856: The industrial computer is using slimbootloader, but there is no display.

victor0726, 2023-03-29:
I am a newbie in the customization of BIOS and FW.

I purchased an industrial computer (not UP Squared board) with an Intel Celeron N3350 processor, DDR3L 4G 1866MHz (204-pin SODIMM slot) memory, SATA 250G hard drive, and 4 sets of serial ports (RS-232/422/485). It has Windows 10 IoT Ent2019 installed.

I want to learn slimbootloader, so I used the Leaf Hill-0.71-Firmware Images provided by Intel and followed the instructions on https://slimbootloader.github.io/supported-hardware/apollo-lake-crb.html and used LEAFHILD.X64.0071.D01.1809040132.bin for stitching.

However, after flashing the BIOS, the display did not turn on.

1、My board has 4 sets of COM ports. How can I configure them to have console access?

2、Currently, I do not know why the screen is not displayed after booting.

3、Do I need to modify any part of the YAML file?

PS. I would be very grateful if you could guide me through the process.

testriscv, 2023-03-29:
How did you flash the bios ? can you restore the original one ?

victor0726, 2023-03-30:
> How did you flash the bios ? can you restore the original one ?

Hi testriscv,

I used the CHA341A programmer to flash the BIOS. The industrial computer uses AMI BIOS and I have kept the original BIOS version, so it can be restored to the original state. I want to learn how to port a new board using Slimbootloader instead of AMI. Do you need the original BIOS version? The original BIOS used a W25Q128JWSQ chip. Thank you for your help

stanleyintel, 2023-03-30:
Hi Victor,

Assume you use the latest SBL source from github.

1. About serial ports (COM ports), please find out which port connects to LPSS_UART2 (pin name: J41, H41, L41, M41, or GPIO number - GPIO_46~49). With the default configuration, SBL uses UART2 as its debug console on APL. 

2. Before the display issue, you may need to check which "board id" (or platform id) the SBL detects. Different board id has different platform configuration delta file (i.e., DLT file). E.g., 7 is for LeafHill and 8 is for Up2. You can find "PLATFORMID_CFG_DATA.PlatformId" in many DLT files under Platform/ApollolakeBoardPkg/CfgData/. Then, different DLT file decides which VBT file is loaded, and the VBT (video bios table) controls how your display is connected/initialized by SOC. You can set/check board id by,

- Method 1: specify the board id by adding the "-p" option in executing StitchLoader.py:
e.g., StitchLoader.py -p 0xAA000208 ...(other options), which will set board id = 8
More discussion about the -p option for APL can be found in https://groups.io/g/slimbootloader/topic/74665746

- Method 2. if you can get debug logs from serial port, you can know the board id from boot log. The board id info is shown in Stage1B, and it looks like:   BoardID: <value>
 
- Method 3: GPIO configuration: APL SBL code calculates the Board ID from  GPIO_224 | GPIO_213 << 1 | GPIO_223 << 2. Refer to [here](https://github.com/slimbootloader/slimbootloader/blob/04e872d77b25fe42c3b9ffd18aeb93c9785a4577/Platform/ApollolakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c#L248) for implementation detail.

3. About YAML: you need to modify YAML or DLT file to align with your HW (board design) and SW (features). Instead of directly modifying YAML, we suggest you modify DLT file. The relation between YAML and DLT is similar to the parent class and child class in Java/C++. Because board id decides which DLT configuration is active, you must know the board id before modifying a DLT file.

victor0726, 2023-03-30:
> 1. LPSS_UART2

Hi stanley,

I'm using an IPC that is not a LeafHill or Up2 board, and it originally used an AMI BIOS. I don't have a schematic for this computer, so I don't know how it's connected to serial port( IPC has 4 serial ports ).

I want to try using slimbootloader on this IPC , even though it may require changing many settings.
The IPC's specifications are: Intel Celeron N3350, DDR3L 4G 1600MHz (204-pin SODIMM slot), VGA, and 4 serial ports (RS-232/422/485).

1、The Intel website only provides Leaf Hill-0.71-Firmware Images and UP Squared-0.71-Firmware Images. Which one should I choose?

2、"How do I modify the YAML for DDR3L 4G 1600MHz (204-pin SODIMM slot)? What are the items that need to be modified?"

PS: I hope to have the serial port output on the screen before proceeding to the next step.

Thank you.

stanleyintel, 2023-03-30:
Hi Victor,

It is difficult to power on a board without serial console and boot log. You may use try&error to find the correct serial port. If the RS-232 comes from SOC directly, possibly it is the LPSS_UART2. Anyway, for each port, try the configuration: 115200 baud (8N1, with no flow control). 

1. About Stitching.. oops... I unexpectedly deleted the answer in my previous post. The answer is: Please keep AMI BIOS, and stitch the SBL image with your AMI BIOS IFWI. No need to use Leaf Hill image.

2. Let's enable serial port first. After that, you can know which DLT file to modify and then, you can read https://slimbootloader.github.io/how-tos/configure-memory-down.html to learn how to configure your DDR3L.

victor0726, 2023-03-30:
Hi stanley,
I stitch the SBL image with AMI BIOS IFWI and  used Tera Term to configure the settings to 115200 baud (8N1, with no flow control) and tested all four serial ports, but  it doesn't prints on serial port.. However, I discovered that the RS-232 do not come directly from the SOC, but instead go through a south bridge chip (SCH3106). In this case, what should I do to enable serial port output?"

[photo.zip](https://github.com/slimbootloader/slimbootloader/files/11109418/photo.zip)

stanleyintel, 2023-03-30:
Hi Victor,

Unfortunately, there has no efficient and correct method to enable serial port without schematic. The RS232 thru SCH3106 might be designed for other propose. In addition, even you can resolve the serial port issue, you still need schematic to configure GPIO pads and even display (VGA).

I can only give you few directions. Hope others in this forum can help you more. By the way, you may also get some help from https://groups.io/g/slimbootloader/messages.

My feedback:
1. On APL, we suggest to use LPSS_UART2 as the debut output port. (see doc # 558402, which you can get from your Intel rep). Thus, the board designer might already reserve UART2. However, in a production system, it is possible to leave only test points for UART2 pins. If so, you will need to rework the board to enable it. 
2. APL FSP, without customization, does not support serial port over Super I/O. Same for SBL.
3. If you are sure the RS232/SCH3106 was designed for SOC debug, you need to customize SBL. The effort roughly include: (a) disable UART GPIO Init in [Stage1A](https://github.com/slimbootloader/slimbootloader/blob/04e872d77b25fe42c3b9ffd18aeb93c9785a4577/Platform/ApollolakeBoardPkg/Library/Stage1ABoardInitLib/Stage1ABoardInitLib.c#L75); (b) enable super I/O (i.e., IOE.SE bit in LPC (BDF: 0:31:0)); and (c) rewrite the whole [SerialPortLib.c](https://github.com/slimbootloader/slimbootloader/blob/04e872d77b25fe42c3b9ffd18aeb93c9785a4577/Silicon/ApollolakePkg/Library/SerialPortLib/SerialPortLib.c) to use the I/O Port 2E/2F to communicate with SCH3106 (you need detail spec from SCH3106 vendor to know how to route serial msg to its serial port).

victor0726, 2023-03-31:
Hi stanley,

1、I found that my Architecture is IA32 when using BuildLoader.py. How can I change it to x64?(the 
[image]
attached file.)

2、Does "stitch the SBL image with your AMI BIOS IFWI" mean that if we want to use Slim Bootloader on our self-manufactured IPC, we must use the CPU BIOS firmware provided by Intel?

3、My company wants to design an Intel-based IPC on our own. Besides specifying the specifications, where should we start? Can you give me some advice or steps?

Thank you

stanleyintel, 2023-03-31:
Hi Victor,

Let me try to answer the 2 questions with below info:
1. IFWI: Integrated Firmware Image (or sometimes, Intel fw image) is the image that you can use DediProg to flash the SPI chip on your board.
2. IFWI contains many firmware ingredients and configurations. Many ingredients are released in binary form by Intel, e.g., CSME and PMC. Some ingredients are prepared by customers, e.g., SBL and PSE. In addition to ingredients, many platform settings are configured by customers during stitching (e.g., a 4-lane PCIe should be 4x1, 2x2 PCIe, or 1x4?) 
3. The "stitching" is to compose an IFWI with given ingredients and user configuration.
4. We (Intel) periodically release BKC (best known configuration) to tell customers what the ingredients should be used and where to download them.
5. The above info seems to be complex, but don't worry, we already summarize BKC and stitching steps/commands in each SBL release notes. For POC, you can simply follow the release notes to build your own IFWI. Please download and study a SBL release note to evaluate the complexity first. For example, doc # 766113 is the release notes for EHL (Elkhartlake).

Next, about stitching:
1. In evaluation phase, developers may use StitchLoader.py to stitch SBL image with an existing IFWI.  The "StitchLoader.py" only updates the BIOS region with SBL image. Thus, all other fw ingredients and configuration in the output IFWI are the same as its input IFWI. So it can simplify the stitching task.
2. However, when you want to customize your IFWI to enable more features, (e.g., enable "boot guard" protection). the StitchLoader.py cannot help.
4. In development phase, developers usually use Fit.exe (or SBL built in StitchIfwi.py) to compose an IFWI. The detail can be found in release notes and related collateral.

Now, back to SBL:
1. SBL is a bootloader that initializes hardware and boots to OS loader or kernel. The role of SBL is the same as UEFI BIOS. That's why it is stored in the BIOS region of IFWI.
2. SBL calls Intel [FSP](https://github.com/intel/FSP/) to initialize many Intel platform features (CPU, PCH, Memory controller, ...). The FSP is also released in binary form from Intel. E.g., for APL, you can find them under Silicon/ApollolakePkg/FspBin.
3. However, a product (says your IPC) is composed of not only Intel platform but also 3rd party components/peripherals, e.g., the SuperIO SCH3106,. This is what you need to customize SBL to enable. Similarly, you also need to change some SBL settings to turn on/off some platform features (e.g., Intel VT-x for virtualization).
4. The customization can be done by your company and/or SBL partners. Please check doc: #730563 to find SBL partners. Besides, you also need help from device vendors (e.g., device-specific init sequence, or even UEFI/ACPI reference code)
5. In community ([groups.io](https://groups.io/g/slimbootloader/topics) and here), we help answer generic questions and resolve common issues. If a question requires further check (e.g., schematics) and/or domain expert (e.g., Intel GbE), we will suggest you file an IPS (Intel® Premier Support) ticket.

Last, about your questions:
1. "stitching SBL with your AMI BIOS IFWI" - you can get the answer from the "stitching" part above.
2. Current SBL supports TGL, EHL, ADL, RPL-P and some sku of 4th gen Xen, and many others You may try to (1) get an RVP or CRB from your Intel rep, (2) download the corresponding SBL release notes, (3) follow the release notes to build and try it, (4) add 3rd party devices and enable them, and (5) if required, approach SBL partners to check what they can help. 
3. Some doc/spec might be helpful:

- ACPI Spec: https://uefi.org/specs/ACPI/6.5/ : if your product runs either Linux or Microsoft Windows
- UEFI Spec: https://uefi.org/specs/UEFI/2.10/ : if running Microsoft Windows

victor0726, 2023-04-06:
Hi stanley,

I would like to use the CRB provided by Intel for efficient development and assistance requests in the future.

1、How to obtain RVP or CRB from Intel representatives? I'm sorry, I really have no idea.
2、What is RVP?
3、Will the CRB provided by Intel come with a  (schematics )?
4、Where can I find the documents #766113 and #730563 mentioned above?

Thank you

stanleyintel, 2023-04-06:
Hi Victor,

   You can get all the info from your Intel rep (e.g., Intel FAE). In particular, 
1. the schematics of RVP and CRB are available.
2. when you check with your Intel rep, please tell him/her that your project is to use NEX (Network Edge Group) CPU/PCH sku. It is because SBL mainly supports NEX platforms.

victor0726, 2023-04-12:
Hi stanley,

It is difficult for a developer who is new to slimbootloader to modify parameters one by one to fit Intel Elkhart Lake CRB when opening Elk's YAML using Config Editor, as there are many options and settings. Is there any tutorial specifically for Elk's YAML settings?  I want to learn all the configuration options. Can you provide a tutorial or guide specifically for configuring YAML for Intel Elkhart Lake CRB?  thanks.

stanleyintel, 2023-04-12:
Hi, 

  I recommend you download and read the following docs first:
    1. Elkhart Lake - Customer Reference Board (CRB) - Getting Started Guide (RDC# 615860): the first doc you should check. It lists many useful docs and as its name - it tells you how to get started for EHL CRB.
    2. Elkhart Lake - CRB User Guide (RDC# 615859) : it describes the CRB layout and many other things.
    3. Elkhart Lake - Message of the Week (MOW): it collects ALL docs related to EHL. Save it for further reference.
   
   Then, to develop SBL, please download EHL MR6, including its Release Notes (RDC# 766113) and Source code (RDC# 616714). Please note, all the doc # are subject to change, but you can always find the latest doc # in MOW.

   The default values set by SBL should be good for most design in power-on phase. Indeed, you may need to fine tune some parameters to optimize SBL for your product. The FSP header files (EHL [FSP-m](https://github.com/intel/FSP/blob/master/ElkhartLakeFspBinPkg/Include/FspmUpd.h) and [FSP-s](https://github.com/intel/FSP/blob/master/ElkhartLakeFspBinPkg/Include/FspsUpd.h)) give a relatively clear description of most parameters. With these description, you can find their corresponding code/YAML in SBL source. However, not all FSP parameters are explicitly set by SBL. If a FSP parameter is not used by SBL, the default values (shown in [bsf](https://github.com/intel/FSP/blob/master/ElkhartLakeFspBinPkg/FspBin/FSP.bsf) will be used. If you want to change it, since SBL is open source, you can easily modify source to add what you need.

   Last, if possible, you may consider close current issue and post further question(s) in https://groups.io/g/slimbootloader/topics. (a github issue is good for bug report, and groups.io is for generic discussion)

closed 2023-05-23 by rprangar