slimbootloader/slimbootloader #1862: 【EHL】How to config GPP_H11 to PWM mode?

lutooog, 2023-04-04:
HI,
      How to config  GPP_H11 or GPP_E17 to output a pwm square wave?
Thank you!

stanleyintel, 2023-04-06:
Hi,
The control is related to the following FSP-s UPDs:

- PchPsePwmEnable
- PchPsePwmPinEnable: [15] is for GPP_H11 and [11] is for GPP_E17
- PchPsePwmPinMux, in [15] = GPIO_VER3_MUXING_PSE_PWM_15_GPP_H11 is for GPP_H11, and [11] = GPIO_VER3_MUXING_PSE_PWM_11_GPP_E17 is for GPP_E17

stanleyintel, 2023-04-06:
Add:
1. When PSE ECLite is enabled, PSE_PWM15 is used by default PSE FW for fan control. Thus, you will need to either disable eclite or modify your PSE FW to use another PWM. Related controls: FSP-s UPD PchPseEcliteEnabled and PSE FW. (i am not familiar with PSE FW, but guess it is controlled [here](https://github.com/intel/pse-fw/blob/main/apps/capabilities/pse_eclite_oob/prj.conf#L69)
2. GPP_E17 might be configured as timed gpio pin (PSE_TGPIO46). Thus, you may also need to check  PchPseTimedGpioPinEnable[46] and related UPDs.

closed 2023-05-15 by lutooog