slimbootloader/slimbootloader #1877: [EHL] stuck Firmware update Elkhart Lake on our design board

Chrysler2016, 2023-04-20:
Hi, community
        I'm trying to use the firmware update function of SLB, but it got stuck during updating.
        
       Our image is based on MR6, with UEFI payload. Here is how I build the image in Linux:
       python3 BuildLoader.py clean
       python3 BuildLoader.py build ehl -p "OsLoader.efi:LLDR:Lz4;UEFIPAYLOAD.fd:UEFI:Lzma"
       I copy Outputs/ehl/SlimBootloader.bin out to windows.
       
       Then I use FIT.exe to update BIOS bin with SlimBootloader.bin to generate a IFWI image. Use Dediprogramer to flash to our board. It works fine. 
       After this, I try to generate a Capsule image with following command:
       python3 ./BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS Outputs/ehl/SlimBootloader.bin -k FirmwareUpdateTestKey_Priv_RSA3072.pem -o FwuImage.bin
       So the SlimBootlaoder.bin is the same as the one in IFWI image.

       I copy the FwuImage.bin to a FAT32 USB disk and connect to our board. In ubuntu, I triggered firware update by:
       echo 1 > /sys/bus/wmi/devices/44FADEB1-B204-40F2-8581-394BBDC1B651/firmware_update_request
       reboot
I got the log as update_step1.txt       
[update_step1.txt](https://github.com/slimbootloader/slimbootloader/files/11284518/update_step1.txt)
line 4 is some debug log I added to check the boot partition, you can ignore it.
I see the same log at https://github.com/slimbootloader/slimbootloader/issues/1339, but not knowing how to solve it.

        Now matter how many times I reboot, power off/on, there is no output or display. 
        Then I remove the on board battery, and power on again, there is output in serial. See update_step2.txt
        
        [update_step2.txt](https://github.com/slimbootloader/slimbootloader/files/11284607/update_step2.txt)
        I checked code of update, found it might relate to the boot partition. If I remove the battery, then it should boot from the primary partition. 

I also tried https://github.com/slimbootloader/slimbootloader/issues/1593, decompress IFWI image, get the bios.bin, then use this one the generate the Capsule image, also get the same result, stuck. 

Hope some one can give me some ideas on how to debug this issue. Thanks in advance.

Chrysler

vincentchchen, 2023-04-21:
Hi Chrysler,

According to your log, the Boot Guard is disabled, so let's exclude the signing key issue. (#1593)
- [Boot Guard] AcmStatus : 0x00000000
- [Boot Guard] BootStatus: 0x00000000
- [Boot Guard] Boot Guard Failed or is Disabled!

Then, we found that you generated the IFWI by using FIT tool, instead of using stitch script.
https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ElkhartlakeBoardPkg/Script/StitchIfwi.py
In case of missing any critical configuration, would you try generating the IFWI with stitch script to see if the fwupdate is working?
The steps are described in the section 7 of EHL MR6 release notes (RDC #766113)
Please remember to disable Boot Guard by using "-b legacy" option when calling StitchIfwi.py

Also, as you mentioned about the boot partition and some hint from #1339, you could also check if the top swap configuration is correct.
For EHL, it should be 512K by default.
https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ElkhartlakeBoardPkg/Script/StitchIfwiConfig.py#L210
In FIT tool, you could find it through "Flash Setting" -> "Top Swap Block Size"

Chrysler2016, 2023-04-21:
Hi, vincentchchen

Thanks for your reply. Follow your instructions, now the update procedure goes smoothly. 

The cause for this issues is the "Top Swap Block Size".  I checked the size in FIT, it was 64KB. So I changed it to 512KB as you mentioned and made a build. After Dedi program into flash, it boot successfully. Then I trigger a firmware update in Linux. It can switch to Backup partition. I can see logs during the update process: (the log I added manually after stage 1A header print)

        ------------set current boot partition: backup 
        -------------set current boot partition: Primary 
        -------------set current boot partition: Primary  
Also, I can see update success log:

        Get current FWU state: 0x7D
        Set next FWU state: 0x7C
        P2sb is hidden, unhide it
        P2sbBar=0xFD000000, Data32=0x10
        write Data32=0x10
        Hide p2sb again.
        Read it to ensure data is written. Data32=0x10
        Exiting Firmware Update (Status: Success)
        Set next FWU state: 0x77
        Reset required to proceed.
Maybe the P2sbBar Data32 is not correct. I trigger a new update and get a full log for those who want to compare:

[topswap_512KB_fulllog.txt](https://github.com/slimbootloader/slimbootloader/files/11292505/topswap_512KB_fulllog.txt)

Thanks again. 

BR
Chrysler

closed 2023-04-21 by Chrysler2016