slimbootloader/slimbootloader #1885: [EHL] HDMI has no audio output

sharkblue2009, 2023-04-26:
Hi,

We try to enable ElkhartLake HDMI audio output by change these configs:

GRAPHICS_CFG_DATA.PchHdaEnable    | 1
GRAPHICS_CFG_DATA.PchHdaDspEnable | 1
GRAPHICS_CFG_DATA.PchHdAudioLinkMode | 0

but there is no audio device found in Ubuntu (linux-5.15.0-1027-intel-iotg), we don't know if there are other changes/configs need to enable HDMI audio.

When we set PchHdAudioLinkMode to 0x1 and the on board ALC5660 codec is working. 

The DDI1 is used for our HDMI interface.

randylintw, 2023-04-26:
Hi, 
We will check it and feedback later.

randylintw, 2023-04-27:
Hi, 
can you test this?

sharkblue2009, 2023-04-28:
ok, will test and get back to you.

sharkblue2009, 2023-05-04:
Hi Randy,

The patch works, the HDMI can output audio correctly.
Another question is, if we want to use HDMI audio and an external I2S codec such as ALC5660 at the same time,  how can we modify the HDAUDIO_VERB_TABLE, if there any guidance documents?

Thanks,

randylintw, 2023-05-16:
SBL on EHL isn't enabled the ALC5660 codec by default.

randylintw, 2023-05-16:
EHL SBL don't enable the ALC5660 [I2S codec]  by default.

sharkblue2009, 2023-05-17:
OK, Thanks, we will check if can enable it!