slimbootloader/slimbootloader #1899: [EHL]Burn Slim Bootloader from Linux

lutooog, 2023-05-15:
Hi,
I want to burn SBL from linux
refer to https://groups.io/g/slimbootloader/topic/82351644#465
SBL will lock down SPI flash write-protect at the end of Stage2 for normal boot flow. 
So, How to  disable SPI write protection in SBL?
Thank you!

stanleyintel, 2023-05-16:
Hi Lutooog,

First, I repeat what Maurice mentioned in that post first:

> For security concerns,  SBL will lock down SPI flash write-protect at the end of Stage2 for normal boot flow.   
> And as a result, it prevents any SPI write access from OS or application.  So flashrom won't work by default.
> 
> However, if you fully understand the security risk, or the platform is used in a debug / testing environment, and you want to allow runtime SPI write in OS, it is also possible to disable SPI write protection in SBL.  But it is not recommended in production since any malware in OS might be able to write to the flash.
> 

If you fully understand the risk, you may try to find "PchLockDownBiosInterface" in Stage2 code, which usually comes with "GetBootMode() == BOOT_ON_FLASH_UPDATE" condition.

lutooog, 2023-05-16:
”Fspscfg->PchLockDownBiosInterface = FALSE; “
Does it mean that SPI write protection has been disabled?

stanleyintel, 2023-05-16:
Hi Lutoong,

   It is to disable the BIOS Lock feature and keep EISS bit (D31:F5:RegDCh[5]) unset. On EHL, you may want to set it via DLT or YAML (Platform/ElkhartlakeBoardPkg/CfgData/CfgData_Silicon.yaml). Please also comment out ProgramSecuritySetting.

lutooog, 2023-05-16:
Hi,
I have setted PchLockDownBiosLock to disable and commented out ProgramSecuritySetting, but in linux,
Device is not writable.
```
root@localhost:~# mtdinfo /dev/mtd0
mtd0
Name:                           BIOS
Type:                           nor
Eraseblock size:                4096 bytes, 4.0 KiB
Amount of eraseblocks:          8192 (33554432 bytes, 32.0 MiB)
Minimum input/output unit size: 1 byte
Sub-page size:                  1 byte
Character device major/minor:   90:0
Bad blocks are allowed:         false
Device is writable:             false
```

stanleyintel, 2023-05-16:
Hi,

   Do you have PchLockDownBiosInterface = FALSE ? If possible, may you show the "PCH Lock Down Config" logs of running with DEBUG FSP?

lutooog, 2023-05-16:
Hi，
It works for me， Thank you very much！

closed 2023-05-23 by lutooog