slimbootloader/slimbootloader #1911: [EHL]how to create a new thread to monitor status of a gpio at sbl running time

lutooog, 2023-05-26:
Hi,
   I want to monitor status of a gpio at sbl running time. How to config to do that?
Thank you!

stanleyintel, 2023-05-26:
Hi,

SBL, as well as most bootloaders and UEFI BIOS, do not support multithreading.
Monitoring gpio status by multithreading might be practical, because the boot time is short (i.e., a typical SBL boot time is < 1 sec).

However, we know some bootloader developer may design a bootloader to delay loading an OS until certain GPIOs states. If you like to do so, you can implement a poll-based design (e.g., a while-loop to poll the GPIOs state). An analogy example is the OS Loader shell to read key events (see the "Shell" function in BootloaderCommonPkg/Library/ShellLib/Shell.c).

lutooog, 2023-05-26:
Hi,
What I need is a way to monitor GPIOs at all the running time(even OS is running). Or Can I use gpio interruption to do that?

stanleyintel, 2023-05-26:
SBL will hand over to OS (stop running by this mean) when OS is running. Thus, SBL itself does not run in background when OS is running.

The feature you are developing is related to (1) a safety system run by PSE (EHL only); (2) a secure OS run by hypervisor, like ACRN or KVM; or (3) a SMI handler for SMM: SBL+Osloader does not support SMM, but SBL+UEFI Payload supports it.

Maybe my colleagues can give better suggestions.

lutooog, 2023-05-26:
Thanks for your help.

lutooog, 2023-05-26:
Hi,
you mentioned that
`a typical SBL boot time is < 1 sec`
but why does FSP MemoryInit take me a long time at first boot?
```
Not a AHCI controller or Disabled
 Id   | Time (ms)  | Delta (ms) | Description
------+------------+------------+----------------------------------
 1000 |      28 ms |      28 ms | Reset vector
 1010 |      43 ms |      15 ms | Stage1A entry point
 1040 |      44 ms |       1 ms | Board PostTempRamInit hook
 1060 |      50 ms |       6 ms | Stage1A continuation
 1080 |      75 ms |      25 ms | Load Stage1B
 2000 |      75 ms |       0 ms | Stage1B entry point
 2020 |     161 ms |      86 ms | Board PreMemoryInit hook
 **2030 |   34299 ms |   34138 ms | FSP MemoryInit**
 2040 |   34308 ms |       9 ms | Board PostMemoryInit hook
 2050 |   34346 ms |      38 ms | Board PreTempRamExit hook
 2060 |   34365 ms |      19 ms | FSP TempRamExit
 2070 |   34365 ms |       0 ms | Board PostTempRamExit hook
 2080 |   34370 ms |       5 ms | Load Stage2
 2090 |   34398 ms |      28 ms | Copy Stage2 to memory
 20A0 |   34398 ms |       0 ms | Verify Stage2
 20B0 |   34399 ms |       1 ms | Decompress Stage2
 20C0 |   34399 ms |       0 ms | Extend Stage2 hash
 20D0 |   34400 ms |       1 ms | Rebase Stage2
 3000 |   34408 ms |       8 ms | Stage2 entry point
 3010 |   34436 ms |      28 ms | Board PreSiliconInit hook
 3020 |   34690 ms |     254 ms | Save NVS data
 3030 |   35125 ms |     435 ms | FSP SiliconInit
 3040 |   35128 ms |       3 ms | Board PostSiliconInit hook
 3050 |   35196 ms |      68 ms | Display splash
 3060 |   35202 ms |       6 ms | MP wake up
 3080 |   35216 ms |      14 ms | MP init run

```

stanleyintel, 2023-05-26:
Hi,

  Memory training is always a special case. Some customers (e.g., laptop and desktop) will like to save MRC Data in a special SPI region in manufacturing so it will not take long time even at first boot. However, the other customers (especially safety-related products) will ask to run memory training every time. Please find "FindNvsData" and "SaveNvsData" in to know current SBL implementation and optimize it based on your project requirement.

lutooog, 2023-05-26:
I still don't understand How to do. I just use default configuration at present. If I want to reduce boot time, Should I save MRC Data in a special SPI region? What is it MRC Data? Where do I get  MRC Data?

stanleyintel, 2023-05-26:
Hi,

MRC Data is the memory training result (trained by FSP MRC). SBL can gets it from FSP (check GetFspNvsDataBuffer).

In EHL default configuration, SBL saves MRC data in the "MRCD" in BIOS region at the first boot after flashing.
Because the default data stored in "MRCD" region is 0xFFFFFFFF (invalid), FSP must train memory and then SBL can save the result.
In the next boot, the data stored in "MRCD" is valid, so FSP can reuse it to save boot time.
Thus, every time you flash the BIOS region, the MRCD is erased and that's why the first boot is always slow.

lutooog, 2023-05-27:
So how to config to save MRC Data before first boot? Or can I not erased the BIOS region every time I flash the BIOS region?

stanleyintel, 2023-05-27:
Hi,

   You need to customize your SBL. In development or manufacturing, the MRC Data can be saved (e.g., using Dediprog in lab). Then, some customers save/load MRC Data in a different media and the other customers customize the flash and firmware update process to skip flashing the MRCD region.

stanleyintel, 2023-05-29:
Hi,

Without too many customization, this is the method: 
1. the file name of your save MRC data should be: MRCDATA.bin
2. comment out the "create MRC data" code snippet in BuildLoader.py
3. place your MRCDATA.bin in ./Build/BootloaderCorePkg/<compiler>/FV/MRCDATA.bin
Then build (but you cannot clean SBL.).

If you want to do more customization, you can copy your MRCDATA.bin to self._fv_dir (e.g., to replace the gen_file_with_size with "shutil.copyfile" in the "create MRC data" code snippet)

By the way, it is better to dump the hex binary of your MRCD while SBL calls FindNvsData, so you can verify if the saved content is correct.

lutooog, 2023-05-29:
Hi,
 I follow your steps, but it still take a long time to the first boot. I am sure that  the saved MRCD data is correct.

stanleyintel, 2023-05-29:
One thing you can debug is to compare logs from (1) 1st boot (without your MRCD; (2) 2nd boot (without your MRCD); and (3) 1st boot (with your MRCD). All logs should include (1) FSP debug messages and (2) full hex dump of MRCD in SaveNvsData() and FindNvsData()

stanleyintel, 2023-05-30:
There is one more thing you can try to ensure your existing customization does not affect the memory re-training: 

(1) flash board (without your MRCD); 
(2) 2nd boot - and type "perf" in OS Loader shell to check the boot time; 
(3) power off board; 
(4) read the whole flash from board by Dediprog and save it into a file (says, xxx.bin); 
(5) flash your board with xxxx.bin;
(6) power on, and type "perf" in OS Loader shell to check the boot time.

lutooog, 2023-06-06:
Thank you for your help. It works for me. I will close this issue.

closed 2023-06-06 by lutooog