slimbootloader/slimbootloader #1922: [EHL]How to config to test HDMI with a dynamic picture

lutooog, 2023-06-14:
Hi,
    I can see SBL logo on the HDMI display. I just want to test HDMI interface comprehensively. How to display a dynamic picture?
Thank you!

stanleyintel, 2023-06-14:
Hi,

Do you mean something like a boot animation? You can check DisplaySplash() to see how SBL display a BMP to framebuffer. The function is in BootloaderCorePkg/Stage2/Stage2Support.c. 

However, it may not what you are looking for, because it only tests with BMP.
If you want to run comprehensive tests, possibly you still need to boot into an embedded OS and run test utilities.

lutooog, 2023-06-14:
bmp is static picture. How to display boot animation?

lutooog, 2023-06-14:
Can you give me an example?

stanleyintel, 2023-06-15:
Hi,

I didn't try it, but I think it is not difficult to implement it. E.g.,
```
for (i=0; i<count; i++)
{
   MicroSecondDelay (100);
   DisplayBmpToFrameBuffer(bmp[i], ...);  // where bmp[] is a bmp pointer array
}

```
or if you don't want to manage a bmp array, you may still use a single BMP (i.e., PcdSplashLogoAddress) but modify Blt->Blue/Green/Red in DisplayBmpToFrameBuffer in every loop, so the color will be changed. E.g.,
```
DisplayBmpToFrameBuffer ( ..., loop_count) {
   ...
   Blt->Red ^= (UINT8)loop_count;
   Blt->Blue ^= (UINT8)loop_count;
   Blt->Green ^= (UINT8)loop_count;
```

randylintw, 2023-09-25:
@lutooog have you completed this feature?

randylintw, 2024-05-28:
Close it due to a prolonged period of inactivity.

closed 2024-05-28 by randylintw