slimbootloader/slimbootloader #1952: [EHL] HDMI audio output disappered when PchHdaDspEnable=1

sharkblue2009, 2023-07-03:
For EHL code, with the commit "[feat: Support HDMI audio playback on EHL]", the HDMI audio is working, but when set PchHdaDspEnable=1(we are using ALC5660 I2S codec so DSP need to be enabled), the HDMI audio output will disappered.  if there are conflicts between PchHdaDspEnable and HDMI audio? or need addtional configs?

the OS is Ubuntu 20/22 IOTG and kernel is linux-intel-iotg-5.15.74

randylintw, 2023-07-04:
If choose PchHdaDspEnable, the legacy hda will also be controlled by dsp. 
The NhltAcpiTable is not supported so that hdmi interface is disappear.

sharkblue2009, 2023-07-04:
Thanks for your quick feedback, is there any way to enable HDMI audio output and external I2S Codec(ALC5660) at the same time?

randylintw, 2023-07-06:
I am trying to see if SOF driver can be used in SBL.

randylintw, 2023-07-06:
try this. 
https://github.com/randylintw/slimbootloader/commit/059cf238db14b709ee5b956364f2fb63ad92362a

sharkblue2009, 2023-07-07:
@randylintw 
I tested with your commit, it works fine! both HDMI and ALC5660 are working.
Appreciate for your support, thanks.

closed 2023-07-07 by sharkblue2009