slimbootloader/slimbootloader #1953: [EHL] TCO Watchdog time left not count down in Ubuntu 20/22

Chrysler2016, 2023-07-06:
Hi, We are trying to use TCO watchdog feature, but it seems the watchdog doesn't work on EHL.

OS: Ubuntu 22.04.1  Kenerl: 5.19.0-32-generic

Here I list the steps:
`sudo modprobe iTCO_wdt`
`sudo dmesg`

    [ 1344.810883] iTCO_wdt iTCO_wdt: Found a Intel PCH TCO device (Version=6, TCOBASE=0x0400)
    [ 1344.811644] iTCO_wdt iTCO_wdt: initialized. heartbeat=30 sec (nowayout=0)

config /etc/default/watchdog, set

    watchdog_module="iTCO_wdt"

uncomment line in:    /etc/watchdog.conf

    watchdog-device  = /dev/watchdog

`sudo systemctl start watchdog`
`systemctl status watchdog`

    ● watchdog.service - watchdog daemon
         Loaded: loaded (/lib/systemd/system/watchdog.service; enabled; vendor preset: enabled)
         Active: active (running) since Thu 2023-07-06 09:56:34 CST; 17min ago
       Main PID: 2090 (watchdog)
          Tasks: 1 (limit: 8874)
         Memory: 540.0K
            CPU: 251ms
         CGroup: /system.slice/watchdog.service
                 └─2090 /usr/sbin/watchdog

    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  file: no file to check
    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  pidfile: no server process to check
    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  interface: no interface to check
    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  temperature: no sensors to check
    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  no test binary files
    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  no repair binary files
    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  error retry time-out = 60 seconds
    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  repair attempts = 1
    7 06 09:56:34 ElkhartLake-Embedded-Platform watchdog[2090]:  alive=[none] heartbeat=[none] to=root no_act=no force=no
    7 06 09:56:34 ElkhartLake-Embedded-Platform systemd[1]: Started watchdog daemon.

`wdctl`

      Device:      /dev/watchdog0
      Identity:    iTCO_wdt [version 0]
      Timeout:   60 seconds
      Timeleft:   60 seconds

Then simulate hang:
`sudo su`
`echo 1 > /proc/sys/kernel/sysrq`
`echo c > /proc/sysrq-trigger`

**After 2 minutes, the system is still not rebooting**.

I also tried softdog:
`sudo su`
`modprobe softdog`
`cat >> /dev/watchdog`

wati about 30 seconds, system reboots.

There are posts:
[Issue with Linux watchdog timer where system did not restart](https://community.intel.com/t5/Intel-NUCs/Issue-with-Linux-watchdog-timer-where-system-did-not-restart-for/m-p/1388055#M91502%3Fwapkw=iTCO_wdt)
But not knowing how to solve, so I submit this issue here, check if anyone meet this problem.

Thanks.

Chrysler2016, 2023-07-07:
Sorry, watchdog should be installed first.
`sudo apt install watchdog`

stanleyintel, 2023-07-09:
Hi Chrysler,

  Please modify Platform/ElkhartlakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c
to set "Fspscfg->EnableTcoTimer  = 1" in UpdateFspConfig (about line 1257, its default assignment is 0) .

Chrysler2016, 2023-07-10:
Hi, Stanley
      Thank you for your reply, I will try to update the code as you mentioned.
      I forget to mention, I previously enabled another cfg in fsp:
[image]
      It seems the WatchDogEnabled is related to AMT, not TCO.

BR
Chrysler

Chrysler2016, 2023-07-10:
Hi, Stanley.
      Thank you. After I enable the config in fsp. The timer begin to count down.
`Fspscfg->EnableTcoTimer    = 1;`

I manually disable code in iTCO_wdt_ping() to not refresh the RLD counter.
Then I use 'wdctl' to check the timeleft, in dmesg:

    [  124.127457] cccc iTCO_wdt_ping
    [  124.332443] dddd watchdog_get_status:0x8000
    [  124.332593] cccc get timeleft:35
    [  124.332597] dddd watchdog_get_timeleft:35
    [  125.085085] dddd watchdog_get_status:0x0
    [  125.085229] cccc get timeleft:34
    [  125.085233] dddd watchdog_get_timeleft:34
    [  125.128122] cccc iTCO_wdt_ping
    [  125.757983] dddd watchdog_get_status:0x8000
    [  125.758135] cccc get timeleft:33
    [  125.758139] dddd watchdog_get_timeleft:33
    [  126.128720] cccc iTCO_wdt_ping
    [  126.654054] dddd watchdog_get_status:0x8000
    [  126.654198] cccc get timeleft:33
    [  126.654202] dddd watchdog_get_timeleft:33
    [  127.129633] cccc iTCO_wdt_ping
    [  127.413863] dddd watchdog_get_status:0x8000
    [  127.414008] cccc get timeleft:31
    [  127.414012] dddd watchdog_get_timeleft:31

closed 2023-07-10 by Chrysler2016