slimbootloader/slimbootloader #1967: [EHL]HSIO configuration

hashyuan, 2023-07-18:
we want to use Lane 7 as a SGMII GBE, Lane 9 as a SGMII GBE, Lane 10 and Lane 11 as two PCIe NVMe.  Does this configuration works?

[image]

stanleyintel, 2023-07-18:
Hi Hashyuan,

   Your question is related to use either FIT or HSIO softstrap tool to configure flex I/O which should be discussed in Intel® Premier Support (IPS). Let me try to provide some info. You can get details and more help by filing an IPS.

   It is okay to assign PSE0 to Lane7 as SGMII GBE, similar for PSE1. I am not sure about the "two PCIe NVMe". Do you mean "x2 NVMe" ? The x2 NVMe M.2 SSD should work on Lane 10/11, while "two" NVMe should not work. Last, please get design review from your Intel rep.

hashyuan, 2023-07-18:
Hi Stanley,
Thanks. Yes, I mean "x2 NVMe".
The Lane 10/11 part of our design is copied from EHL-CRB. When pluged in a PCIe 4.0 NVMe, The switch on Lane7 doesn't work, the SGMII on Lane7 is fixed 2.5G link, but pluged in a SATA SSD, the switch works. So it makes me wondering what restraints cause it.

stanleyintel, 2023-07-18:
Hi,

   I didn't understand, what does it mean "the switch on Lane7 doesn't work, the SGMII on Lane7 is fixed 2.5G link"? The GbE controller should operate at 2500 Mbps for SGMII.

hashyuan, 2023-07-18:
Hi Stanley,
The SGMII between Lane7 and a switch is fixed 2.5G link. 
Without plugin any devices, the switch doesn't work too. 
[EHL-HSIO-PSE0-fails.txt](https://github.com/slimbootloader/slimbootloader/files/12077400/EHL-HSIO-PSE0-fails.txt)

stanleyintel, 2023-07-18:
Hi Hashyuan,

   Did you update FSP UPD and change fit parameter to enable SGMII? Also, did update flex I/O configuration by fit or hsio softstrap tool?

hashyuan, 2023-07-18:
Hi Stanley,
Yes.

> SILICON_CFG_DATA.PchPseGbeEnable		| { 0x02, 0x02 }
SILICON_CFG_DATA.TsnLinkSpeed		| 2
SILICON_CFG_DATA.PseTsnGbe0SgmiiEnable	| 1
SILICON_CFG_DATA.PseTsnGbe1SgmiiEnable	| 1
SILICON_CFG_DATA.PseTsnGbe0PhyInterfaceType	| 2
SILICON_CFG_DATA.PseTsnGbe1PhyInterfaceType	| 2
> 

> ('./FlexIO/PcieLaneReversalConfiguration/PCIeCtrl0LnReversal',               'No'),
#('./FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont1Conf',                 'x1 PCIe on Lane 8'),
('./FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont3Conf',                 'x1 PCIe on Lane 6'),
 ('./FlexIO/SataPcieComboPortConfiguration/SataPCIeComboPort0',               'PCIe'),
 ('./FlexIO/SataPcieComboPortConfiguration/SataPCIeComboPort1',               'GPIO Polarity PCIe'),
 ('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/SgmiPse0',            'Lane 7'),
 ('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/SgmiPse1',            'Lane 9'),
 ('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/Psegbe0PhyIfMode',    'SGMII'),
('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/Psegbe1PhyIfMode',    'SGMII'),
> 
and changed PseTsnGbeLinkSpeed to { 2, 2}

stanleyintel, 2023-07-18:
Hi,

How about:  (and use "fit" tool to check the generated IFWI)

('./FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont1Conf', 'x2 PCIe on Lane 10 and 11'),
('./FlexIO/SataPcieComboPortConfiguration/SataPCIeComboPort0', 'PCIe'),  # comment it out should be okay too
('./FlexIO/SataPcieComboPortConfiguration/SataPCIeComboPort1', 'PCIe'),  # same as above

hashyuan, 2023-07-18:
Hi Stanley,
With these changes, It get worse.

> ('./FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont1Conf', 'x2 PCIe on Lane 10 and 11'),
> ('./FlexIO/SataPcieComboPortConfiguration/SataPCIeComboPort0', 'PCIe'), # comment it out should be okay too
> ('./FlexIO/SataPcieComboPortConfiguration/SataPCIeComboPort1', 'PCIe'), # same as above

1. NO plugin, SGMII on Lane 7 fails
2. plugin a NVMe SSD，SGMII on Lane 7 fails
3. plugin a SATA SSD, SGMII on Lane 7 fails

The Soft strap configuration in sbl_ifwi_ehl.bin match the result of 615858_EHL_HSIO_Configuration_Softstraptables_Tool_rev1.1.3

> 000001c0  00 00 00 00 00 00 00 00  06 51 55 b5 0a 5a 05 00

hashyuan, 2023-07-19:
Hi Stanley,
In the TSN part of doc.601458, it mentions about MIPI PLL or SATA PLL. So the testes makes sense now.

stanleyintel, 2023-07-19:
Hi,

  Did you keep your TSN configuration in your previous test?

  Also, please use your previous configuration and add the following 2 entries to mGpioTablePreMemEhl in Platform/ElkhartlakeBoardPkg/Library/Stage1BBoardInitLib/GpioTableEhlPreMem.h

  {GPIO_VER3_GPP_G12, {GpioPadModeNative3, GpioHostOwnGpio, GpioDirNone, GpioOutDefault, GpioIntDis, GpioHostDeepReset, GpioTermNone}},  //M.2_SSD_SATAXPCIE_1
  {GPIO_VER3_GPP_E8,  {GpioPadModeNative2, GpioHostOwnGpio, GpioDirNone, GpioOutDefault, GpioIntDis, GpioHostDeepReset, GpioTermNone}},  //M.2_SSD_SATA_DEVSLP_1

hashyuan, 2023-07-19:
Hi Stanley,
still the same result.
1.NO plugin, SGMII on Lane 7 fails
2.plugin a NVMe SSD，SGMII on Lane 7 fails
3.plugin a SATA SSD, SGMII on Lane 7 fails

With the testes I have done, it seems when Lane7/8/9 is configured as SGMII,  the SATA must be enabled, there will be no x2 NVMe M.2 SSD option on lane 10/11. 
Please correct me if I am wrong.

stanleyintel, 2023-07-19:
Hi,

   I cannot understand about your test results. When you use your previous configuration plus the GPIO code change, supposedly, it will go back to "When plugged in a PCIe 4.0 NVMe, The switch on Lane7 doesn't work, the SGMII on Lane7 is fixed 2.5G link, but plugegd in a SATA SSD, the switch works"

stanleyintel, 2023-07-19:
here, the "previous configuration" may mislead you? I mean YOUR original configuration.

stanleyintel, 2023-07-19:
May you check with your "previous configuration" to see if you still get the same result? If so, then add the GPIO code change.

hashyuan, 2023-07-19:
Hi Stanley,
The original config (with/without the GPIO code changes):

> SILICON_CFG_DATA.PchPseGbeEnable | { 0x02, 0x0 }
> #SILICON_CFG_DATA.TsnLinkSpeed | 2
> SILICON_CFG_DATA.PseTsnGbe0SgmiiEnable | 1
> #SILICON_CFG_DATA.PseTsnGbe1SgmiiEnable | 1
> SILICON_CFG_DATA.PseTsnGbe0PhyInterfaceType | 2
> #SILICON_CFG_DATA.PseTsnGbe1PhyInterfaceType | 2
> 
> ('./FlexIO/PcieLaneReversalConfiguration/PCIeCtrl0LnReversal', 'No'),
> #('./FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont1Conf', 'x1 PCIe on Lane 8'),
> ('./FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont3Conf', 'x1 PCIe on Lane 6'),
> ('./FlexIO/SataPcieComboPortConfiguration/SataPCIeComboPort0', 'PCIe'),
> ('./FlexIO/SataPcieComboPortConfiguration/SataPCIeComboPort1', 'GPIO Polarity PCIe'),
> ('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/SgmiPse0', 'Lane 7'),
>#('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/SgmiPse1', 'Lane 9'),
> ('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/Psegbe0PhyIfMode', 'SGMII'),
> #('./NetworkingConnectivity/TimeSensitiveNetworkingConfiguration/Psegbe1PhyIfMode', 'SGMII'),
>
---"When plugged in a PCIe 4.0 NVMe, The switch on Lane7 doesn't work, the SGMII on Lane7 is fixed 2.5G link, but plugegd in a SATA SSD, the switch works"
or
1.NO plugin, SGMII on Lane 7 fails
2.plugin a NVMe SSD，SGMII on Lane 7 fails
3.plugin a SATA SSD, SGMII on Lane 7 works

--------------------------------------------------------------------------------------
The previous config (with/without the GPIO code changes):
> SILICON_CFG_DATA.PchPseGbeEnable | { 0x02, 0x02 }
> SILICON_CFG_DATA.TsnLinkSpeed | 2
> SILICON_CFG_DATA.PseTsnGbe0SgmiiEnable | 1
> SILICON_CFG_DATA.PseTsnGbe1SgmiiEnable | 1
> SILICON_CFG_DATA.PseTsnGbe0PhyInterfaceType | 2
> SILICON_CFG_DATA.PseTsnGbe1PhyInterfaceType | 2

> ('./FlexIO/PcieLaneReversalConfiguration/PCIeCtrl0LnReversal', 'No'),
> #('./FlexIO/PcieMultiVcPortConfiguration/PCIeMltVcCont1Conf', 'x1 PCIe on Lane 8'), [...]
---
1.NO plugin, SGMII on Lane 7 fails
2.plugin a NVMe SSD，SGMII on Lane 7 fails
3.plugin a SATA SSD, SGMII on Lane 7 fails

hashyuan, 2023-07-19:
Hi Stanley,
The Figure 9-3 from Doc.636112 explains the test results.  Maybe codes has changed, please correct me.

[image]

stanleyintel, 2023-07-19:
Hi Hashyuan,

  May you try with "fit.exe"? Thus, you may generate an IFWI image first. And then use fit.exe to load it and modify the "Networking & Connectivity" and "Flex I/O" settings.

hashyuan, 2023-07-19:
Hi Stanley,
Could The "Networking & Connectivity" and "Flex I/O" settings the linux fit generated be wrong? So using the windows fit.exe to check the setting?
Do you think "Lane 7 as a SGMII GBE, Lane 9 as a SGMII GBE, Lane 10/11 as x2 NVMe" could work?
Have you try Lane10/11 as PCIe, SGMII on Lane 7 works? Sorry, I don't have a EHL-CRB to test.

stanleyintel, 2023-07-19:
Hi,

   You may file an IPS case to get more support from HSIO experts and CSME (fit tool) team. I can keep helping you once the solution is linked to slimbootloader (source).

hashyuan, 2023-07-20:
Hi Stanley,
Thanks！

closed 2023-07-20 by hashyuan