slimbootloader/slimbootloader #1987: Managing New Version of FSP Binaries

Joursoir, 2023-08-10:
Hello,

I have encountered an issue with FSP binaries on the Alder Lake-S board. About a month ago, I successfully built SBL and set up everything without any issues. However, when I tried to update the git repository yesterday and compile the code, an error occurred. After a bit of investigation, I traced the problem back to commit 9b8d2cdce12a9ed49e0685bbed1d9aff3b8826e2 (this commit was supposed to fix things but ended up causing some unexpected issues :D). 

As I delved deeper into the FSP-related problem, I have faced commit b3388892520c805a6c7c227cd9e8df59f63b5887, which actually modifies the FSP binaries that should be used. So, I had to manually delete FSP files. `BuildLoader.py` does NOT clean up `Silicion/{platformPkg}/{board}` directory (in my case `Silicon/AlderlakePkg/Adls`).

I'm uncertain if there exists an option for changing FSP files not manually, but in case it doesn't, I'm eager to introduce such a feature or find a workaround. Currently, I'm going to make modifications to the `PrepareBuildComponentBin.py` script. My aim is to check not only existence but also considers whether files have been changed.

I would appreciate any insights & thoughts you might have!

vincentchchen, 2023-08-11:
Hi,

The first build issue you encountered is due to the variable type change in FSP header file, from one-byte UINT8 to two-byte array.
https://github.com/intel/FSP/blob/master/AlderLakeFspBinPkg/IoT/AlderLakeS/Include/FspsUpd.h#L680
For ADL-S, this change is introduced since the version "IoT ADL-S MR5 (0C.00.A4.13)", to support individual TSN ports Enable/Disable.

So on SBL side, need to apply the commit to update the FSP headers and binaries accordingly.
https://github.com/slimbootloader/slimbootloader/commit/24b62383addeabfb86f87c66be739d4ba96a68c4

Regarding the option you mentioned for deleting FSP files, this patch probably fulfills the request.
https://github.com/slimbootloader/slimbootloader/commit/eb98e8b8aa07f6f04d43fcf5499c6b23d011e160
You could add "board_name" in the clean command. e.g. python BuildLoader.py clean adls

Joursoir, 2023-08-11:
Thank you for confirming my assumptions. The command `python BuildLoader.py clean adls` does exactly what I need.

closed 2023-08-11 by Joursoir