slimbootloader/slimbootloader #2001: [EHL] soft-power-off can't shutdown board power

sharkblue2009, 2023-08-22:
when we do a soft-power-off from Ubuntu Linux/Win10 system, the OS can go off, monitor is off, but the power on board can't be shutdown, we measured the signal PMC_SLP_S3_N/PMC_SLP_S4_N/PMC_SLP_S5_N are all NOT asserted at this time,  what's the problem? The PMC_SLP_Sx_N pins are configured as the functions by FIT tool. and we are using SBL_MR6 with Intel_CSE_15.40.30.2879. 

As comparison, pushing PowerBotton for couple seconds, the power on board can be shutdown, which is in S5 state.

stanleyintel, 2023-08-22:
Hi Sharkblue,

  Can you test with the command: "reset off" under **OS Loader**? Before testing it, please cherry pick the patch in the PR: https://github.com/slimbootloader/slimbootloader/pull/2002.

sharkblue2009, 2023-08-23:
Hi Stanley,

By using the CRB BIOS “Elkhart_Lake_Platform_CRB_IFWI_v5083_00” decomposed images, the "power down" and "reset defer"  issue do not exist anymore.

We are using FIT tool to stitch the final BIOS image, compare with previous failed situation, the CSE/PMC/PCH/SI images are from Intel_CSE_15.40.30.2879 package,  although the images’ version string shown in the FIT tool is the same, the binary images are different with the images decomposed from the CRB BIOS,.

closed 2023-08-30 by sharkblue2009