slimbootloader/slimbootloader #2012: [EHL] How to enable SATAXPCIE auto detect function.

sharkblue2009, 2023-08-30:
We want to enable the HSIO lane 10/11 SATAXPCIE port auto detect function, what we did include:

1. set GPIO E0 to function 2.
     {GPIO_VER3_GPP_E0, {GpioPadModeNative2, GpioHostOwnGpio, GpioDirNone, GpioOutDefault, GpioIntDis,     GpioHostDeepReset, 
        GpioTermNone}}
2. use FIT tool to configure the Flex IO->Sata/Pcie combo ports to "GPIO Polarity PCIE" or "GPIO Polarity SATA"

But the auto detect function is not working,  what we missed?

stanleyintel, 2023-08-31:
Hi Sharkblue,

   In addition to GPIO E0 to function 2 and Fit configuration, you may do the following check:

   (GPIO E0)
    1. *Not* enable SATA LED (i.e., FSPs UPD SataLedEnable should be zero)
    2. Remove existing GPP_E0 configuration in SBL code, e.g., mGpioTablePreMemEhl in Platform/ElkhartlakeBoardPkg/Library/Stage1BBoardInitLib/GpioTableEhlPreMem.h
    3. Check MMIO register 0xFD6A0A70 (e.g., use "mm" command under OSLoader). Its [12:10] should be 2h.

   (FIT)
    1. SGMII Host 0 is either disabled or on Lane 8.
    2. SGMII PSE 1 does not occupy lane 11 
    3. PCIe Multi VC Controller 1 is disabled

   There are also some HW settings should be checked, but I am not a HW guy to tell it. According to Table 21-1 in EDS vol1, it mentions a 20K pullup is required. And also please check all requirements related to PCIe and SATA in PDG.

   If nothing incorrect in all checks, please file an IPS case (Intel Premier Support) to get help from our HW team.

sharkblue2009, 2023-09-01:
Hi Stanley,

1. The SATA LED is disabled by default , the value in FSP.bsf of "SataLedEnable“ is zero
2. the mGpioTableEhlSimics in GpioTableEhlPreMem.h is not used in any code, so GPP_E0 configuration is not used.
3. we checked the register 0xFD6A0A70, value is 0x44000B03, which bits [12:10] is 2h.

In FIT, Sgmii host is disabled, PSE1 sgmii is lane 7, and PCIE Multi VC 1 is disabled.

We are using PCIE_REFCLK3 for this slot,  our current value for "PcieClkSrcUsage"  in FSP is:
{ 0x00, 0x06, 0x04, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF }
which value should we change?

in FIT, which value should we use for the "Flex IO->Sata/Pcie combo Port"?

stanleyintel, 2023-09-01:
Hi Sharkblue

   About configuring GPP_E0, it is to check "mGpioTablePreMemEhl".  Because your register value is already correct, it is okay to not touch it.

   About PcieClkSrcUsage, please check RDC# 638817.

   About FIT, it depends on your design:  set "GPIO Polarity PCIe" if GPP_E0 = 1 for PCIe in your design; similar for "GPIO Polarity SATA (0 for PCIe).

sharkblue2009, 2023-09-04:
Hi Stanley,

I noticed there is FSP config which may relate this: 

  - FiaLaneReversalEnable :
      name         : FIA Lane Reversal Enable/Disable config mask
      type         : EditNum, HEX, (0x0,0xFF)
      help         : >
                     Set 0x10 for M.2 NVME storage support on J4U1.
      length       : 0x2
      value        : 0x0

But I tried to set the value to 0x10 or 0x1, it seems no effect.

I tried the debug FSP, the PcieRp Init log is:
-----------------
InitPcieSingleRootPort (5) Start
VID: FFFF DID: FFFF  MLS: 0
Pcie RP Multi VC: V0CTL = 0x800000FF
Printing PCIe equalization settings
PcieLinkEqMethod = 0
PcieLinkEqMode = 1

It should mean the pcie device is detected, but the VID/DID read out value is invalid. we had checked the CLKREQ/REF_CLK, these signals are ok.

Anything else need to check?

stanleyintel, 2023-09-04:
Hi,

 Few questions:

 1. Is that a x1 or x2 PCIe device? Also tell us your current configuration of "Flex IO->Sata/Pcie combo Port 0" and "Sata/Pcie combo Port 1" in Fit.
 2. What's the values of your PcieClkSrcUsage?  Also, not sure about the 1st index of your mentioned PCIE_REFCLKx. Is that PCIE_REFCLK0 or PCIE_REFCLK1 ?

sharkblue2009, 2023-09-05:
we are using an x4 NVME disk to test and it compatible with x1 & x2 mode.  Both "Flex IO->Sata/Pcie combo Port 0" and "Sata/Pcie combo Port 1"  are set to "GPIO polarity Pcie",  I also tested with only Port 0 set to "polarity Pcie" and  Port 1 disabled.

PcieClkSrcUsage  | { 0x00, 0x06, 0x04, 0x04, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF }
                                                               |
                                                               L  we are using PCIE_CLK3 for the auto-detection slot. (PcieRp 5) 

The first index is PCIE_REFCLK0, we are connect PCIE_CLK3 to this slot.

On the slot side, we are using PCIE_8 lane for slot's PE_0 and PCIE_9 lane 9 for slot's PE_1( No lane reversal), which is different with the CRB board,  so I think we should keep the "FiaLaneReversalEnable"  to zero.

stanleyintel, 2023-09-05:
- PcieClkSrcUsage[3] should not be 4.

- It is correct to have No Lane Reversal: EDS notes that lane reversal is not applicable for x2 controller (i.e., PCIE 1)

- Did you get a "Port 5 has a device attached" in your log?

- what are the values of SILICON_CFG_DATA.PcieRpClkReqDetect and MEMORY_CFG_DATA.PcieClkSrcClkReq?

- how about the status of GPP_E0 in your test?

sharkblue2009, 2023-09-05:
yes, I got "Port 5 has a device attached." in boot log.

When NVME card plug in, GPP_E0 is high.

PcieClkSrcClkReq and PcieRpClkReqDetect  is the default value:
--------------
PcieRpClkReqDetect |   { 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }
PcieClkSrcClkReq  |  { 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F }
--------------

For PcieClkSrcUsage[3] = 4, what I think 4 is for PchPcie4, which is Pcie MultiVC controller 1.  is it wrong? what value shoud use?

stanleyintel, 2023-09-05:
It's typo, I mean PcieClkSrcUsage[2] should not be 4. We should not have 2 clocks for one RP.

Is GPP_H10 also designed as ClkReq for RP#4 in your board design? If so, it should be configured as Native Function 1. You can check its mmio value at 0xFD6D08A0 where [12:10] should be 1.

sharkblue2009, 2023-09-06:
we are not using GPP_H10 and this pin is float. 

I had changed the PcieClkSrcUsage[2] to 0xff to avoid conflict with CLK_REQ3, but the result is the same. 

"
InitPcieSingleRootPort (5) Start
VID: FFFF DID: FFFF MLS: 0
"

stanleyintel, 2023-09-06:
Hi,

  1. What's the log related to "Multilane check for PCIE Controller 1" from your side?
  2. When executing "pci" under Shell or "lspci" under Linux, can you find B0:D28.F4?

stanleyintel, 2023-09-06:
Also, if ClkReq is unused, the PcieRpClkReqDetect[4] should be 0.

sharkblue2009, 2023-09-06:
The log:
Multilane check for PCIE Controller 0: Disabling RP 2/3/4
Multilane check for PCIE Controller 1: No ports disabled
Multilane check for PCIE Controller 2: No ports disabled
Multilane check for PCIE Controller 3: No ports disabled

For "pci" command,  I didn't find B0:D27:F4, is it pci device id?

stanleyintel, 2023-09-06:
B0:D28.F4, I mean bus 0, device 28 (1C), function 4, which is PCIe Root Port #4. Its device id is 4B3C.

Please check the [11:8] of MMIO reg SC3 (0xFDCF0288) should be 1100b.

sharkblue2009, 2023-09-06:
Ok, got it, I only saw the 00:1c.0 exist but no 00:1c.4  for "pci" command.

>  00   00   1C    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 8086 Device 4B3C Prog Interface 0

> 

The value:
0x00000000FDCF0288 : 0x0000CCA0
which [11:8] is 1100b

The complete log for the Port  init is:

> InitPcieSingleRootPort (5) Start
VID: FFFF DID: FFFF  MLS: 0
Pcie RP Multi VC: V0CTL = 0x800000FF
Printing PCIe equalization settings
PcieLinkEqMethod = 0
PcieLinkEqMode = 1
Ph1DownstreamPortTransmitterPreset = 7
Ph1UpstreamPortTransmitterPreset = 5
Ph3NumberOfPresetsOrCoefficients = 3
Phase 3 Coefficient 0, pre-cursor = 6, post-cursor = 8
Phase 3 Coefficient 1, pre-cursor = 6, post-cursor = 12
Phase 3 Coefficient 2, pre-cursor = 8, post-cursor = 8
Phase 2 local transmitter override enable = 0
Phase 2 local transmitter override preset = 0
PipePresetToCoefficientQuerySupported = 1
PCIe EQ: Setting coefficient 0, pre-cursor = 6, post-cursor = 8
PCIe EQ: Setting coefficient 1, pre-cursor = 6, post-cursor = 12
PCIe EQ: Setting coefficient 2, pre-cursor = 8, post-cursor = 8
RootportDownstreamConfiguration 1C:4
RecursiveBusAssignmentE 0:1C:4 -> 0,2,F
RecursiveBusAssignmentL 0:1C:4 -> 0,2,2
RecursiveMpsCheck 0:1C:4
HasChildBus00:1C:04: yes, 2
RecursiveMpsConfiguration 0:1C:4 : MPS 1
HasChildBus00:1C:04: yes, 2
RecursiveIoApicCheck 0:1C:4
HasChildBus00:1C:04: yes, 2
IoApic status 0 @0:0:1C:4
RecursiveCccConfiguration 0:1C:4
HasChildBus00:1C:04: yes, 2
RecursiveBusAssignmentE 0:1C:4 -> 0,2,F
RecursiveBusAssignmentL 0:1C:4 -> 0,2,2
MultiVcConfiguration for 0:1C:4
RecursiveMultiVcConfig for 0:1C:4
MyExtVc.CapOffset 00000280
Extended Multi VC Count 1
HasChildBus00:1C:04: yes, 2
RootPort 4 supports 1 ExtendedVCD

>

stanleyintel, 2023-09-06:
It looks like you have the following x2 configuration, right?
- "Flex IO -> PCIe Multi VC Controller 1 = "Disabled"
- "Flex IO -> Sata/Pcie combo Port 0" = "GPIO Polarity PCIe"
- "Flex IO -> Sata/Pcie combo Port 1" = "GPIO Polarity PCIe"
I believe if you use a hex editor to check the IFWI, the values at offset 0x1CD and 0x1CE should be: C0 and 0C.  (under default FLMAP1.FPSBA)

From my side, I cannot find anything abnormal, in terms of:
  - the PCIE_1 is not disabled
  - Fit has (1) "Flex IO->Sata/Pcie combo Port 0" = "GPIO Polarity PCIe" ; (2) "Flex IO->Sata/Pcie combo Port 1" = "GPIO Polarity PCIe" ; (3) PCIe Multi VC Controller 1 is disabled; (4) SGMII PSE 1 on Lane9.
  - GPP_E0: correctly configured as native function 2 (verified by reading MMIO register); and test PCIe detection with HIGH
  - REFCLK3 is in-used and PcieClkSrcUsage[3] = 4
  - No FIA Lane reverse on Lane10/11, complying with your hw design.

Please file an IPS to get help from our HW team and HSIO expert.

stanleyintel, 2023-09-06:
I mean, I cannot find anything incorrectly configured. Therefore, it is better to file an IPS to get review by HW team and HSIO experts.

sharkblue2009, 2023-09-06:
Stanley, 

Thanks, very appreciate for your support!

I will file an IPS as your suggestion.

stanleyintel, 2023-09-06:
One notable thing is that you can find the device 4B3C, which is the PCIe RP#4. Wondering to know if your PCIe device is already present (under Shell or OS)?

stanleyintel, 2023-09-07:
Another test you can try: configure LANE10 and 11 as PCIe1 x 2 mode only (PCIe Multi VC Controller 1 = "x2 PCIe on Lanes 10 and 11").  Just to rule out any hw issue.

sharkblue2009, 2023-09-08:
I tried this configure, for "PCIe Multi VC Controller 1" to Lane 10 & 11 or Just Lane 10, but the result is the same, the VID/DID read out for the device is still 0xFFFF.

One more question, if the PCIE Lane "Polarity Inversion" need configuration in BIOS?  One LANE RX side, we used the feature.