slimbootloader/slimbootloader #2050: UDK Url at slimbootloader.github.io is wrong

hanetzer, 2023-10-15:
Sorry, couldn't find a better place to file this.

The url given at https://slimbootloader.github.io/developer-guides/debugging-with-udk.html#prerequisite
to get Intel UDK Debugger for linux (https://firmware.intel.com/develop/intel-uefi-tools-and-utilities/intel-uefi-development-kit-debugger-tool#overlay-context=develop) does not appear to actually have a download for this tool.

I managed to obtain a copy of version 1.5.1 elsewhere, but I think perhaps the edk version I'm
building SourceLevelDebugPkg from may not be compatible with this version; while it does handshake:

```
Send INIT break packet and try to connect the HOST (Intel(R) UDK Debugger Tool v1.5) ...
HOST connection is successful!
```
It never actually spawns the gdbserver on `:1234` to connect to.

ongeelim, 2024-06-19:
@stanleyintel I would like to check if Intel UDK Debugger for Linux download link is still available in Intel website? Or is the tool already deprecated? Thank you.

randylintw, 2024-06-20:
@ongeelim , from Intel® UEFI Development Kit (Intel® UDK) Debugger Tool (version 1.5) Configuration and Setup Guide, 
the https://uefidk.intel.com/ isn't accessible: 504 DNS look up failed.

randylintw, 2025-01-09:
no activity for this issue for a long time, close it.

closed 2025-01-09 by randylintw

hanetzer, 2025-06-26:
There's nothing I or anyone else can do to fix this error and no other info we can provide. Inactivity is a problem on your/intel's end, not us.