slimbootloader/slimbootloader #2053: GetContainerHeaderSize in ContainerLib returns 0

TagSouth, 2023-10-18:
Function GetContainerHeaderSize in BootloaderCommonPkg/Library/ContainerLib/ContainerLib.c rejects all containers as invalid, returning a size of 0.

By the time line 82 
`if ((Offset < 0) || (Offset >= ContainerHdr->DataOffset)) {`
is reached on the last iteration, Offset is the distance between (one byte beyond the last component entry) and the start of the container header.  But that is the value stored in ContainerHdr->DataOffset.  So a well-formed container is always rejected.

Replacing that line with
`if ((Offset < 0) || (Offset > ContainerHdr->DataOffset)) {`
makes the function return a non-zero offset.

But it would be a lot simpler to replace the whole function with a
`return ContainerHdr->DataOffset;`

bejeanmo, 2023-10-31:
Can you attach a container or container header where this is the case? ContainerHdr->DataOffset will be aligned to 4KB (or larger, if a sub container uses a larger alignment), so the actual container header size will typically be less than the data offset. 

There may be an edge case here you pointed out where the last component entry ends at exactly the alignment size which would trigger the issue, in which case changing the **>=**'to **>** is likely the correct solution. However, returning ContainerHdr->DataOffset is typically incorrect.

TagSouth, 2023-10-31:
[sbl_os.zip](https://github.com/slimbootloader/slimbootloader/files/13219392/sbl_os.zip)
Here is the file I am trying to load.  I know that the operating system inside has issues, as it still doesn't boot even with my change above, but I think its structure as a container file is correct.

bejeanmo, 2023-10-31:
Yes, it looks like your container structure is more or less sane. Your component alignment is 16-byte and the components have no authentication data. Both of these things are fine, but it means the DataOffset falls immedately after the last component entry in the header. The comparison change to greater than should resolve this issue.

PR #2068 (2023-11-02): fix: Container header parsing failure in some cases.
GetContainerHeaderSize() parses header component entries to find the offset of the end of the last component entry and checks this against ContainerHdr->DataOffset. However, there is a valid case where this can exactly equal to ContainerHdr->DataOffset, when the component alignment is small enough and there is no component auth data (AuthType is none). This check is changed to fail on greater than instead of greater or equal to cover this case. fixes #2053
closed 2023-11-02 by gdong1